patman15/BMS_BLE-HA #153: Add seplos stacked batteries attributes

ouinouin, 2025-01-24:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hi, first of all thanks for your implementation of the seplos BLE protocol.
I ows 2 set of 48V LIFEPO4 batteries equipped with seplos 3.0 BMS rev 11.
while stacking multiple seplos bms and connecting the bms with RJ45 cable, they do something clever , only the first battery is reporting  through BLE , and its reproting all theinfos from all other racks, so for examples intead of having 16 attributes for the cells voltage, i have 32 attributes for the cell voltage hidden behind the deltaV sensor.
but on the official application, we can also see , the voltage for both units , the SOC for bith units  etc etc.
i see this is working for the temperatures and for the cells voltages (i see 8 temperatures (4 per rack). it would be nice to see at least as attributes all individual values, specially for voltage and SOC.

my set of 2 racks is available to test if you re interrested, i m not a coder but can fiddle around if necessary

ouinouin, 2025-01-25:
Hi, @patman15 
happy to give a try , the calm evening is also a prerequisite here :-) .

ouinouin, 2025-02-06:
@patman15  ok, i ll give a try , dont expect new for the upcoming weeks ( rotational job). i ll be able to test end of feb.

wrobela, 2025-02-06:
Hi 

I was able to download and test this branch. I can see data from 2 packs. 
It works for:
SOC, Cycles, Voltage, Current

[image]

[image]

Thank you!

ouinouin, 2025-02-06:
@wrobela , excellent, is it working for power ?  (indeed your batteries SOC seems to diverge.... a bit :-)

wrobela, 2025-02-06:
Power is not shown for packs, only one value is available. Thanks to this test I know there is something wrong with my packs (I need to take a closer look)

[home-assistant_bms_ble_2025-02-06T18-47-15.032Z.log](https://github.com/user-attachments/files/18694810/home-assistant_bms_ble_2025-02-06T18-47-15.032Z.log)

patman15, 2025-02-06:
@wrobela thanks a lot for the logs. Looks good to me. Please report in case you find an issue, otherwise I will merge with the next release.
@ouinouin the updated [README](https://github.com/patman15/BMS_BLE-HA/tree/153-add-seplos-stacked-batteries-attributes?tab=readme-ov-file#provided-information) shows the available attributes. I decided only to implement the ones reported directly by the BMS. You can of course get the power, by setting up a template sensor, multiplying voltage with current.

closed 2025-02-09 by patman15