patman15/BMS_BLE-HA #154: list devices - logs

erlendoyen, 2025-01-24:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi

i see that this should autodetect supported batteries.

where ca you see logs etc for devices that's been discovered but not supported?

i don't see any new sesors to add anywhere

batmon is able to retrieve data from the bms'es

patman15, 2025-01-25:
Hi! If you are looking for devices that are not automatically added, please see this [README section](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#if-your-device-is-not-recognized) especially list item 4 and 5.

Which configuration do you use for batmon? If you provide me that info and the output of `bluetoothctl info <MAC>` I can correct the auto detection.

erlendoyen, 2025-01-26:
aha ok, i just read Zero configuration

config batmon

- address: a5:c2:37:2a:c3:44
  type: jbd
  alias: RE25J001-1
- address: a5:c2:37:31:21:6b
  type: jbd
  alias: RE25J001-2

batmon is working now, so it will most likely use that

patman15, 2025-01-26:
> aha ok, i just read Zero configuration

Well it works for **supported** batteries and yours is not yet known, so I need to add it. After that, you do not have to do anything. I still would need the output of `bluetoothctl info <MAC>`

> batmon is working now, so it will most likely use that

Ok, if you are happy, you can also just close the issue.

closed 2025-01-29 by erlendoyen