patman15/BMS_BLE-HA #155: New Integration? DC House 12V LiFePo4 battery with bluetooth

erikeidt, 2025-01-25:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I am attempting to connect a new BMS, I think.  Here's what I have from "bluetoothctl connect A5:C2:37:2E:A1:17" :

Attempting to connect to A5:C2:37:2E:A1:17
AdvertisementMonitor path registered
hci0 CE:2A:42:46:1E:1C type LE Random connected eir_len 31
[CHG] Device CE:2A:42:46:1E:1C Connected: yes
hci0 type 6 discovering on
[CHG] Device 00:0D:6F:DB:D6:4F RSSI: 0xffffffa2 (-94)
[CHG] Device 4A:8E:15:BB:83:61 RSSI: 0xffffff9e (-98)
[CHG] Device 4A:8E:15:BB:83:61 AdvertisingFlags:
  1a                                               .               
[CHG] Device A5:C2:37:2E:A1:17 RSSI: 0xffffffbc (-68)
hci0 type 6 discovering off
hci0 A5:C2:37:2E:A1:17 type LE Public connected eir_len 24
[CHG] Device A5:C2:37:2E:A1:17 Connected: yes
hci0 type 6 discovering on
[CHG] Device 70:09:71:A0:E2:21 RSSI: 0xffffffa0 (-96)
[CHG] Device 14:B4:57:11:E8:4F RSSI: 0xffffff9e (-98)
[CHG] Device 13:54:A2:03:F2:3C RSSI: 0xffffffc0 (-64)
[CHG] Device 7A:36:E8:34:B0:94 RSSI: 0xffffffa4 (-92)
Connection successful

I'm not sure what to do next, any pointers much appreciated, thanks in advance, Erik

### Additional context

https://play.google.com/store/apps/details?id=com.eco.dchouse

patman15, 2025-01-25:
Hi! Basically I would need a bit more info, if possible for you, as described in the hints for a feature request:
> For new BMS types, please provide the output of `bluetoothctl info <MAC>` and the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py). To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

Worst case I can check from the app if I can recognize the BMS type.
Best regards, Patrick

erikeidt, 2025-01-25:
Device A5:C2:37:2E:A1:17 (public)         
        Name: DP04S007L4S120A                          
        Alias: DP04S007L4S120A                         
        Paired: no                                     
        Bonded: no                        
        Trusted: no                                    
        Blocked: no                                    
        Connected: no                                  
        LegacyPairing: no                              
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
        Modalias: usb:v248Ap8266d0001                             
        ManufacturerData.Key: 0xa117 (41239)                      
        ManufacturerData.Value:                              
  2e 37 c2 a5                                      .7..
        AdvertisingFlags:                                         
  06                                               .

Sometimes "Connected: yes"

I tried running the python script as follows:

Action

Python Scripts: service_explorer

action: python_script.service_explorer
data:
  name: "A5:C2:37:2E:A1:17"

And got:
Failed to process the returned action response data, expected a dictionary, but got <class 'NoneType'>.

patman15, 2025-01-26:
duplicates #157

closed 2025-01-26 by patman15