patman15/BMS_BLE-HA #157: JBD - DC House battery not detected

mbeeuid, 2025-01-26:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

DC House: DBJBD (JBD) Firmware ver: 6.6 Device: DP04S007L4S120A

### Describe the issue

DC House battery not discovered even though it appears to use a JBD BMS.  I see a new "DCHouse" Branch, will this cover this particular BMS? (155-new-integration-dc-house-12v-lifepo4-battery-with-bluetooth)

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

Battery not found.  (no log)

patman15, 2025-01-26:
```
Device A5:C2:37:2E:A1:17 (public)         
        Name: DP04S007L4S120A                          
        Alias: DP04S007L4S120A                         
        Paired: no                                     
        Bonded: no                        
        Trusted: no                                    
        Blocked: no                                    
        Connected: no                                  
        LegacyPairing: no                              
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
        Modalias: usb:v248Ap8266d0001                             
        ManufacturerData.Key: 0xa117 (41239)                      
        ManufacturerData.Value:                              
  2e 37 c2 a5                                      .7..
        AdvertisingFlags:                                         
  06                   
```

patman15, 2025-01-26:
>  I see a new "DCHouse" Branch, will this cover this particular BMS? (155-new-integration-dc-house-12v-lifepo4-battery-with-bluetooth)

Yeah, I yesterday started working on it but did not come far. Please do not open new issues for the same topic. I will now switch over to this issue and add the necessary changes in a branch which is also linked to this ticket.

patman15, 2025-01-26:
@mbeeuid please try the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/157-jbd---dc-house-battery-not-detected) and provide me a debug log, so I can verify the detection and add matching tests for the future.
Thanks, Patrick

mbeeuid, 2025-01-26:
Sorry Patman, I just got called out of the country and will not be able to assist until next week.  Just sent you some "tea money" though for the prompt attention!  Thanks again!

patman15, 2025-01-29:
Thanks, I'll take some tea till you are back. 😃 Let me know if you are ok then I'll merge the changes to the next release.

closed 2025-02-09 by patman15

mbeeuid, 2025-02-10:
V1.11.7 fixed my detection.  Batteries are now connected to HA!  Thank you very much, more tea on the way!

commit bbc15fc4 (2025-02-09): Fix ECO-WORTHY, DCHOUSE detection due to wrong manufacturer ID (#180)

* fix ECO-WORTHY detection due to wrong manufacturer ID
* also fixes #157 for DCHOUSE detection