patman15/BMS_BLE-HA #160: No supported devices found, but bms_ble Bluetooth device detected

herijaya, 2025-01-29:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Daly BMS 16S

### Describe the issue

Why is the integration not detecting BLE devices through the Bluetooth proxy? Is there an additional configuration or a limitation that I might be missing?

Thank you for your help!

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

Here are relevant logs from both the ESP32 proxy and Home Assistant (debug mode enabled for the integration):

2025-01-29 09:49:03.508 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=DL-401805011A26 address=40:18:05:01:1A:26 rssi=-55 manufacturer_data={770: b'\x16@\x18\x05\x01\x1a&DL'} service_data={} service_uuids=[] source=58:BF:25:9E:97:E0 connectable=True time=92661.17286896706 tx_power=-127>

2025-01-29 09:49:03.513 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=DL-401710015C9B address=40:17:10:01:5C:9B rssi=-57 manufacturer_data={770: b'\x16@\x17\x10\x01\\\x9bDL'} service_data={} service_uuids=[] source=58:BF:25:9E:97:E0 connectable=True time=92665.28482341766 tx_power=-127>

2025-01-29 09:50:12.596 DEBUG (MainThread) [custom_components.bms_ble] user step

patman15, 2025-01-29:
Hi! Reason is that the services of the device are not resolved yet. Can you please add the ESP32 proxy log in `VERY_VERBOSE` mode or provide me the advertisement information using e.g. [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)?

wmichael3, 2025-01-30:
Hey, confirming that you don't support SOK BMS. Here is what is in my ESPHome Log under VERY_VERBOSE. I get the "No Supported Devices found using Bluetooth" message (immediately) when attempting to add the integration. The HA server itself does not have a Bluetooth radio, so all is via ESPHome. In this case an Atom Lite, which is an  ESP32-PICO, a dual core device with 4Mb.

[12:02:26][VV][esp32_ble_tracker:453]:   Address: D6:6C:0A:62:9B:D6 (PUBLIC)
[12:02:26][VV][esp32_ble_tracker:456]:   RSSI: -94
[12:02:26][VV][esp32_ble_tracker:457]:   Name: 'SOK-AA01308'
[12:02:26][VV][esp32_ble_tracker:465]:   Ad Flag: 6
[12:02:26][VV][esp32_ble_tracker:468]:   Service UUID: 0xFFF0
[12:02:26][VV][esp32_ble_tracker:489]:   Adv data: 02.01.06.03.03.F0.FF.0C.09.53.4F.4B.2D.41.41.30.31.33.30.38.0C.08.53.4F.4B.2D.41.41.30.31.33.30.38 (33)

I should add that this ESP32 certainly discovers more than 3 BLE devices (even my neighbors Samsung Frame TV shows up)  and picks up data from 4 BLE beacons right now - 2 Mopeka, and 2 Victron. My understanding is that the 3 device limit is only for held connections, but I'm fuzzy on the specifics of that because I thought that all BLE was broadcast, not connected.

Any thoughts appreciated.

patman15, 2025-01-30:
> Hey, confirming that you don't support SOK BMS.

Hey @wmichael3! Correct, if you need support for it, please open a separate [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?template=feature_request.yml) so I can keep track of the different implementations. I need to do a new protocol implementation for this BMS. This ticket is for a Daly BMS battery that is not detected. Thanks!

> My understanding is that the 3 device limit is only for held connections, but I'm fuzzy on the specifics of that because I thought that all BLE was broadcast, not connected.

Correct, the BMS require one held connection per device. Thus, for broadcasting sensors the limit is not relevant, but for BMSs and therefore this integration, it is.

RyuuKiiroi, 2025-01-31:
I have a similar problem.
One BMS JBD SP17S005 connects immediately and is visible in HA.
And BMS Daly Smart active balance (blue), bt name like JHB-, I tried to change to DL-, is visible in the logs VERY_VERBOSE, but no other devices are visible in HA.

[00:54:35][VV][esp32_ble_tracker:434]: Parse Result:
[00:54:35][VV][esp32_ble_tracker:453]:   Address: 50:18:12:01:A4:94 (RANDOM)
[00:54:35][VV][esp32_ble_tracker:456]:   RSSI: -33
[00:54:35][VV][esp32_ble_tracker:457]:   Name: 'DL-01'
[00:54:35][VV][esp32_ble_tracker:465]:   Ad Flag: 6
[00:54:35][VV][esp32_ble_tracker:479]:   Manufacturer ID: 0x0104, data: 01.50.18.12.01.A4.94.4A.48.42 (10)
[00:54:35][VV][esp32_ble_tracker:489]:   Adv data: 02.01.06.0D.FF.04.01.01.50.18.12.01.A4.94.4A.48.42.06.09.44.4C.2D.30.31 (24)

[logs_esp32-bluetooth-proxy-5d34a8_run.txt](https://github.com/user-attachments/files/18625460/logs_esp32-bluetooth-proxy-5d34a8_run.txt)

herijaya, 2025-02-01:
[07:17:43][C][logger:186]:   Level: VERY_VERBOSE
[07:17:43][C][logger:188]:   Log Baud Rate: 115200
[07:17:43][C][bluetooth_proxy:091]: Bluetooth Proxy:
[07:17:43][C][bluetooth_proxy:092]:   Active: YES
[07:17:43][C][bluetooth_proxy:093]:   Connections: 3
[07:17:44][C][bluetooth_proxy:094]:   Raw advertisements: YES
[07:17:44][C][esp32_ble_tracker:663]:   Scan Interval: 320.0 ms
[07:17:44][C][captive_portal:089]: Captive Portal:
[07:17:44][C][mdns:116]: mDNS:
[07:17:44][C][mdns:117]:   Hostname: esp32-ble-bms
[07:17:44][V][mdns:118]:   Services:
[07:17:44][V][mdns:120]:   - _esphomelib, _tcp, 6053
[07:17:44][V][mdns:122]:     TXT: friendly_name = bluetooth proxy esp32-1
[07:17:44][V][mdns:122]:     TXT: version = 2024.12.4
[07:17:44][V][mdns:122]:     TXT: mac = 58bf259e97e0
[07:17:44][V][mdns:122]:     TXT: platform = ESP32
[07:17:44][V][mdns:122]:     TXT: board = esp32dev
[07:17:44][V][mdns:122]:     TXT: network = wifi
[07:17:44][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[07:17:44][C][esphome.ota:073]: Over-The-Air updates:
[07:17:44][C][esphome.ota:074]:   Address: 192.168.45.72:3232
[07:17:44][C][esphome.ota:075]:   Version: 2
[07:17:44][C][esphome.ota:078]:   Password configured
[07:17:44][C][safe_mode:018]: Safe Mode:
[07:17:44][C][safe_mode:020]:   Boot considered successful after 60 seconds
[07:17:44][C][safe_mode:021]:   Invoke after 10 boot attempts
[07:17:44][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[07:17:45][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:17:45][V][bluetooth_proxy:059]: Proxying raw packet from 40:17:10:01:5C:9B, length 33. RSSI: -69 dB
[07:17:45][V][bluetooth_proxy:061]: Proxying 1 packets
[07:17:45][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70467796950171
  rssi: -69
  address_type: 1
  data: '\xff@\\x9bDL	DL-401710015C9B'
}
}
[07:17:46][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:17:46][V][bluetooth_proxy:059]: Proxying raw packet from 40:18:05:01:1A:26, length 33. RSSI: -56 dB
[07:17:46][V][bluetooth_proxy:061]: Proxying 1 packets
[07:17:46][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70471907351078
  rssi: -56
  address_type: 1
  data: '\xff@&DL	DL-401805011A26'
}
}
[07:17:49][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:17:49][V][bluetooth_proxy:059]: Proxying raw packet from 40:18:05:01:1A:26, length 33. RSSI: -63 dB
[07:17:49][V][bluetooth_proxy:061]: Proxying 1 packets
[07:17:49][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70471907351078
  rssi: -63
  address_type: 1
  data: '\xff@&DL	DL-401805011A26'
}
}
[07:17:51][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:17:51][V][bluetooth_proxy:059]: Proxying raw packet from 40:17:10:01:5C:9B, length 33. RSSI: -69 dB
[07:17:51][V][bluetooth_proxy:061]: Proxying 1 packets
[07:17:51][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70467796950171
  rssi: -69
  address_type: 1
  data: '\xff@\\x9bDL	DL-401710015C9B'
}
[... 170 lines cut ...]
  data: '\xff@&DL	DL-401805011A26'
}
}
[07:19:02][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:19:02][V][bluetooth_proxy:059]: Proxying raw packet from 40:18:05:01:1A:26, length 33. RSSI: -64 dB
[07:19:02][V][bluetooth_proxy:061]: Proxying 1 packets
[07:19:02][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70471907351078
  rssi: -64
  address_type: 1
  data: '\xff@&DL	DL-401805011A26'
}
}
[07:19:05][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[07:19:05][V][bluetooth_proxy:059]: Proxying raw packet from 40:17:10:01:5C:9B, length 33. RSSI: -54 dB
[07:19:05][V][bluetooth_proxy:061]: Proxying 1 packets
[07:19:05][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70467796950171
  rssi: -54
  address_type: 1
  data: '\xff@\\x9bDL	DL-401710015C9B'
}
}

herijaya, 2025-02-01:
[13:22:08][C][logger:185]: Logger:
[13:22:08][C][logger:186]:   Level: VERY_VERBOSE
[13:22:08][C][logger:188]:   Log Baud Rate: 115200
[13:22:08][C][logger:189]:   Hardware UART: UART0
[13:22:08][C][bluetooth_proxy:091]: Bluetooth Proxy:
[13:22:08][C][bluetooth_proxy:092]:   Active: YES
[13:22:08][C][bluetooth_proxy:093]:   Connections: 3
[13:22:08][C][bluetooth_proxy:094]:   Raw advertisements: YES
[13:22:08][C][esp32_ble:418]: ESP32 BLE:
[13:22:08][C][esp32_ble:420]:   MAC address: 58:BF:25:9E:97:E2
[13:22:08][C][esp32_ble:421]:   IO Capability: none
[13:22:08][C][esp32_ble_tracker:661]: BLE Tracker:
[13:22:08][C][esp32_ble_tracker:662]:   Scan Duration: 300 s
[13:22:08][C][esp32_ble_tracker:663]:   Scan Interval: 320.0 ms
[13:22:08][C][captive_portal:089]: Captive Portal:
[13:22:08][C][mdns:116]: mDNS:
[13:22:08][C][mdns:117]:   Hostname: esp32-ble-bms
[13:22:08][V][mdns:118]:   Services:
[13:22:08][V][mdns:120]:   - _esphomelib, _tcp, 6053
[13:22:08][V][mdns:122]:     TXT: friendly_name = bluetooth proxy esp32-1
[13:22:08][V][mdns:122]:     TXT: version = 2024.12.4
[13:22:08][V][mdns:122]:     TXT: mac = 58bf259e97e0
[13:22:08][V][mdns:122]:     TXT: platform = ESP32
[13:22:08][V][mdns:122]:     TXT: board = esp32dev
[13:22:08][V][mdns:122]:     TXT: network = wifi
[13:22:08][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[13:22:08][C][esphome.ota:073]: Over-The-Air updates:
[13:22:08][C][esphome.ota:074]:   Address: 192.168.45.72:3232
[13:22:08][C][esphome.ota:075]:   Version: 2
[13:22:08][C][esphome.ota:078]:   Password configured
[13:22:08][C][safe_mode:018]: Safe Mode:
[13:22:08][C][safe_mode:020]:   Boot considered successful after 60 seconds
[13:22:08][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:22:08][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[13:22:08][C][api:140]: API Server:
[13:22:08][C][api:141]:   Address: 192.168.45.72:6053
[13:22:08][C][api:143]:   Using noise encryption: YES
[13:22:12][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:22:12][V][bluetooth_proxy:059]: Proxying raw packet from 7C:BC:80:EB:27:C4, length 17. RSSI: -93 dB
[13:22:12][V][bluetooth_proxy:061]: Proxying 1 packets
[13:22:12][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 137149058590660
  rssi: -93
  address_type: 1
  data: '

\xffL
[13:22:17][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:22:17][V][bluetooth_proxy:059]: Proxying raw packet from 40:17:10:01:5C:9B, length 33. RSSI: -59 dB
[13:22:17][V][bluetooth_proxy:061]: Proxying 1 packets
[13:22:17][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70467796950171
  rssi: -59
  address_type: 1
  data: '\xff@\\x9bDL	DL-401710015C9B'
}
}
[13:22:27][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:22:27][V][bluetooth_proxy:059]: Proxying raw packet from 40:18:05:01:1A:26, length 33. RSSI: -59 dB
[13:22:27][V][bluetooth_proxy:061]: Proxying 1 packets
[13:22:27][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70471907351078
  rssi: -59
  address_type: 1
  data: '\xff@&DL	DL-401805011A26'
}
}
[... 262 lines cut ...]
  data: '\xff@&DL	DL-401805011A26'
}
}
[13:24:04][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:24:04][V][bluetooth_proxy:059]: Proxying raw packet from 40:17:10:01:5C:9B, length 33. RSSI: -59 dB
[13:24:04][V][bluetooth_proxy:061]: Proxying 1 packets
[13:24:04][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 70467796950171
  rssi: -59
  address_type: 1
  data: '\xff@\\x9bDL	DL-401710015C9B'
}
}
[13:24:08][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:24:08][V][bluetooth_proxy:059]: Proxying raw packet from 7C:BC:80:EB:27:C4, length 17. RSSI: -93 dB
[13:24:08][V][bluetooth_proxy:061]: Proxying 1 packets
[13:24:08][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 137149058590660
  rssi: -93
  address_type: 1
  data: '

\xffL

patman15, 2025-02-01:
@herijaya can you please not spam the issue, but attach the logs as file? I would need
> data: '���\xff���@���\x9bDL� DL-401710015C9B'
in a (machine) readable form and that does not work by copy/paste as any binary data is replaced with �.

patman15, 2025-02-01:
@RyuuKiiroi 
> I have a similar problem. One BMS JBD SP17S005 connects immediately and is visible in HA. And BMS Daly Smart active balance (blue), bt name like JHB-, I tried to change to DL-, is visible in the logs VERY_VERBOSE, but no other devices are visible in HA.
> 
Sorry, v1.11.4 was not able to detect _Daly Smart active balance (blue)_ BMS (the README unfortunately always shows the latest code on main branch, while the release might be a bit behind). 
Please rename your device to `JHB-` and try the v1.11.5 version.

herijaya, 2025-02-01:
oh yeah sorry
here are the logs:

[logs_esp32-ble-bms_logs.txt](https://github.com/user-attachments/files/18628042/logs_esp32-ble-bms_logs.txt)

patman15, 2025-02-01:
@herijaya no worries, thanks for the cleanup, will check the logs. 👍

RyuuKiiroi, 2025-02-01:
> [@RyuuKiiroi](https://github.com/RyuuKiiroi)
>
>
> Sorry, v1.11.4 was not able to detect _Daly Smart active balance (blue)_ BMS (the README unfortunately always shows the latest code on main branch, [...]

Updated to the latest version and changed the name.

[logs_esp32-bluetooth-proxy-5d34a8_run (1).txt](https://github.com/user-attachments/files/18628263/logs_esp32-bluetooth-proxy-5d34a8_run.1.txt)

patman15, 2025-02-01:
@herijaya, @RyuuKiiroi can you both please enable debug logging of the integration and try the branch linked to this ticket by [manually installing](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) it? Seems that the detection pattern did not fit your devices.

RyuuKiiroi, 2025-02-01:
> [@herijaya](https://github.com/herijaya), [@RyuuKiiroi](https://github.com/RyuuKiiroi) can you both please enable debug logging of the integration and try the branch linked to this ticket by [...]

It worked ))

[image]

[home-assistant_bms_ble_2025-02-01T11-26-02.945Z.log](https://github.com/user-attachments/files/18628554/home-assistant_bms_ble_2025-02-01T11-26-02.945Z.log)

patman15, 2025-02-01:
@RyuuKiiroi you saw that your name change was not full successful? :wink: Target was JHB-...

RyuuKiiroi, 2025-02-01:
> [@RyuuKiiroi](https://github.com/RyuuKiiroi) you saw that your name change was not fully successful? 😉 Target was JHB-...

Yes, I saw it, but the name Bluetooth is correct.

[15:01:25][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[15:01:25][V][bluetooth_proxy:058]: Proxying raw packet from 50:18:12:01:A4:94, length 35. RSSI: -34 dB
[15:01:25][V][bluetooth_proxy:061]: Proxying 1 packets
[15:01:25][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 88064311534740
  rssi: -34
  address_type: 1
\xffP\xa4\x94JHB	JHB-50181201A494'
}
}
[15:01:25][VV][api.service:971]: on_bluetooth_gatt_write_request: BluetoothGATTWriteRequest {

patman15, 2025-02-01:
> Yes, I saw it, but the name Bluetooth is correct.

Hmm, probably a caching effect.
Could you do me a favor and look in the home-assistant logs on your side for a line containing `DEBUG (MainThread) [custom_components.bms_ble]`... and `AdvertisementData(local_name='`... ? I'd like to check what HA sees of the device to finally fix the automatic detection.

RyuuKiiroi, 2025-02-01:
That's all it shows

[image]

[image]

patman15, 2025-02-01:
@RyuuKiiroi you need to [enable debug logging](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and if you want to view them like this, enable `show raw logs` behind the 3 dots menu.

Note: You are aware that running the integration and Batmon in parallel, can, will, does cause issues?

herijaya, 2025-02-01:
@patman15 debug logs:

[logs.txt](https://github.com/user-attachments/files/18629042/logs.txt)

RyuuKiiroi, 2025-02-01:
> [@RyuuKiiroi](https://github.com/RyuuKiiroi) you need to [enable debug logging](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and if you want to view them like this, enable `show raw logs` behind the 3 dots menu.

2025-02-01 14:25:48.434 DEBUG (MainThread) [custom_components.bms_ble] DL-JHB-50181201A494: BMS data sample {'temp#0': 14.0, 'voltage': 52.5, 'current': 0.0, 'battery_level': 88.0, 'cycle_charge': 92.4, 'cell_count': 16, 'temp_sensors': 2, 'cycles': 0, 'delta_voltage': 0.002, 'temp#1': 14.0, 'temp#2': 14.0, 'cell#0': 3.283, 'cell#1': 3.284, 'cell#2': 3.284, 'cell#3': 3.283, 'cell#4': 3.284, 'cell#5': 3.283, 'cell#6': 3.284, 'cell#7': 3.284, 'cell#8': 3.284, 'cell#9': 3.284, 'cell#10': 3.283, 'cell#11': 3.282, 'cell#12': 3.282, 'cell#13': 3.283, 'cell#14': 3.282, 'cell#15': 3.284, 'cycle_capacity': 4851.0, 'power': 0.0, 'battery_charging': False, 'temperature': 14.0}

2025-02-01 14:25:55.382 DEBUG (MainThread) [custom_components.bms_ble] SP17S005P17S120A: BMS data sample {'temp_sensors': 3, 'voltage': 52.49, 'current': 0.0, 'battery_level': 10, 'cycle_charge': 12.27, 'cycles': 0, 'temp#0': 16.7, 'temp#1': 13.2, 'temp#2': 15.3, 'cell#0': 3.285, 'cell#1': 3.285, 'cell#2': 3.284, 'cell#3': 3.278, 'cell#4': 3.277, 'cell#5': 3.284, 'cell#6': 3.287, 'cell#7': 3.279, 'cell#8': 3.28, 'cell#9': 3.278, 'cell#10': 3.278, 'cell#11': 3.28, 'cell#12': 3.279, 'cell#13': 3.281, 'cell#14': 3.278, 'cell#15': 3.28, 'delta_voltage': 0.01, 'cycle_capacity': 644.052, 'power': 0.0, 'battery_charging': False, 'temperature': 15.067}

Where to look for AdvertisementData(local_name=' ?
There is no such line in batmon debug and esphome log.

> Note: You are aware that running the integration and Batmon in parallel, can, will, does cause issues?

What integration?
I didn't quite understand )

patman15, 2025-02-01:
> There is no such line in batmon debug and esphome log.

You might need to restart the integration/HA and check again.

>>  Note: You are aware that running the integration and Batmon in parallel, can, will, does cause issues?

> What integration?
> I didn't quite understand )

This, BMS_BLE-HA integration providing you data. From the logs it looks like you have running `batmon` in parallel. Not a good idea, as BMSs usually only allow one connection at a time, so you might run into conflicts.

RyuuKiiroi, 2025-02-01:
Sorry.
I just mixed up the integrations.
But batmon is disabled, I tried to connect via uart, and via proxy it probably doesn't support connection.

[image]

[image]

patman15, 2025-02-02:
@herijaya, @RyuuKiiroi unfortunately, the logs you provided so far did not contain the information I need. Could you please follow the [guidelines](https://github.com/patman15/BMS_BLE-HA/blob/main/README.md#in-case-you-have-troubles-youd-like-to-have-help-with), i.e.
- please [enable the debug protocol](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) for the [BLE Battery Management integration](https://my.home-assistant.io/redirect/integration/?domain=bms_ble),
- restart Home Assistant, wait till it is fully started up,
- disable the log (Home Assistant will prompt you to download the log), 

and attach it here? To give you an idea, why I'm so picky: I need a verification of your setup so that I can add tests, to ensure that I do not mess up the setup of other people and keep yours intact for future changes. Thanks for your support!

RyuuKiiroi, 2025-02-02:
2025-02-02 12:08:51.476 DEBUG (MainThread) [custom_components.bms_ble] JHB-50181201A494: advertisement: {'name': 'JHB-50181201A494', 'address': '50:18:12:01:A4:94', 'rssi': -36, 'manufacturer_data': {260: b'\x01P\x18\x12\x01\xa4\x94JHB'}, 'service_data': {}, 'service_uuids': [], 'source': 'B8:D6:1A:5D:34:A8', 'advertisement': AdvertisementData(local_name='JHB-50181201A494', manufacturer_data={260: b'\x01P\x18\x12\x01\xa4\x94JHB'}, tx_power=-127, rssi=-36), 'device': BLEDevice(50:18:12:01:A4:94, JHB-50181201A494), 'connectable': True, 'time': 59143.799713134766, 'tx_power': -127}

2025-02-02 12:09:42.985 DEBUG (MainThread) [custom_components.bms_ble] SP17S005P17S120A: advertisement: {'name': 'SP17S005P17S120A', 'address': 'A5:C2:37:34:85:42', 'rssi': -31, 'manufacturer_data': {34114: b'47\xc2\xa5'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'B8:D6:1A:5D:34:A8', 'advertisement': AdvertisementData(local_name='SP17S005P17S120A', manufacturer_data={34114: b'47\xc2\xa5'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-31), 'device': BLEDevice(A5:C2:37:34:85:42, SP17S005P17S120A), 'connectable': True, 'time': 59217.015641174, 'tx_power': None}

closed 2025-02-02 by patman15