patman15/BMS_BLE-HA #162: Negative Temperature shown as 65.535°C on Redodo

Stiles96, 2025-02-01:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Redodo

### Describe the issue

Temperature above 0°C are shown ok in HA. Is the temperature below 0°C, I get only 65.535,0°C as current temperature.

[image]

### Reproduction steps

```text
1. Wait until temperature is below 0°C
2. Check temperature in HA.
```

### Debug logs

No logs to find.

patman15, 2025-02-01:
Hi! If you could get me a debug log to setup a regression test that would be great. I can fix it tomorrow I guess.

patman15, 2025-02-02:
@Stiles96 Could you please try the linked branch in this ticket via [manual install](https://github.com/patman15/BMS_BLE-HA#manual) and let me know if it solves the issue (debug log still appreciated). Thanks!

closed 2025-02-02 by patman15