patman15/BMS_BLE-HA #163: Add NEEY active balancer support

NAP141, 2025-02-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

NEEY is a very popular Active Balancer. In some forums it states that it is equivalent to Heltec. Android Store app is called NEEY 

### Additional context

Some work has been done at ESPHome level here: https://github.com/syssi/esphome-jk-bms 
in specifics here: https://github.com/syssi/esphome-jk-bms/tree/main/components/heltec_balancer_ble 
However this did not work for me when replacing my esp32 ble-proxy with it. 

Service explorer: 
[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
   [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read), Value: bytearray(b'EK-24S4EB')
   [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: bytearray(b'\x00\x00')
   [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters (read), Value: bytearray(b'\x10\x00\x10\x00\x00\x00,\x01')
[Service] 0000ffe0-0000-1000-8000-00805f9b34fb (Handle: 8): Vendor specific
  [Characteristic] 0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 9): Vendor specific (write-without-response,write,notify), Max write w/o rsp size: 514
     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 11): Client Characteristic Configuration, Value: bytearray(b'\x01\x00')
   [Characteristic] 0000ffe2-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific (write-without-response,write,notify), Max write w/o rsp size: 514
     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 14): Client Characteristic Configuration, Value: bytearray(b'\x01\x00')
   [Characteristic] 0000ffe3-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (write-without-response,write), Max write w/o rsp size: 514

A youtube review here: 
https://www.youtube.com/watch?v=jpG3siEzll8 
note that in this review Device Name starts with "GW"

patman15, 2025-02-02:
Hi! Thanks for the request. I guess this can be done, but I need to work through my queue of request before I can have a look. I'll come back to this issue as soon as I can spare some time.

NAP141, 2025-02-02:
Thank you! There is an ongoing issue request with a lot more info here: 
https://github.com/syssi/esphome-jk-bms/issues/640

patman15, 2025-02-16:
@NAP141 just had a deeper look and I do not think the device qualifies as "BMS". There is no current sensor or SoC as far as I can see, if that is correct, I need to reject this request unfortunately, in order to keep my efforts focused. From the linked ticket I can see that you already succeeded integrating the device to HA, so I hope you are good?

NAP141, 2025-02-16:
You are correct, this is an active balancer and it does not manage the current flow. Heltec makes also full BMS and I am guessing the protocol would be the same. Balancing cells is still technically a "Battery Management System" :-). 
It is up to you, your Daly implementation works for Daly Balancers. 
I have now a solution for this device through this https://github.com/syssi/esphome-jk-bms so I am good.

closed 2025-02-16 by NAP141