patman15/BMS_BLE-HA #168: Add SOK BMS support

patman15, 2025-02-02:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

@wmichael3 I raised this new issue in case you are still interested (or anyone else) please comment. I will close the issue with the original post in https://github.com/patman15/BMS_BLE-HA/issues/160#issuecomment-2625393278 
```
[12:02:26][VV][esp32_ble_tracker:453]: Address: D6:6C:0A:62:9B:D6 (PUBLIC)
[12:02:26][VV][esp32_ble_tracker:456]: RSSI: -94
[12:02:26][VV][esp32_ble_tracker:457]: Name: 'SOK-AA01308'
[12:02:26][VV][esp32_ble_tracker:465]: Ad Flag: 6
[12:02:26][VV][esp32_ble_tracker:468]: Service UUID: 0xFFF0
[12:02:26][VV][esp32_ble_tracker:489]: Adv data: 02.01.06.03.03.F0.FF.0C.09.53.4F.4B.2D.41.41.30.31.33.30.38.0C.08.53.4F.4B.2D.41.41.30.31.33.30.38 (33)
```

### Additional context

https://github.com/fl4p/batmon-ha/blob/master/bmslib/models/sok.py

wmichael3, 2025-02-02:
> issue with the original post in #160 (comment)

Thank you. I'm still interested, but did not feel this was justified for my
need alone and was waiting to see if anyone else chimed in.

patman15, 2025-02-17:
@rmkilc thanks for expressing your interest! I will soon take a look at that BMS. JFYI, I needed to remove the file from your comment because I do not want to run into legal issues.

patman15, 2025-02-22:
Hey @rmkilc, @wmichael3! Could someone please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery. That would help a lot and provides me test data to verify the implementation. Thanks!

rmkilc, 2025-02-22:
> Hey [@rmkilc](https://github.com/rmkilc), [@wmichael3](https://github.com/wmichael3)! Could someone please see if you are able to capture the Bluetooth traffic on [...]

I've got a btsnoop_hci.log file off my phone that you can view in Wireshark. I'd rather not post it here publicly though as I don't know if there is any identifying information. Anyway I can send it directly to you?

patman15, 2025-02-22:
Ah great! You can send me a PM on the HA community or via Discord. Links are in my profile. Thanks!

patman15, 2025-02-22:
@rmkilc there seem to be 2 different BMS types. Which app are you using / what type of battery is it?

rmkilc, 2025-02-22:
The app is ABC - BMS on Android and the battery I'm using is [this one](https://www.us.sokbattery.com/product-page/sokess-24v-100ah-lifepo4-battery).

rmkilc, 2025-02-23:
[Here](https://github.com/user-attachments/files/18933011/home-assistant_bms_ble_2025-02-23T19-54-45.466Z.log) is the debug log. It does find the SOK BMS, so that's a start! But it fails to initialize.

patman15, 2025-02-24:
Sorry that was a typo, please update and try again.

patman15, 2025-02-24:
Ok, just to be sure the real value of the current is 1/10 of the shown one meaning? Real lower than shown. Then all the other issues are subsequent calculation issues.
Cell voltages I can add hopefully with your log.
Thanks.

rmkilc, 2025-02-24:
Yes, when it says 30 amps, it is really only 3 amps.

patman15, 2025-02-25:
@rmkilc next version ready to be tested. Please provide a log again and let me know your thoughts!

rmkilc, 2025-02-26:
It connects briefly, and seems to show the new sensors correctly (including the individual cell voltages!), but then disconnects. At that point, it shows failed setup.

I can only get it to connect again briefly if I delete the device, reboot Home Assistant, and add it back.

Here are a couple logs: [log 1](https://github.com/user-attachments/files/18974500/home-assistant_bms_ble_2025-02-25T20-26-16.902Z.log) [log 2](https://github.com/user-attachments/files/18974659/home-assistant_bms_ble_2025-02-26T00-25-05.706Z.log)

patman15, 2025-02-26:
Ok, thanks. Seems the BMS is very pick on request timing and the protocol design is really bad ...
Will work around that and provide another update.

rmkilc, 2025-02-26:
Sounds good. I am out of town for a week so I can't do any testing until I'm back. Thanks for your work on this!

patman15, 2025-02-26:
> Thanks for your work on this!

You're welcome, most likely I will then merge it for a release on the weekend.

> By the way, it looks like it was connecting by itself without a reboot occasionally.

Somehow I assumed so, but now it should be much better.

rmkilc, 2025-02-27:
I am still getting the same result. I'm doing it all from my phone so I'm not sure if I'm missing something. I copy and pasted the updated contents of 'abc_bms.py' using the file editor and rebooted HA. [Here](https://github.com/user-attachments/files/19003564/error.log) is a log.

patman15, 2025-02-27:
Ok, that was me late in the evening not paying enough attention. Sorry for that. Fixed. Hope it works now. Test coverage on that BMS model is still low so it did not catch this issue.

wmichael3, 2025-02-27:
Hi - found time to try this with my SOK battery. It is discovered, but fails to add. Could be the connection limit problem, as the error you display in the UI is:

> Failed setup, will retry: device communicating failed: D6:6C:0A:62:9B:D6 - D6:6C:0A:62:9B:D6: Failed to connect after 10 attempt(s): No backend with an available connection slot that can reach address D6:6C:0A:62:9B:D6 was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device (BleakOutOfConnectionSlotsError)

The other BLE devices processed by this ESP are a Victron SmartShunt, Victron SmartSolar, and two Mopeka Propane sensors - I had thought they were all broadcast BLE that did not take connection slots (since there are 4 devices, 2 Victron, 2 Mopeka), but certainly don't know that for certain and not sure how I can tell. There is only the one SOK BMS. (this ESP is placed near a travel trailer that has these devices inside)

Can you confirm from the attached log that this is indeed a connection limit issue, or instead something else?

Thanks so much for working on this.

[SOK BLE error.log](https://github.com/user-attachments/files/19013840/SOK.BLE.error.log)

patman15, 2025-02-27:
Hey @wmichael3 , thanks for reporting, good to have another tester for this type of BMS! 😃 
First log was definitely some sort of Bluetooth issue. In the second one I can see that the BMS simply ignores the command. Strange. I can see that your SOK BMS uses a different naming scheme than the one of rmklic. Maybe the two batteries use a different BMS? Need to research a bit.

wmichael3, 2025-02-27:
If it helps at all, here is what the BMS app shows.  I've never not been able to connect the app from the same locations as the ESPs, but mostly I use the SmartShunt over the BMS app for almost everything and its BLE has been perfect.  But the shunt doesn't reflect the internal battery heater power consumption and that bit me once over a cold snap - which is why I was interested in this. Thanks again.

[image]

patman15, 2025-02-27:
@wmichael3 as mentioned, your signal strength is really bad. Have a look at the Bluetooth monitor in HA and try to fix the reception issue first. `RSSI` shall be `> -70dBm`. Then I can have a look at a new debug log, what is happening.

wmichael3, 2025-02-27:
RSSI of -62, maybe 18 inches from the battery to an ESP, no change that I can see.

```
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JN45M87D6215SPAWK8MA67V3 version=1 domain=bms_ble title=SOK-AA01308 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=D6:6C:0A:62:9B:D6>
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: initializing Chunguang Song ABC BMS, BT address: D6:6C:0A:62:9B:D6
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for SOK-AA01308 (D6:6C:0A:62:9B:D6) as Chunguang Song ABC BMS
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: advertisement: {'name': 'SOK-AA01308', 'address': 'D6:6C:0A:62:9B:D6', 'rssi': -62, 'manufacturer_data': {}, 'service_data': {'00001827-0000-1000-8000-00805f9b34fb': b'\xa8\x01Q\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'service_uuids': ['0000fff0-0000-1000-8000-00805f9b34fb', '00001827-0000-1000-8000-00805f9b34fb'], 'source': 'DC:54:75:CE:B3:D8', 'advertisement': AdvertisementData(local_name='SOK-AA01308', service_data={'00001827-0000-1000-8000-00805f9b34fb': b'\xa8\x01Q\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fff0-0000-1000-8000-00805f9b34fb', '00001827-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-62), 'device': BLEDevice(D6:6C:0A:62:9B:D6, SOK-AA01308), 'connectable': True, 'time': 2686180.710009606, 'tx_power': None}
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: BMS data update
2025-02-27 13:35:55.652 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: connecting BMS
2025-02-27 13:35:55.652 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:62:9B:D6 - D6:6C:0A:62:9B:D6: Connection attempt: 1
2025-02-27 13:35:56.489 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:62:9B:D6 - D6:6C:0A:62:9B:D6: Connected after 1 attempts
2025-02-27 13:35:56.910 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: TX BLE data: ee c4 00 00 00 4f
2025-02-27 13:36:06.911 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: device communication timed out
2025-02-27 13:36:06.911 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching SOK-AA01308 data in 11.259 seconds (success: False)
2025-02-27 13:36:06.911 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (SOK-AA01308)
2025-02-27 13:36:06.911 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: disconnecting BMS
2025-02-27 13:36:07.248 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: disconnected from BMS

```

rmkilc, 2025-02-27:
I installed the latest, and so far, it's not disconnecting!

patman15, 2025-02-27:
@wmichael3! Ok, then I would need you to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery. Seems like the current code works for rmklic but not for you. Can you get me such a trace?

wmichael3, 2025-02-27:
Ok, if I understand this correctly, this is what you're looking for?

[btsnoop_hci.log](https://github.com/user-attachments/files/19016264/btsnoop_hci.log)

rmkilc, 2025-02-27:
It all seems to still be working. I can remotely trigger my battery to discharge and charge so I got that going for testing so the data is constantly changing. It's the battery my UPS for my networking gear runs on. My battery does not have a heater like wmichael3's has.

For a few minutes the rssi sensor was unavailable but the rest were all working. Then it came back. I have debugging enabled and can upload a log later if need be.

[image]

patman15, 2025-02-27:
Thanks @rmkilc . Looks like I need to find a way to read both types of batteries. If @wmichael3 confirms that the change helps his setup, I need to combine them and then would need to ask both of you to test again. Unfortunately, without a proper protocol description it's a bit trial and error as any change could break one of the two types.
Thanks!

wmichael3, 2025-02-27:
Yes, my battery is a 12V/100AH with self-heating. Bought about 2 years ago.

Still fails, but more going on here.

[SOK BLE error update 3.log](https://github.com/user-attachments/files/19017582/SOK.BLE.error.update.3.log)

patman15, 2025-02-28:
@wmichael3, @rmkilc I just pushed an update that should make the setup work for both of you. 🤞 
Let me know, preferable with a debug log for final verification.

wmichael3, 2025-02-28:
It worked, and entities and devices were created!

Later today I'll exercise the battery a bit and look at the data, but at first glance things look good. Below are the pertinent lines from the log.

```
2025-02-28 09:58:26.870 DEBUG (MainThread) [custom_components.bms_ble] Unloaded config entry: D6:6C:0A:62:9B:D6, ok? True!
2025-02-28 09:58:26.870 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (SOK-AA01308)
2025-02-28 09:58:26.871 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: disconnecting BMS
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: disconnected from BMS
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JN45M87D6215SPAWK8MA67V3 version=1 domain=bms_ble title=SOK-AA01308 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=D6:6C:0A:62:9B:D6>
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: initializing Chunguang Song ABC BMS, BT address: D6:6C:0A:62:9B:D6
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for SOK-AA01308 (D6:6C:0A:62:9B:D6) as Chunguang Song ABC BMS
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: advertisement: {'name': 'SOK-AA01308', 'address': 'D6:6C:0A:62:9B:D6', 'rssi': -71, 'manufacturer_data': {}, 'service_data': {'00001827-0000-1000-8000-00805f9b34fb': b'\xa8\x01Q\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'service_uuids': ['00001827-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], 'source': 'DC:54:75:CE:B3:D8', 'advertisement': AdvertisementData(local_name='SOK-AA01308', service_data={'00001827-0000-1000-8000-00805f9b34fb': b'\xa8\x01Q\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['00001827-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-71), 'device': BLEDevice(D6:6C:0A:62:9B:D6, SOK-AA01308), 'connectable': True, 'time': 2759336.003635168, 'tx_power': -127}
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: BMS data update
2025-02-28 09:58:27.176 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: connecting BMS
2025-02-28 09:58:27.177 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:62:9B:D6 - D6:6C:0A:62:9B:D6: Connection attempt: 1
2025-02-28 09:58:27.577 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:62:9B:D6 - D6:6C:0A:62:9B:D6: Connected after 1 attempts
2025-02-28 09:58:27.605 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: TX BLE data: ee c4 00 00 00 4f
2025-02-28 09:58:28.606 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: TX BLE data: ee c2 00 00 00 46
2025-02-28 09:58:28.807 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: RX BLE data: bytearray(b'\xcc\xf0\xa65\x00\x00\x00\x00\xa0\x86\x01\xc9U\x01x\x00d\x00\x00\xd7')
2025-02-28 09:58:28.808 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: RX BLE data: bytearray(b'\xcc\xf3\x15\n\x1e\x002\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2025-02-28 09:58:28.909 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: RX BLE data: bytearray(b'\xcc\xf4\x01\x12\r\x00\x02\x07\r\x00\x03$\r\x00\x04Y\x0e\x00\x00\x07')
2025-02-28 09:58:28.909 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: TX BLE data: ee c1 00 00 00 ce
2025-02-28 09:58:29.012 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: RX BLE data: bytearray(b'\xcc\xf0\xa45\x00\x00\x00\x00\xa0\x86\x01\xc9U\x01x\x00d\x00\x00\x89')
2025-02-28 09:58:29.114 DEBUG (MainThread) [custom_components.bms_ble.plugins.abc_bms] SOK-AA01308[9B:D6]: RX BLE data: bytearray(b'\xcc\xf2\x00\x01\x01\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x94')
2025-02-28 09:58:29.114 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA01308: BMS data sample {'temp_sensors': 1, 'voltage': 13.732, 'current': 0.0, 'battery_level': 100, 'cycle_charge': 87.497, 'cycles': 120, 'cell#0': 3.346, 'cell#1': 3.335, 'cell#2': 3.364, 'cell#3': 3.673, 'temp#0': 6, 'delta_voltage': 0.338, 'cycle_capacity': 1201.509, 'power': 0.0, 'battery_charging': False, 'temperature': 6.0, 'problem': False}
2025-02-28 09:58:29.114 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching SOK-AA01308 data in 1.938 seconds (success: True)

```

closed 2025-03-02 by patman15

rmkilc, 2025-03-06:
Just for completeness, here is my [debug log](https://github.com/user-attachments/files/19098863/home-assistant_bms_ble_2025-03-04T22-14-32.298Z.log). So far it's working great!