patman15/BMS_BLE-HA #169: No Supported Devices Found - Ecoworthy 12V 150AH (metal case version)

Scope666, 2025-02-03:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi, I just setup a bluetooth proxy using the ESPHome "Ready-Made Projects" page.  It appears to be proxying requests, and I temp installed the Bermuda BLE integration to verify I can "see" the BMS chip.  My HA setup is HA Core running in Docker on an Unraid server.  I've tried some of the bluetoothctl commands I've seen you list but they seem to hang, not sure if they works with a setup like mine.   (docker + proxy)

Here's some extra info:

```
            "name": "DP04S007L4S120A",
            "address": "A5:C2:37:2E:A5:C3",
            "rssi": -49,
            "advertisement_data": [
              "DP04S007L4S120A",
              {
                "42435": {
                  "__type": "<class 'bytes'>",
                  "repr": "b'.7\\xc2\\xa5'"
                }
              },
              {},
              [
                "0000ff00-0000-1000-8000-00805f9b34fb"
              ],
              -127,
              -49,
              []
            ],
            "details": {
              "source": "3C:8A:1F:81:55:18",
              "address_type": 0
            }
          },
```

[image]

[image]

Scope666, 2025-02-03:
Not sure how I managed to do this, but digging around in other issues, I saw there's an alternative branch for an ECO-WORTHY 100 AH with an open issue.  I changed the 5 digit vendor ID to match what I see in the ESPHome Diagnostics and now I seem to be getting info from it:

[image]

Scope666, 2025-02-03:
manifest.json edit:

[image]

jbd_bms.py edit:

[image]

patman15, 2025-02-03:
Hi! Thanks for the info. I start suspecting that ECO-WORTHY uses the MAC address for the vendor id, which is really stu...
Need to check and adapt the detection. If you could provide a debug log that would be great. I would need one including a restart of the integration.

Scope666, 2025-02-03:
Let me know if this works for you, if not, happy to dig out anything you need:

```
2025-02-03 08:00:06.097 DEBUG (MainThread) [custom_components.bms_ble] Unloaded config entry: A5:C2:37:2E:A5:C3, ok? True!
2025-02-03 08:00:06.097 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (DP04S007L4S120A)
2025-02-03 08:00:06.097 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: disconnecting BMS
2025-02-03 08:00:06.177 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: disconnected from BMS
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JK5182RMV8RH4MTSNC8YXMBT version=1 domain=bms_ble title=DP04S007L4S120A state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=A5:C2:37:2E:A5:C3>
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: initializing Jiabaida Smart BMS, BT address: A5:C2:37:2E:A5:C3
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for DP04S007L4S120A (A5:C2:37:2E:A5:C3) as Jiabaida Smart BMS
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S120A: advertisement: {'name': 'DP04S007L4S120A', 'address': 'A5:C2:37:2E:A5:C3', 'rssi': -49, 'manufacturer_data': {42435: b'.7\xc2\xa5'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': '3C:8A:1F:81:55:18', 'advertisement': AdvertisementData(local_name='DP04S007L4S120A', manufacturer_data={42435: b'.7\xc2\xa5'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-49), 'device': BLEDevice(A5:C2:37:2E:A5:C3, DP04S007L4S120A), 'connectable': True, 'time': 1644011.069238759, 'tx_power': None}
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S120A: BMS data update
2025-02-03 08:00:06.178 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: connecting BMS
2025-02-03 08:00:06.178 DEBUG (MainThread) [bleak_retry_connector] A5:C2:37:2E:A5:C3 - A5:C2:37:2E:A5:C3: Connection attempt: 1
2025-02-03 08:00:08.324 DEBUG (MainThread) [bleak_retry_connector] A5:C2:37:2E:A5:C3 - A5:C2:37:2E:A5:C3: Connected after 1 attempts
2025-02-03 08:00:08.342 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: TX BLE data: dd a5 03 00 ff fd 77
2025-02-03 08:00:08.529 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (start): bytearray(b'\xdd\x03\x00"\x054\x00\x0034@t\x00\x001\x1a\x00\x00\x00\x00')
2025-02-03 08:00:08.530 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (cnt.): bytearray(b'\x00\x00fO\x03\x04\x01\x0b<\x00\x00\x00@t34\x00\x00\xfc ')
2025-02-03 08:00:08.531 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (cnt.): bytearray(b'w')
2025-02-03 08:00:08.532 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: TX BLE data: dd a5 04 00 ff fc 77
2025-02-03 08:00:08.659 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (start): bytearray(b'\xdd\x04\x00\x08\r\x03\r\x04\r\x03\r\x03\xff\xb7w')
2025-02-03 08:00:08.659 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S120A: BMS data sample {'temp_sensors': 1, 'voltage': 13.32, 'current': 0.0, 'battery_level': 79, 'cycle_charge': 131.08, 'cycles': 0, 'temp#0': 14.5, 'cell#0': 3.331, 'cell#1': 3.332, 'cell#2': 3.331, 'cell#3': 3.331, 'delta_voltage': 0.001, 'cycle_capacity': 1745.986, 'power': 0.0, 'battery_charging': False, 'temperature': 14.5}
2025-02-03 08:00:08.659 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DP04S007L4S120A data in 2.481 seconds (success: True)
2025-02-03 08:00:08.659 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up bms_ble.binary_sensor
2025-02-03 08:00:08.660 INFO (MainThread) [homeassistant.components.sensor] Setting up bms_ble.sensor
2025-02-03 08:00:38.988 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S120A: BMS data update
2025-02-03 08:00:38.988 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: BMS already connected
2025-02-03 08:00:38.988 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: TX BLE data: dd a5 03 00 ff fd 77
2025-02-03 08:00:39.140 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (start): bytearray(b'\xdd\x03\x00"\x054\x00\x0034@t\x00\x001\x1a\x00\x00\x00\x00')
2025-02-03 08:00:39.143 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (cnt.): bytearray(b'\x00\x00fO\x03\x04\x01\x0b<\x00\x00\x00@t34\x00\x00\xfc ')
2025-02-03 08:00:39.147 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (cnt.): bytearray(b'w')
2025-02-03 08:00:39.147 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: TX BLE data: dd a5 04 00 ff fc 77
2025-02-03 08:00:39.241 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] DP04S007L4S120A[A5:C3]: RX BLE data (start): bytearray(b'\xdd\x04\x00\x08\r\x04\r\x05\r\x03\r\x04\xff\xb4w')
[... cut ...]
2025-02-03 08:01:09.251 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S120A: BMS data sample {'temp_sensors': 1, 'voltage': 13.32, 'current': 0.0, 'battery_level': 79, 'cycle_charge': 131.08, 'cycles': 0, 'temp#0': 14.5, 'cell#0': 3.332, 'cell#1': 3.333, 'cell#2': 3.331, 'cell#3': 3.331, 'delta_voltage': 0.002, 'cycle_capacity': 1745.986, 'power': 0.0, 'battery_charging': False, 'temperature': 14.5}
2025-02-03 08:01:09.251 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DP04S007L4S120A data in 0.263 seconds (success: True)
```

Scope666, 2025-02-03:
I think somebody needs to show the BMS manufacturer this, it's the beginning part that they should be using, certainly not the end, that's unique per device.

[image]

patman15, 2025-02-03:
Or they could read chapter 1.4 of https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/CSS_v11/out/en/supplement-to-the-bluetooth-core-specification/data-types-specification.html :wink:

patman15, 2025-02-08:
@Scope666 could you please try the linked branch via manual install and report back? Best case with a debug log? Thanks!

closed 2025-02-09 by patman15