patman15/BMS_BLE-HA #171: Add the ability to read/get information about a stacked Seplos v2 battery system

bogi9999, 2025-02-03:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

This is a request to read several Seplos Mason 280 battery packs via Bluetooth.
The first pack (master) has Bluetooth, the others are connected to the master via RS485.
In a Setup like that, the master combines the Value SOC of each Pack to a sum of all and send this information to an Energy Storage System Controler (eg. Victron Cerbo) via CAN Bus. The Master knows all values of each Pack and each Cell. You cann access thes information through an bluetooth connection to the master (BP00), and switch then to the next Pack in the Bluetooth App (BP01/02/03.....) 
The benefit of this reuquest whould be, to have information of the whole setup for health monitoring of that system. 
In my case, i have a setup of five battery-packs available for testing. I am not a developer, but I will contribute to the implementation with a few helpful hints.

patman15, 2025-02-03:
Hi! Thanks for raising the issue. There was a long discussion in #101 about the sub-packs and how to query them. Seems that multiple possible setups are possible.
Anyway, could you please attach a debug log of your setup? The result information should already contain the information about your 5 packs. I can then add further queries to add Infos like I did for the V3 as well.

bogi9999, 2025-02-12:
Hello Patrick,
I had already written my assumption in the Home Assistant Community.
Here it is again. I already wrote that the data is disappearing. The connection quality and signal strength should be fine.
The problem with the disappearance of the data seems to be the retrieval function of the additional batteries.
There is probably a round trip in BP01, then BP02, then BP03 etc.
When I use the RS485 interface and the Seplos Battery Monitor to poll the packs via a multiple poll, this is exactly how it is handled.
This is probably how the query of the other packs is handled.
Is the debug log enough, I’ve attached to this mail?

Bye, Leigh

Antworten an: patman15/BMS_BLE-HA ***@***.***>
Datum: Montag, 3. Februar 2025 um 16:16
An: patman15/BMS_BLE-HA ***@***.***>
Cc: bogi9999 ***@***.***>, Author ***@***.***>

patman15, 2025-02-13:
@bogi9999 the attachment did not go through via e-mail, can you please attach it directly to this issue?

patman15, 2025-02-13:
> The problem with the disappearance of the data seems to be the retrieval function of the additional batteries.
> There is probably a round trip in BP01, then BP02, then BP03 etc.

I'm not sure what you mean. I can see from the log that you only added BP00, no other device is there. So why do you say the data disappears? When was it available?
The functionality to query all packs via a single BT connection is not yet there, so I'm confused about your question.

patman15, 2025-02-13:
Ok, got it, I can have a look at it, but this is a different issue than mentioned in the title, so let's open a new one -> #189

his1220, 2025-02-17:
I will help testing whenever needed. Thanks for your great work!!

idanny50, 2025-02-18:
as I know seplos v2, when in packed, bp00 it is master, and from there you access slaves units. When i connect with phone I only see bp00 pack, I connect to it and then I can change to slave packs and read info about them.

patman15, 2025-02-21:
@bogi9999, @his1220, and **anyone** who has a Seplos v2 and is willing to help. I have redone and smoke-tested the new code.
Please install the [beta release](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.13.0-beta.0), see if it does what you expect and be so kind to provide me:
- any findings you have (deviations of values to the app)
- info about your setup (how many battery packs in total, how many BT modules, the address of the individual packs)
- a debug log of a few minutes

As per the definition in the [README](https://github.com/patman15/BMS_BLE-HA#provided-information) you should the sensors should provide the overall system information and each sensor (as listed) should have an attribute, providing the individual pack information. Thanks a lot for helping!

> [!IMPORTANT]
> This release requires HA version 2025.2.2 or later!

bogi9999, 2025-02-21:
Okay, the result is different than expected.
I now have the capacity of the entire system. So all five packs. Also the total current and the total power as well as the total SoC. But the whole thing is only displayed briefly and the Name is still BP00.
I have cross-checked the displayed information with the Victron Dashboard - the data is valid.
There is only the Information from the "multi-pack" no informations/data from one of the singe packs.
After downgrading to the current release, only one pack is available again.

[home-assistant_bms_ble_2025-02-21T21-59-42.575Z.log](https://github.com/user-attachments/files/18916299/home-assistant_bms_ble_2025-02-21T21-59-42.575Z.log)

patman15, 2025-02-21:
> There is only the Information from the "multi-pack" no informations/data from one of the singe packs.

Have you checked the attributes, as mentioned _... and each sensor (as listed) should have an attribute ..._?
Anyway, the system does not report all pack information, just the first. Can you generate a BT recording of the app when switching to the pack tabs?

patman15, 2025-02-28:
@bogi9999 any follow up action you want me to take?

bogi9999, 2025-03-02:
@patman15 Sorry, I was away for a few days.
I'm not sure whether the access to the data of the “slaves” with the V3 BMS also works with the V2 BMS.
I can't actually select them in the App. I can only select BP00 and the sum of all data. The total is also currently displayed. If someone is reading along here who also has several V2 BMS, could they perhaps create a log?

andreas-bulling, 2025-03-02:
According to this repository/tool, access to multiple batteries should be possible through rs485
https://github.com/mworion/dbus-seplos

bogi9999, 2025-03-03:
> According to this repository/tool, access to multiple batteries should be possible through rs485 https://github.com/mworion/dbus-seplos

Of course this is an option, but unfortunately it does not work with the master if it is connected to the PV system via CAN. Then only the BP01-BPxx packs are available.

andreas-bulling, 2025-03-03:
As far as I understand, reading out both batteries via RS485 should work?!

bogi9999, 2025-03-03:
I use more than two of these packs. And as soon as the master is connected to CAN, it can no longer be read out via RS485, but via Bluetooth. Only the slaves can still be accessed via RS485.
In addition, only one RS485 connection is possible. And I need the connection primarily for the Cerbo (in my case the Victron CAN protocol). The Bluetooth option would be to monitor the health of the cells/packs.

patman15, 2025-03-04:
Hi @bogi9999, @andreas-bulling! My main question is: is it possible to query the cell states of packs other than the main (BP00) through the BP00? There are commands to query the packs (you can give the number in the query). This would mean that there is some sort of connection between the packs and the master. So far all packs were queried individually using BP00, BP01, ... but there are users who have only one Bluetooth module.
Can someone investigate, clarify, provide logs ... ?

bogi9999, 2025-03-05:
Hi @patman15 
I get the other packs displayed, if I want to select them, an error message appears, presumably in Chinese. I can select BP00 or “Multi” which is the sum of all packs. 
In my case, only the master actually has a Bluetooth module.

patman15, 2025-03-05:
Hi @bogi9999 ,

Now I'm confused, in the formulation of this [request](https://github.com/patman15/BMS_BLE-HA/issues/171#issue-2827638421) you wrote:
> The Master knows all values of each Pack and each Cell. **You cann access thes information through an bluetooth connection to the master (BP00)**, and switch then to the next Pack in the Bluetooth App (BP01/02/03.....)

but now you say:
> I get the other packs displayed, if I want to select them, an error message appears, presumably in Chinese. I can select BP00 or “Multi” which is the sum of all packs. In my case, only the master actually has a Bluetooth module.

In the logs of the modified version I can clearly see that asking the master for data from the other packs just returns the information of the first (master) pack. Numbers are just identical, as I think I saw also when doing the original implementation.

So can you please summarize your view, because mine is, the app does not show other data, the commands just deliver x-times the same, so in my opinion, you **cannot** ask the master for the other packs in Seplos v2 (v3 works).
Unless, there is an additional issue, e.g. addressing. I assume addresses 0, 1, 2, 3, ...
Maybe you want to use Google translate to figure out what the error message means?

Please let me know how you think this can work. Unfortunately, I do not see much input from other Seplos v2 users that could clarify this issue.

bogi9999, 2025-03-05:
Hi @patman15 
But it is as I say. The master has all the information for the slave packs, because it outputs total information for all packs.
I've been getting this information since your last update.
Previously, I only had the information from the master itself, i.e. from BP00.
What can I say. On the other hand, I can reach and read out all the slaves via the RS485 port on the master.
I'll try to take a screenshot of the message/app later.

patman15, 2025-03-05:
> But it is as I say. The master has all the information for the slave packs, because it outputs total information for all packs.

Via CAN or RS485 you mean? Can be, but maybe the information is internally not provided to the BT module that is what needs to be clarified.

> I've been getting this information since your last update.

I know there is **some** information. But, emphasizing my [previous message](https://github.com/patman15/BMS_BLE-HA/issues/171#issuecomment-2700083448), did you notice that the voltage level of cell 0 is equal to 16 to 32 and 48? Same for cell 1, 17, 33, 49. Also SoC and current are equal among all packs. So I highly doubt this is correct information.

bogi9999, 2025-03-05:
[image] 

[image] 

[image] 

[image] 

[image] 

The message is "Equipment not asked for, No. re-sweep"

patman15, 2025-03-05:
My translator says "device is not available, please scan again" That sounds to me as if the app is trying to connect directly via Bluetooth and not via the master and is consistent with the other hints I have. I fear that Seplos V2 cannot access the other packs unless they have their own BT module.

bogi9999, 2025-03-05:
> My translator says "device is not available, please scan again" That sounds to me as if the app is trying to connect directly via Bluetooth and not via the master and is consistent with the other hints I have. I fear that Seplos V2 cannot access the other packs unless they have their own BT module.

But how does the master, which also provides the PV system with the real sum of the individual packs, know that there are 4 other packs and gets the values from there?
At most, it could be that it does not function as a BT gateway.

andreas-bulling, 2025-03-06:
I noticed that, as long as two packs are disconnected from each other, they each show up as individual BT devices. Once connected, the slave disappears and only the master remains. Can you confirm that's the same for you?

bogi9999, 2025-03-06:
> I noticed that, as long as two packs are disconnected from each other, they each show up as individual BT devices. Once connected, the slave disappears and only the master remains. Can you confirm that's the same for you?

Thats right.

patman15, 2025-03-06:
> But how does the master, which also provides the PV system with the real sum of the individual packs, know that there are 4 other packs and gets the values from there?
At most, it could be that it does not function as a BT gateway.

@bogi9999 Maybe there is one misunderstanding. What _is_ possible is to provide the overall sum for the entire system. That is what the patched version shows. What is _not_ possible is to show the values of each individual pack then, because the main system reports for each pack the same value. No idea why. There might be a mistake, or you just can't query it using Bluetooth, as you said.

maps0n, 2025-03-06:
I have two Mason SEPLOS Battery packs, each with SEPLOS BMS 10E (V2) and BT Modul.
The Mastepack (BP0) is connected by CAN bus to a Victron Multiplus GX and the two packs are interconnected by RS485
There is a design flaw with this SEPLOS BMS: When the CAN bus is in use, the desktop software can only monitor the slave battery pack(s), but not the master.  :-(
So, I am using the iOS app to monitor both packs:
So, I am really excited about @patman15 's  work to give access via home assistant:

I'd like to provide feedback information as well, but I am stuck at setting the system up.
• Installed a ESP32 BT Proxy via ESPhome, and add "active: true" in yaml
• I have downloaded the BLE Battery Management System (BMS) V1.13.0 in the HACS
• When I try to add this integration, there is the message: "Keine unterstützen Geräte via Bluetooth gefunden."
• The advertise monitor proofs that the ESP32 Proxy found the two BMS as BP00 and BP01 with RSSI of -79
• In terminal the command     bluetoothctl devices >> No default controller available"
• manually connecting does not work neither

Did I forget a step or did something wrong?

Right click in the advertise monitor not the BP01 und BP00 entries produces: {"name":"BP01\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","address":"04:7F:0E:13:21:33","rssi":-79,"manufacturer_data":{},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"94:54:C5:E8:38:24","connectable":true,"time":1741291451.996939,"tx_power":null}

{"name":"BP00\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","address":"04:7F:0E:13:21:DB","rssi":-77,"manufacturer_data":{},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"94:54:C5:E8:38:24","connectable":true,"time":1741291452.2929354,"tx_power":null}

bogi9999, 2025-03-06:
> There is a design flaw with this SEPLOS BMS: When the CAN bus is in use, the desktop software can only monitor the slave battery pack(s), but not the master. :-(

Well, that is only partly correct.
I learned a few days ago that if you use a splitter on the CAN port, you can also get the RS485 info from the master, so I think I'll build a cable that will allow me to do that. 
Pin 4/5/6 for CAN and 6/7/8 for RS485 on the CAN port (attention in this case settings to 9800Baud). You can then connect the slave packs to the second RS485 port of the master and query them as usual via RS485 and 19200baud.

andreas-bulling, 2025-03-06:
> Well, that is only partly correct. I learned a few days ago that if you use a splitter on the CAN port, you can also get the RS485 info from the master, so I think I'll build a cable that will allow [...]

Is this the wiring that mworion is using for his dbus-seplos project (sorry, I still struggle to understand how to set this all up. Now you also mention specific pins - which I haven't seen in his documentation..)

bogi9999, 2025-03-06:
Oh, I've already had success with exactly that splitter. --> https://amzn.eu/d/0swxTuy
But i do not want to use a Splitter there.

The Wiring wil be -->

CAN PORT

Pin 4 - CAN H --> Victron PIN 7
Pin 5 - CAN L --> Victron PIN 8
Pin 6 - Ground --> Victron PIN 3
also
PIN 6 - Ground --> RS485 GND
Pin 7 --> RS485 A
PIN 8 --> RS485 B

patman15, 2025-03-08:
@maps0n 

I opened a separate issue to follow-up with your topic.

bogi9999, 2025-03-12:
@maps0n @andreas-bulling 

Got that shit done as explained. 

@patman15 

i will get another Pack on Saturday. Probably next week i can try again the BLE Connection to the Slaves with a new Pack.

his1220, 2025-03-18:
Just testing but what concerns me now is that the BP01 Battery is now named

BP0��0� AT+NAME=BP01
so that it can be added as battery 

[image]

Privatecoder, 2025-03-27:
@bogi9999 it is even better to combine each's packs RS485-part of the CAN-port to a separate RS485 bus as described [here](https://github.com/syssi/esphome-seplos-bms/issues/116).

I updated my script and included a [drawing in my repo as well](https://github.com/Privatecoder/seplos-mqtt-remote-rs485/).

This way the RS485-ports on each pack can be used soley for intra-pack communication without an additional master on the bus to read the packs information.

patman15, 2025-03-27:
@Privatecoder does this mean that with the right cabling, the BT module of the master should be able to provide the information of all packs? So far I could not see a prove for that. Would you, by any chance, be able to test my BT implementation?
Thanks, Patrick

Privatecoder, 2025-03-27:
@patman15 I don't think so as this method is somewhat exploiting the _inverter_-interface (its RS485 part) to grab the packs information.

Tulexcorp, 2025-04-15:
Hi, per your recommendation in #260, the log and some observations. If you need a redo on the log or any other info, let me know.

[home-assistant_bms_ble_2025-04-15T06-26-54.719Z.log](https://github.com/user-attachments/files/19752382/home-assistant_bms_ble_2025-04-15T06-26-54.719Z.log)

Just for reference, of the 13 entities for each battery, only a few are different for the master battery. Not an issue.
"Problem" - Master is fine, says OK, all the others say problem, "show more" shows the problem being "unavailable".
Battery Current, Cycles, Volt, and Battery Level all say Pack Value under Attributes, but far as I can tell, the value is actually the master battery's value, not some pack average. Didn't check Link Quality or Signal strength, as I left those unenabled.

Setup is 6 Seplos batteries in parallel, master battery connected to inverter via can. Installing the master battery first did not expose any of the other batteries.

patman15, 2025-04-15:
@Tulexcorp sorry, I need to bug you again, the log is too short. I need it to be around 3 minutes long to see all packs at least once.

Tulexcorp, 2025-04-15:
[home-assistant_bms_ble_2025-04-15T08-29-25.787Z.log](https://github.com/user-attachments/files/19754256/home-assistant_bms_ble_2025-04-15T08-29-25.787Z.log)

No Problem. Forgot I was logging, so it's kind of long. Tried to do a shorter one, but it just keeps adding to this one.

patman15, 2025-04-15:
@Tulexcorp could you please install the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/171-add-stacked-seplos-v2-battery-attributes) via [manual install](https://github.com/patman15/BMS_BLE-HA/tree/171-add-stacked-seplos-v2-battery-attributes?tab=readme-ov-file#installation) and provide me another debug log?
The previous one was excellent and everything seems to work fine. I fixed the problem reporting, which was due to a strange message that the slaves send.
I hope that I finally can aggregate the data of the slaves into the master device, but let's see what your system reports.

Tulexcorp, 2025-04-15:
[home-assistant_bms_ble_2025-04-15T20-34-01.598Z.log](https://github.com/user-attachments/files/19765298/home-assistant_bms_ble_2025-04-15T20-34-01.598Z.log)

After this install, most entities for batteries 2-6 show no values.

Tulexcorp, 2025-04-15:
Readings I'm now getting.

[image]

patman15, 2025-04-15:
> After this install, most entities for batteries 2-6 show no values.

The log is very limited because you seemed to have renamed the directory, which impacts debug logging.

Tulexcorp, 2025-04-15:
OK, let me do it over. I renamed the directory and then did a new install so that I could just undo it if it didn't work. 
I will delete both and start over.

Tulexcorp, 2025-04-15:
Deleted everything related to this integration from the custom components directory. Removed the Hacs. Did a full system reboot.
Nothing showed up in devices. Copied the files to the directory, restarted HA. The six batteries show up as bms_ble devices even before installing the bms_ble integration. 

[home-assistant_bms_ble_2025-04-15T22-22-21.358Z.log](https://github.com/user-attachments/files/19766294/home-assistant_bms_ble_2025-04-15T22-22-21.358Z.log)

Tulexcorp, 2025-04-16:
Just wanted to verify something. As noted earlier, because HA doesn't play well with some Realtek BMS adapters, I can only have one devise on that adapter, and it HAS to be the master battery. Any other combination breaks it. Knowing that, I realized having the Realtek adapter still in play could be causing issues, even though it's not breaking the readings. So, I removed one battery from the approved bluetooth adapter and moved the master to that adapter, allowing me to disable the Realtek adapter. Rebooted the system.  Of the 4 slaves that I can now see, the results are still the same.  Eight of the entities still show no values for the slaves.

patman15, 2025-04-16:
Hey @Tulexcorp, your data is incredibly helpful! Thanks a lot! I had one parsing error, mostly due to incorrect test data. No idea, where I had that from, but I could now fix it hopefully. Can you check again with the updated branch? Basically, you should find all data in the attributes of the BP00 device. I just kept the other device to verify how they behave, so in theory you should be able to ignore them.
Please provide me a debug log again!

Tulexcorp, 2025-04-16:
Log below for just the master battery installed. One device, 13 entities. I can see 96 values under attributes for cell delta. Because it only reports 2 places in attribues, I can't tell if what the entity is reporting is the master battery or some kind of average. Other entities also list multiple attributes, but don't think they are correct. For example,  current might show as 2 amps for current entity, but attributes all show 0. 

[home-assistant_bms_ble_2025-04-16T17-08-34.291Z.log](https://github.com/user-attachments/files/19781458/home-assistant_bms_ble_2025-04-16T17-08-34.291Z.log)

Will install the others and send a log.

Tulexcorp, 2025-04-16:
[home-assistant_bms_ble_2025-04-16T17-22-16.492Z.log](https://github.com/user-attachments/files/19781597/home-assistant_bms_ble_2025-04-16T17-22-16.492Z.log)

Log with master and 4 slaves. Attributes stayed the same for master battery, but no attributes available for slaves. Same results, no values for slave batteries for most entities.

Tulexcorp, 2025-04-17:
I went into the bluetooth app. Won't connect if HA is connected to it, which makes sense. There is no logging ability. There is the ability to view it as multi which sees it as one big battery, or each individual battery. Everything looked normal.

I'm good with using the regular version. The only real benefit to what I believe you are trying to do here is to eliminate using multiple slots. Only really an issue if you are running other bluetooth items or have a lot of batteries.

patman15, 2025-04-17:
Thanks for the logs, the master still does not report the other packages, just clones of it's own battery values. 😞 

> I went into the bluetooth app. Won't connect if HA is connected to it, which makes sense. There is no logging ability. There is the ability to view it as multi which sees it as one big battery, or each individual battery. Everything looked normal.

I had it in my previous comment, here are two links on how to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/).

> There is the ability to view it as multi which sees it as one big battery, or each individual battery. Everything looked normal.

I'd need a recording of the BT traffic when using this view, this is the last chance to figure out whether also the app uses multiple BT connection at the same time.

> I'm good with using the regular version. The only real benefit to what I believe you are trying to do here is to eliminate using multiple slots. Only really an issue if you are running other bluetooth items or have a lot of batteries.

That and the fact that you get e.g. the current automatically summed up. All other BMSs do it like this, even Seplos v3.

Tulexcorp, 2025-04-17:
Sorry, didn't catch that those were links. I installed wireshark and followed the directions. I thought I captured the bluetooth, but it doesn't create a log I can read or at least understand. Or, it's above my head and I'm doing it wrong. Is a .pcapng file what you would be looking for?

patman15, 2025-04-17:
> Is a .pcapng file what you would be looking for?

Yes. that sounds good to me!

Tulexcorp, 2025-04-17:
[seplos.txt](https://github.com/user-attachments/files/19800853/seplos.txt)

You will have to rename it as a .pcapng file as it wouldn't let me attach it as was.

patman15, 2025-04-18:
@Tulexcorp file is readable, but USB communication, not the BT traffic. Are you sure you selected the right interface when recording?

Tulexcorp, 2025-04-18:
I will try again. As I can't tell what it is, I have no idea what I'm sending. The directions make it sound simple, but in reality, setting up Wireshark and using it is not intuitive.

patman15, 2025-04-18:
Sorry, for that. As I'm no iOS user, I cannot help more here.

Tulexcorp, 2025-04-18:
Unless someone can walk me through the setup, I don't believe I can help with this. The directions say:

"Helpfully, newer versions of [Wireshark](https://www.wireshark.org/) include an [androiddump](https://www.wireshark.org/docs/man-pages/androiddump.html) utility to capture Bluetooth traffic directly from Android phones. **The setup couldn't be simpler**: connect the phone to the computer via USB, enable USB debugging, select the "Android Bluetooth" capture source in Wireshark, and start recording!"

If  you click on the androiddump link, it says:

androiddump(1) Manual Page
NAME
androiddump - Provide interfaces to capture from Android devices

SYNOPSIS
androiddump [ --help ] [ --version ] [ --extcap-version ] [ --debug ] [ --extcap-interfaces ] [ --extcap-dlts ] [ --extcap-interface=<interface> ] [ --extcap-config ] [ --extcap-capture-filter=<capture filter> ] [ --capture ] [ --fifo=<path to file or pipe> ] [ --adb-server-ip=<IP address> ] [ --adb-server-tcp-port=<TCP port> ] [ --logcat-text=<TRUE or FALSE> ] [ --bt-server-tcp-port=<TCP port> ] [ --bt-forward-socket=<TRUE or FALSE> ] [ --bt-local-ip=<IP address> ] [ --bt-local-tcp-port=<TCP port> ]

androiddump --extcap-interfaces [ --adb-server-ip=<IP address> ] [ --adb-server-tcp-port=<TCP port> ]

androiddump --extcap-interface=<interface> [ --extcap-dlts ]

androiddump --extcap-interface=<interface> [ --extcap-config ]

androiddump --extcap-interface=<interface> --fifo=<path to file or pipe> --capture

DESCRIPTION
Androiddump is a extcap tool that provide interfaces to capture from an Android device. There are only two requirements:

1. You must have the Android SDK and add it to your PATH environment variable. PATH should contain directory with tools like "adb" and "android". Android SDK for various platform are available on: https://developer.android.com/sdk/index.html#Other

2. You must have permission to capture from the Android device. Some Android devices require on-screen authentication.

Don't know if all the synopsis doesn't apply as it's part of Wireshark
Don't know about Android SDK at all.  Have it where, and about adding it to what path?

Simply, I don't see an actual setting in wireshark to enable USB debugging, and I don't see Android bluetooth as an option.

patman15, 2025-04-18:
And why don't you use my suggestion to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/)? It's much simpler ...

[11 more replies]

closed 2025-04-19 by patman15