patman15/BMS_BLE-HA #172: Hi I have a problem with the temperature sensor on the JK BMS. Maybe someone know how to solve it?

patman15, 2025-02-04:
Hi I have a problem with the temperature sensor on the JK BMS. Maybe someone know how to solve it? 
I have several JK BMS conected,  some work well, but others - incorrect temperature sensore value ~ 2550 C`

_Ursprünglich gepostet von @vchaban88 in https://github.com/patman15/BMS_BLE-HA/issues/150#issuecomment-2631650206_

patman15, 2025-02-04:
@vchaban88 please do not mix topics and create new issues to be able to separately track them.

Regarding your issue, I would need a debug log in order to be able to help you. Can you generate one and attach it to this ticket? Thanks!

patman15, 2025-02-08:
@vchaban88 can you please check the linked branch via manual install? I think I found the issue.

vchaban88, 2025-02-08:
Yes, thank you, I'll definitely try it.

vchaban88, 2025-02-09:
After editing jikong_bms.py I got -73 degrees on the sensor (before it was 2550 degrees). The real battery temperature is 10 degrees.

patman15, 2025-02-09:
Can you please attach a debug log? Otherwise I cannot check the issue.

vchaban88, 2025-02-09:
[home-assistant_bms_ble_2025-02-09T09-14-06.795Z.log](https://github.com/user-attachments/files/18723912/home-assistant_bms_ble_2025-02-09T09-14-06.795Z.log)

patman15, 2025-02-09:
I need to dig deeper, seems to be an issue with your firmware version. Meanwhile you can take the first temperature sensor (from the attributes) only, by adapting the templates in the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking).

patman15, 2025-02-15:
Hi @vchaban88! Ok, I think I fixed it now finally. Can you check the version in [this branch](https://github.com/patman15/BMS_BLE-HA/tree/172-jk-bms-wrong-temperature-values)?

mrazekfilip, 2025-02-21:
I apologize for the delay. I just tried it, but it doesn’t seem to solve the problem. Instead of approximately -66°C, I’m now getting around -99°C. And the runtime is decreasing (I would expected it to increase).

What can I do to give you more information?

patman15, 2025-02-21:
>  Instead of approximately -66°C, I’m now getting around -99°C.

Strange.

>  runtime is decreasing (I would expected it to increase).

Sounds correct as
 	_runtime 	[s]: 	remaining discharge time till SoC 0%, unavailable during idle/charging_

> What can I do to give you more information?

If you could get me a debug log of a few minutes that would be awesome. Just click the _enable debug logging_, wait a bit, click again and it will prompt for download. Attach me the file here. Thanks!

[image]

mrazekfilip, 2025-02-21:
> Sounds correct as
> runtime [s]: remaining discharge time till SoC 0%, unavailable during idle/charging

I realized it is correct when I sent the comment. 

Here is the debug log:

[home-assistant_bms_ble_2025-02-21T21-09-53.672Z.log](https://github.com/user-attachments/files/18915513/home-assistant_bms_ble_2025-02-21T21-09-53.672Z.log)

patman15, 2025-02-21:
Hmm, your BMS claims that all sensors are connected: `'temp_sensors': 255`

patman15, 2025-02-21:
@mrazekfilip Ok, I updated the branch and excluded the -200 sensors. Finally, should work now. Let me know!

mrazekfilip, 2025-02-22:
It works. 👍 Thank you

closed 2025-02-22 by patman15