patman15/BMS_BLE-HA #173: BMS not found after updating from v1.10.0 to v1.11.6

karolmc02, 2025-02-04:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Manufacturer: Eleksol / Asembly plant: Jiangxi Jingjiu Power Energy 

### Describe the issue

After updating from v1.10.0 to v1.11.6 there was some kind of a problem and my device lost connection (the sensors started to give unavailable). I have removed it and tried to add it again but it was no longer found.

Searching the devices with `bluetoothctl` it is there:
```
bluetoothctl devices
Device A5:C2:37:27:AE:24 12300DE00013

bluetoothctl info A5:C2:37:27:AE:24
Device A5:C2:37:27:AE:24 (public)
        Name: 12300DE00013
        Alias: 12300DE00013
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
        Modalias: usb:v248Ap8266d0001
        AdvertisingFlags:
  06                                               .               
```

I have downgraded to the v1.10.0 and it was instantly detected again after restart.

I don't know the exact type of the BMS my battery has.

Thank you in advance.

### Reproduction steps

```text

```

### Debug logs

.

patman15, 2025-02-04:
> I don't know the exact type of the BMS my battery has.

Can you please name the BMS the integration shows? Please also add a debug log from the old release including a restart so I have all info together.

Thanks!

karolmc02, 2025-02-04:
Hi!

My bad. I did not realised the BMS name was displayed.

The name displayed in the old version is: Smart BMS by Jiabaida.

Also I attach the logs. (Please ingnore the helpers errors. Those are because before adding the integration again I had the entities with different ids)

[home-assistant_bms_ble_2025-02-04T13-51-44.083Z.log](https://github.com/user-attachments/files/18658052/home-assistant_bms_ble_2025-02-04T13-51-44.083Z.log)

patman15, 2025-02-04:
Did you see the note in the release notes from 1.11? https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.11.3

I'll fix the detection for your device end of this week. 

Thanks for the report,
Patrick

karolmc02, 2025-02-04:
> Did you see the note in the release notes from 1.11? https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.11.3
>

If you are talking about the warning then yes. That's why I opened the issue. I did not realize earlier that this affected me.

Thank you very much,
Karol

patman15, 2025-02-08:
Can you please check the linked branch via manual install? It should work now.

karolmc02, 2025-02-08:
I have tried it and it does not seem to be working.

karolmc02, 2025-02-09:
I tried changing the UUID in matcher list for this one: `0000ff00-0000-1000-8000-00805f9b34fb` and with this one it seams it is detected like before.

patman15, 2025-02-09:
> I tried changing the UUID in matcher list for this one: `0000ff00-0000-1000-8000-00805f9b34fb` and with this one it seams it is detected like before.

Well that is like a "catch-all" clause. Can you please provide a debug log that contains _AdvertisementData_? This usually requires a restart of the integration. Or, if you you are already on HA2025.02, you can check the [advertisement monitor](https://github.com/patman15/BMS_BLE-HA/tree/main#if-your-device-is-not-recognized). The device seems to broadcast less info than listed by `bluetoothctl`. Could get challenging ...

patman15, 2025-02-09:
Ah and do you know about the naming scheme they use? `12300DE00013` could I filter for `123*`, I mean also for other users, not only for you ... 😄

karolmc02, 2025-02-09:
I have done some more digging. The original bluetoothclt info seams incorrect. I believe it returned all those UUIDs because i had the device connected already.

Sorry about that. It's my first time doing any code with bluetooth.

This is the output after I removed the device, scanned it again and get the info:
```
bluetoothctl info A5:C2:37:27:AE:24
Device A5:C2:37:27:AE:24 (public)
        Name: 12300DE00013
        Alias: 12300DE00013
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0xae24 (44580)
        ManufacturerData.Value:
  27 37 c2 a5                                      '7..            
        RSSI: 0xffffffc6 (-58)
        AdvertisingFlags:
  06                                               . 
```

I tried to replace the matcher you added using uuid with the following one using manufacturer_id and it recognises the bms:
```
[
            { # Eleksol
                "service_uuid": BMS.uuid_services()[0],
                "manufacturer_id": manufacturer_id,
                "connectable": True,
            }
            for manufacturer_id in [0xae24]
]
```

> Ah and do you know about the naming scheme they use? 

I have also looked into it and it seams the use the battery serial number as name and as far as I found there is not any kind of a pattern.

patman15, 2025-02-09:
> I tried to replace the matcher you added using uuid with the following one using manufacturer_id and it recognises the bms

Agreed, but the issue is that the value `0xae24` is the last part of the MAC address which changes per device. If it would be the first two bytes (like it should be) 🤕  then I could use it.

> it seams the use the battery serial number as name 

Is the whole name the serial number or could I use `123??DE*` as pattern?

karolmc02, 2025-02-09:
> Is the whole name the serial number or could I use 123??DE* as pattern?

I will try to look into it a bit more.

karolmc02, 2025-02-11:
Hi!

I've looked into it more. I tried finding more Eleksol baterries and looking into documentation (not many on the net). The only pattern I was able to determine would be:
* It starts with 12
* Then 3 digits
* Then 2 letter
* Ending with 5 digits

Writing it as a RegEx would be:  `12\d{3}[A-Z]{2}\d{5}`.

patman15, 2025-02-11:
Ok, now I got it, it is 12V -> 12, 300AH -> 300, so 12300DE... knowing that and knowing that I need at least 3 characters, I will look at `121?0*`, `12200*`, `12300*` that should work for the batteries currently on the website. Thanks!

karolmc02, 2025-02-14:
Hi!

I have checked it with the updated branch and it work fine.

Thank you!

PR #178 (2025-02-15): add detection for Eleksol
fixes #173
closed 2025-02-15 by patman15