patman15/BMS_BLE-HA #175: Battery #2 not being found

haysdb, 2025-02-06:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I installed BLE Battery Management a few days ago. I connected one of two identical brand new Seplos batteries with Seplos v3.0 BMS. It took hours and hours of “initializing” but it did eventually populate the sensors. 

When I try to connect the second battery I get an immediate message “no supported devices found”. If I enable logging I see nothing but a “user step” message. No errors.

I have no trouble finding and connecting to the second bms with the Seplos BMS app and the LightBlue iPhone app.

The HA Pi is about 7’ from the batteries with clear line of sight. There’s no reason it shouldn’t have a strong signal. 

Link quality is 96%
Signal strength is -61 dBm

The first battery is logging a lot of “Failed to connect” errors, so it’s BARELY working. 

At present I’m using the built-in bms in a Raspi 4 while I wait for a BT adapter from your recommended list. 

On a different topic, why does the attributes of delta_voltage show just one decimal of precision in Settings > Devices > Integrations but 3 in Developer tools? The attributes in STATES shows 3 decimals so the data is there but it’s getting rounded for display. Display precision is set to .001 in configuration but that doesn’t seem to be applied to the attributes.

haysdb, 2025-02-06:
I've been trying to find the MAC address of the second BMS with no success. Apparently MAC addresses are random these days when it comes to BT devices.

patman15, 2025-02-06:
Hi! If you see a lot of `failed to connect` errors I wonder why the `link quality` sensor is still showing 96%. If you want me to take a deeper look, I would need a debug log to analyze. In general, since you already ordered it, please try the new BT adapter, best with a USB extension (~30cm) that keeps the BT adapter away from the Raspberry. There might be disturbances with the on-board wireless chips.

> why does the attributes of delta_voltage show just one decimal of precision in Settings > Devices > Integrations but 3 in Developer tools?

Unfortunately, I have no clue, but this is "feature" of Home Assistant. I cannot influence the presentation of the attributes. See the FAQ on how to generate individual sensors in order to see full precision. Basically, anytime you use it yourself, you can also influence the presentation.

haysdb, 2025-02-07:
I have created the individual sensors.

I've gone a step beyond a bluetooth adapter and ordered an ESP32 to use as a Bluetooth Proxy. I didn't want to put the HA Pi near the batteries anyway.

Since there are changes on the way there no point in trying to figure out why the built-in bluetooth is not working.

patman15, 2025-02-07:
> Since there are changes on the way there no point in trying to figure out why the built-in bluetooth is not working.

Meaning that I should close the ticket?

haysdb, 2025-02-09:
Yeah. O\I'll create a new one if i still have there issue with one of there recommended USB adapters

closed 2025-02-09 by patman15