patman15/BMS_BLE-HA #176: Battery creabest not found  by the integration

Rik83, 2025-02-06:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I use bluetooth proxies with esp32 modules.
It works because I can find some bluetooth devices such as thermometer and add corresponding integration.

If I use the monitor of the bluetooth integration, I can see my battery (creabest 135a bluetooth) :
{"name":"140R000288","address":"38:AB:41:1C:B8:C6","rssi":-82,"manufacturer_data":{"0":"ffffffff6400ff000000000000000000"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"30:C6:F7:24:C6:A8","connectable":true,"time":1738855336.3501194,"tx_power":null}

But my battery creabest is not detected by the integration.
When I try to install I get the message :
BLE Battery Management
No supported devices found via Bluetooth.

What did I miss ?

(no app is connected to the battery and only one bluetooth device is connected to the proxy)

Thank you

patman15, 2025-02-06:
Hmm, looks like the automatic detection does not catch your battery. For how long did you wait? At least 5 minutes? If so, the I need to fix it tomorrow. Please let me know if it recovered or still is stuck.

patman15, 2025-02-07:
@Rik83 when trying to fix your detection, I stumbled over an issue in Home Assistant (see referenced bug report) that hinders a fix. Need to wait for a reply by the maintainers. Sorry for the delay.

Rik83, 2025-02-09:
Excellent, thank you for your great work !

agnuwi, 2025-02-10:
> [@Rik83](https://github.com/Rik83) I will update the integration after HA 2025.2.2 is published next Friday. It's required for the tests to pass and that my change take effect.

Thanks for your effort. I have the same issue withe my new Creabest battery (11/2024) and can provide debug logs. The integration of my "old" battery (05/2024) works fine. Creabest has changed their BMS software.

Rik83, 2025-02-18:
Battery creabest still not found after updates...

patman15, 2025-02-18:
Hi! You still need to use the [branch ](https://github.com/patman15/BMS_BLE-HA/tree/176-battery-creabest-not-found-by-the-integration) from PR #185 to get it working. I cannot merge it before finishing the work on upgrading the CI/CD setup (PR #193) to HA 2025.2.2. I really need to carefully move forward as this upgrade forces >150 people to upgrade their HA installation and I cannot go back with reasonable effort.