patman15/BMS_BLE-HA #183: BLE Proxy how to limit to 3 devices

his1220, 2025-02-09:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

How can I limit the BLE Proxy (Olimex) to only 3 devices. I use 6 Seplos V2 Boxes which are found and correctly read in principle , but I have a lot of other bluetooth devices which I want to exclude (filtering?) from HA at all and also from the proxy. They are the cause for interruptions in reading the battery boxes.

Any idea what I could do, or can we have a filter option in the settings?

Thanks in advance

his1220, 2025-02-09:
Many thanks for you reply.
I will try another Proxy but what I see from logging is the fact th)at mostly the boxes connect to hci0 from the Raspberry Board , only 2 boxes connect with the olimex proxy (e.g. just at that moment  and sometimes none. They randomly switch around. Can I bind three of them to the proxy?

patman15, 2025-02-09:
Raspberry BT is not the best, especially for many devices. I would recommend you disable the internal BT and use 2 or more proxies (or an [USB BT adapter](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters) instead). Check your `RSSI` for a good value.
Other option: maybe you could connect your boxes via RS485 and then use only the master via Bluetooth? See #171 and #153 depending on which Seplos version you have. The V3 is ready, the V2 will come soon.

> Can I bind three of them to the proxy?

Sorry again, I'm not aware of such an option, but if you find it, please let me know for others ...

Best regards,
Patrick

his1220, 2025-02-11:
Many thanks for your great work and replies !! can`t await seplos V2 , 
if I find anything usefull during my tests I will post it

closed 2025-02-13 by patman15