patman15/BMS_BLE-HA #184: No supported devices found JK BMS

mrazekfilip, 2025-02-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

The integration says, "No supported devices found via Bluetooth." However, I can connect to my BMS JK_BD4A8S4P via bluetoothctl. 

Here is the information from the HA 2025.02 bluetooth integration:
```
{"name":"JK-LiFePo4","address":"98:DA:20:09:DF:DF","rssi":-73,"manufacturer_data":{"2917":"88a098da2009dfdf"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000ffe0-0000-1000-8000-00805f9b34fb"],"source":"50:84:92:26:2D:38","connectable":true,"time":1739096921.0349336,"tx_power":null}
```

blueotoothctl logs:
```
bluetoothctl connect 98:DA:20:09:DF:DF
Attempting to connect to 98:DA:20:09:DF:DF
hci2 type 6 discovering off
hci2 type 6 discovering on
[CHG] Device 18:3C:B7:F9:FC:58 RSSI: 0xffffffb0 (-80)
[CHG] Device 48:CD:ED:60:FD:C1 RSSI: 0xffffffb6 (-74)
[CHG] Device 79:75:2B:C1:19:5B RSSI: 0xffffffb8 (-72)
[CHG] Device 79:75:2B:C1:19:5B ManufacturerData.Key: 0x004c (76)
[CHG] Device 79:75:2B:C1:19:5B ManufacturerData.Value:
  0c 0e 08 65 bb 9f a3 2c 44 ef f7 e4 60 ee 0e 8d  ...e...,D...`...
  10 06 4c 1d e4 5d 65 28                          ..L..]e(        
[CHG] Device 6D:26:6B:87:B0:1C RSSI: 0xffffffbc (-68)
[CHG] Device E5:26:A7:5B:D1:55 RSSI: 0xffffffb6 (-74)
hci0 type 6 discovering off
hci0 type 6 discovering on
[CHG] Device 6B:E0:7A:AC:E6:AC RSSI: 0xffffffac (-84)
[CHG] Device 45:7B:87:F5:E9:34 RSSI: 0xffffffa2 (-94)
[CHG] Device CA:A7:48:0F:80:DF RSSI: 0xffffffb4 (-76)
[CHG] Device 49:24:BC:24:1C:8E RSSI: 0xffffffb6 (-74)
[CHG] Device 49:24:BC:24:1C:8E ManufacturerData.Key: 0x004c (76)
[CHG] Device 49:24:BC:24:1C:8E ManufacturerData.Value:
  0c 0e 08 65 bb 9f a3 2c 44 ef f7 e4 60 ee 0e 8d  ...e...,D...`...
  10 06 4c 1d e4 5d 65 28                          ..L..]e(        
[CHG] Device 98:DA:20:09:DF:DF RSSI: 0xffffffa4 (-92)
hci2 98:DA:20:09:DF:DF type LE Public connected eir_len 21
[CHG] Device 98:DA:20:09:DF:DF Connected: yes
Connection successful
```

```
bluetoothctl info 98:DA:20:09:DF:DF
Device 98:DA:20:09:DF:DF (public)
        Name: JK-LiFePo4
        Alias: JK-LiFePo4
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0x0b65 (2917)
        ManufacturerData.Value:
  88 a0 98 da 20 09 df df                          .... ...        
        RSSI: 0xffffffa4 (-92)
```

JK_BMS information from the app:

[image]

I hope I didn't forget to include any logs. If you would like to know additional info, feel free to ask. Thank you.

patman15, 2025-02-09:
Hi!
I can't spot an issue with the integration, it should work.
Your `RSSI` value is extremely low: -92, try to get the battery closer to your BT adapter. From experience it should well above -80 to connect.

mrazekfilip, 2025-02-10:
Hi, thank you for your quick response.
Unfortunately, I cannot move the BT adapter or the battery closer. I will get an ESP and try it with a Bluetooth proxy on the weekend.

mrazekfilip, 2025-02-16:
Hi, I installed esphome bluetooth proxy and placed it right next to the battery. The esp shows RSSI between -60 - -67 but still, I see No supported devices found when try to add new integration. 

```
{"name":"JK-LiFePo4","address":"98:DA:20:09:DF:DF","rssi":-66,"manufacturer_data":{"2917":"88a098da2009dfdf"},"service_data":{},"service_uuids":["0000ffe0-0000-1000-8000-00805f9b34fb"],"source":"24:62:AB:E6:49:14","connectable":true,"time":1739713096.3925853,"tx_power":null}
```

bluetooth proxy logs:
```
[16:28:28][C][bluetooth_proxy:091]: Bluetooth Proxy:
[16:28:28][C][bluetooth_proxy:092]:   Active: YES
[16:28:28][C][bluetooth_proxy:093]:   Connections: 3
[16:28:28][C][bluetooth_proxy:094]:   Raw advertisements: NO
[16:28:28][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
[16:28:28][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[16:28:28][C][factory_reset.button:011]: Factory Reset Button 'Factory reset'
[16:28:28][C][factory_reset.button:011]:   Icon: 'mdi:restart-alert'
[16:28:28][C][esp32_ble:418]: ESP32 BLE:
[16:28:28][C][esp32_ble:419]:   MAC address: 24:62:AB:E6:49:16
[16:28:28][C][esp32_ble:421]:   IO Capability: none
[16:28:28][C][esp32_ble_tracker:661]: BLE Tracker:
[16:28:28][C][esp32_ble_tracker:662]:   Scan Duration: 300 s
[16:28:28][C][esp32_ble_tracker:663]:   Scan Interval: 320.0 ms
[16:28:28][C][esp32_ble_tracker:664]:   Scan Window: 30.0 ms
[16:28:28][C][esp32_ble_tracker:665]:   Scan Type: ACTIVE
[16:28:28][C][esp32_ble_tracker:666]:   Continuous Scanning: True
```

Do I have to connect the battery manually to the bluetooth proxy?

patman15, 2025-02-16:
I checked the detection again with your data, works. ✅ 
 
- Your are running the [latest release](https://github.com//patman15/BMS_BLE-HA/releases) of the integration?
- Can you update to HA 2025.02 and go to the [bluetooth integration](https://my.home-assistant.io/redirect/integration/?domain=bluetooth). On your BT adapter select configure->advertisement monitor and see if you find the device there?
- Are you sure it has not been added already and/or you disabled the battery by accident? (Please check the hidden/disabled devices!)
- Please send me a debug log if nothing else helps.

Best regards,
Patrick

Gruni22, 2025-02-19:
Hi @patman15 
I also have the same problem with another battery (Creabest).
{"name":"100R0002E3","address":"BA:03:EC:11:09:09","rssi":-76,"manufacturer_data":{"21330":"2134ba03ec110909010500010000"},"service_data":{},"service_uuids":["000003c1-0000-1000-8000-00805f9b34fb"],"source":"2C:CF:67:AB:82:92","connectable":true,"time":1739968077.0545566,"tx_power":0}

The message 'No supported devices found when try to add new integration.' comes up when trying to add the "integration".

patman15, 2025-02-19:
@Gruni22 
- is this the data from the advertisement monitor?
- Are you sure it has not been added already and/or you disabled the battery by accident? (Please check the hidden/disabled devices!)
- Please send me a debug log if nothing else helps.
- Are you able to reproduce the issue with an HAOS setup?
- Did the batteries work on a 2024.x.x version of HA?

Gruni22, 2025-02-19:
- Yes it is
- Yes, it was a complete new install
- Which Debug Log? You will see the JK-BMS only.
- A Supervised or HAOS setup only discovers the JK-BMS (really fine)
- Never tried before. I am knew to Home Assistant and short prior leaving again...
:-)

[home-assistant_bms_ble_2025-02-19T18-12-47.151Z.log](https://github.com/user-attachments/files/18873484/home-assistant_bms_ble_2025-02-19T18-12-47.151Z.log)

Gruni22, 2025-02-19:
Sorry for the confusions...
I had a new Core installation and realised my JK BMS is not recognized. 
Then I realised, the Creabeast is also not there. 
Then I tried Supervised and HAOS where the JK is there. 
You are right. I should raise a new issue for the CBT. 
Perhaps I find a matching one...
And thanks for your effort.

mrazekfilip, 2025-02-19:
Hello, thank you for your advise!  You were absolutelly right, I found the battery in the disabled devices. No clue how it could happend, but thank you! Therefor I'm closing this issue. 

PS: I think I have a simillar problem as in #172. (My BMS shows temperature ~ -66℃ and the the runtime is also somehow strange. I will try the branch mentioned in the issue and I will let you know.

closed 2025-02-19 by mrazekfilip