patman15/BMS_BLE-HA #187: Trouble connecting using bluetooth proxy with a LithTech battery (same Bluetooth as Electronicx battery)

jc259, 2025-02-11:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have a battery from Lithtech Energy  (China company).  The battery Bluetooth uses the same name format as the Electronicx batteries.  LT-........  I am able to connect the Electronicx app to my battery  (https://play.google.com/store/apps/details?id=com.battery.electronicxlifepo4&hl=en_CA) and view all the stats the same way the original bluetooth android app does I was provided by the Chinese company.  

I can see the battery in my bluetooth proxy advertisement data

{"name":"L-12V100AH-0902","address":"02:00:07:B5:4B:1D","rssi":-87,"manufacturer_data":{},"service_data":{},"service_uuids":[],"source":"D0:EF:76:34:04:20","connectable":true,"time":1739233329.3303301,"tx_power":5}

But when I try and add the BLE battery Management integration it just says "No supported devices found via Bluetooth"

I am not sure if I need to somehow pair the battery as I don't see the battery blueetooth address showing up in the proxy log "02:00:07:B5:4B:1D"?  

I know the bluetooth proxy is working as I have a Victron battery monitor that is connecting through it to another plugin.

If I edit my bluetooth proxy device  in ESPhome in Home assistant I get this which is where I believe I need to put active: true;

substitutions:
  name: esp32-bluetooth-proxy-26792c
  friendly_name: Garage-26792c-BLE Proxy
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: m2aEMORqcQnh6HSR1W3cvdYQUb6UKANY6uhleJb=

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

bluetooth_proxy:
    active: true

patman15, 2025-02-11:
Hi! Thanks for the info, I will check against the Elektronicx battery implementation and provide an update soon.
Let's see if it is really the same.

jc259, 2025-02-11:
Let me know if you need any other logs or info. Thanks!  Btw there are a few apps for different batteries which work with these Lithtech batteries I have.  So possibly a common BMS Bluetooth module across them.

Eg Volithium - https://play.google.com/store/apps/details?id=com.bluetooth.voltium.bmsforAndroid
Electronicx - https://play.google.com/store/apps/details?id=com.battery.electronicxlifepo4

Generic app on the play store
https://litio.store/en/products/batteria-lifepo4-12v-145ah-trasparente-bluetooth-litio-store-lfp-150a-bms-1856wh-serie-ghost
https://play.google.com/store/apps/details?id=com.ble.vanomize12

Other manufacturers apks I found to work:
Meritsun Batteries
Lithtech apk from them directly

patman15, 2025-02-11:
@jc259, I fixed the detection given that the battery really uses the same protocol. Can you try the attached branch via manual install?

> The battery Bluetooth uses the same name format as the Electronicx batteries. `LT-`.....

Not sure, because your snippet `{"name":"L-12V100AH-0902"," ...` shows only `L-`...

jc259, 2025-02-11:
Grabbed a sample of the advertisement for the other batteries that are not showing up.

{"name":"LT-12V-0002\r\n","address":"05:23:01:64:00:B1","rssi":-94,"manufacturer_data":{},"service_data":{},"service_uuids":[],"source":"D0:EF:76:34:04:20","connectable":true,"time":1739317503.3669958,"tx_power":5}

jc259, 2025-02-12:
Update on the two batteries that are connecting to the BLE plugin (ie the ones that start with L-12V......... ).    I read through the closed issues for Electronicx and found that previously someone noted the battery was not displaying data in the BLE plugin if the battery wasn't charging or discharging.  Seems this might be the case with these as well.  As soon as I put a charger on the battery I am seeing consistent sensor data, instead of the sensors showing unavailable when the battery is just sitting on its own.  

Is this a result of how the plugin handles being denied a connection to the bluetooth of the BMS?  Could this be due to the BMS bluetooth limiting how often the connection can be established?  

Is there any chance the plugin could hold the most recent value instead of storing unavailable for the latest connection data?

I've attached some snippets of logs for two cases:

1) battery L-12V100AH-0901   which is showing all sensors unavailable
2) Battery  L-12V100AH-0902   which is plugged in and charging and has data for all sensors.

Not sure if I grabbed useful sections or not but figured something was better then nothing.

[901-Unavailable-LOG.TXT](https://github.com/user-attachments/files/18762530/901-Unavailable-LOG.TXT)
[902-withdataLog.txt](https://github.com/user-attachments/files/18762531/902-withdataLog.txt)

patman15, 2025-02-12:
> sensor.l_12v100ah_0902_voltage 13.2580956886548 2025-02-11T18:00:00.000Z
> sensor.l_12v100ah_0902_voltage 13.258 2025-02-11T18:47:08.853Z
> sensor.l_12v100ah_0902_voltage unavailable 2025-02-11T18:48:19.786Z

For me this, together with your low `RSSI` values hints to a bad connection. Please see the [README](https://github.com/patman15/BMS_BLE-HA#provided-information) for more information and try to enhance your BT setup.

> Are you able to add the LT- format as well?

Yes, done, branch updated. Note, with more than 3 batteries, you will run into the BT proxy connection limit!

> Is this a result of how the plugin handles being denied a connection to the bluetooth of the BMS? Could this be due to the BMS bluetooth limiting how often the connection can be established?

Since your battery is not consistently offline, I would first guess for an connection quality issue.

> Is there any chance the plugin could hold the most recent value instead of storing unavailable for the latest connection data?

That doesn't make sense, as the integration, like any other in HA, is supposed to report missing values. In normal conditions the integration tries to reestablish the connection every 30s.

Best regards,
Patrick

jc259, 2025-02-12:
HI Patrick,

Adding the new battery format LT-12......  worked great, thanks!.  I can see them all now so long as there aren't too many batteries enabled in the plugin at once as I only have one bluetooth proxy in the room where all the batteries are.    

From your previous note about signal strength -  the bluetooth proxy is right beside my battery for the testing I am doing now and I moved the proxy much closer to my wifi router.

I disabled all batteries except one and did some testing..... Immediately after plugging a charger into the battery the sensor values  show up for all sensors. And if you remove the charger the values reset back to "UNAVAILABLE" .  

I think what is happening is the BMS bluetooth used in these batteries (Lithtech, Electronix, etc)  is only allowing a poll of the data at a set interval (likely to save power) when the battery is sitting idle (ie not charging). This interval means 99% of the time the sensors show "UNAVAILABLE" since the plujgin is trying to get data it isn't allowed.   The problem is in general lots of these batteries will not be sitting in a charging state and the idea of the plugin at least for me is to be able to check on the state of charge,  monitor cell in balances and then create some automations to alert me.  

Would it be possible in your plugin for these type of batteries bluetooth  - to not update values with UNAVAILABLE when it can't get data?  

Another thing I thought of was could the polling interval be a config in the plugin per battery so that we could reduce how often the plugin polls for the data in general?

Thanks again for all this work, the plug in is a great tool.

patman15, 2025-02-12:
Is the app able to connect when the BMS is idling? If so, can you get me a BT communication recording of that?
For preserving the values, have a look at the post in the community forum about how to [preserve values of a sensor](https://community.home-assistant.io/t/preserve-last-value-of-sensor-before-becoming-unavailable/469875/4).

jc259, 2025-02-13:
I did some more testing - added some notes in a text file I attached and a log of just the BLE plugin debug data.  It seems the battery is showing connected but not getting data a short period after the connection to home assistant is made.

What could I use to capture the bluetooth logs between the Android phone and the battery when it's sitting idle with the app open?  To see if the app is taking updates or not?

[home-assistant_2025-02-13T01-29-47.553Z.log](https://github.com/user-attachments/files/18776222/home-assistant_2025-02-13T01-29-47.553Z.log)
[Testing info.txt](https://github.com/user-attachments/files/18776223/Testing.info.txt)

patman15, 2025-02-13:
I can confirm what you said. Obviously, the BT connection is still alive, but the BMS seems to go to sleep mode. A reconnect seems to solve the issue. Maybe there is some sort of keep-awake command that is missing. Let's see from the app, if possible.
Please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery. Please ensure that it went to sleep ...
I'd like to check if the app reconnects or has some special command to keep alive.

Thanks,
Patrick

jc259, 2025-02-13:
Hi Patrick, I grabbed a log with nRF connect.  Not sure if is helpful or not. I have attached it.   This was with the app sitting open connected to the battery.  I can't really tell if it just captured the first read and then held it.  I can do another test with a charger connected if you want.

[Log 2025-02-12 20_31_13.txt](https://github.com/user-attachments/files/18790344/Log.2025-02-12.20_31_13.txt)

jc259, 2025-02-13:
Hi, Ran one more test where i:
-connected the app to the battery
-let the app sit open
-flipped through the pages in the app
-connected charger
-let the app sit open
-disconnected charger

I attached the log and an summary of what I did with time stamps.

[2025-12-13-testing.txt](https://github.com/user-attachments/files/18791049/2025-12-13-testing.txt)
[Log 2025-02-13 14_51_20.txt](https://github.com/user-attachments/files/18791050/Log.2025-02-13.14_51_20.txt)

patman15, 2025-02-14:
Hi! I looked at the logs but there is nothing special in it, but I'm not sure the nRF connect captured all necessary information, especially at the beginning of the communication. Can you try it using these methods? [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/)
Otherwise I have one suspect, which I fixed in the [branch](https://github.com/patman15/BMS_BLE-HA/tree/187-trouble-connecting-using-bluetooth-proxy-with-a-lithtech-battery-same-bluetooth-as-electronicx-battery). Can you see if that helps?

jc259, 2025-02-14:
No luck with the latest change.  It still goes to showing UNAVAILABLE after about 5 mins.   I didn't notice you put links previously, I'll try that method to capture the bluetooth communication between the phone app and the battery and update you.

jc259, 2025-03-27:
Sorry I've been swamped with work. Still intend to grab more logs for you at some point here!

jc259, 2025-04-29:
I've updated to the latest and still seeing most of the batteries going into an unknown state. I am still planning to get you logs, just haven't got to it yet.

closed 2025-06-05 by patman15