patman15/BMS_BLE-HA #189: Seplos v2 loosing protocol sync due to BT adapter messages

patman15, 2025-02-13:
> [home-assistant_bms_ble_2025-02-13T10-56-41.481Z.log](https://github.com/user-attachments/files/18782694/home-assistant_bms_ble_2025-02-13T10-56-41.481Z.log) 

 _Originally posted by @bogi9999 in [#171](https://github.com/patman15/BMS_BLE-HA/issues/171#issuecomment-2656234180)_

Data disappears and comes back

patman15, 2025-02-14:
I checked the log, the problem is basically, that the BT adapter of the BMS messes up the BMS messages with it's own messages (`AT+...`). That results in corrupted data which is discarded. The only way to fix that is to disable the BT adapter messages. Not sure this can be done and how.
Does the app work 24/7? 
Can you capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while starting up the Seplos app with the battery? Maybe I can find some command disabling the `AT` commands.

bogi9999, 2025-02-17:
Sorry, i don't know if the app work 24/7, because i do not sitt 24/7 in front of the battery. ;)
If i can capture, i first have to check if there is a tool for that available.

patman15, 2025-02-17:
> If i can capture, i first have to check if there is a tool for that available.

I linked you two for [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) and [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) :wink:

idanny50, 2025-02-18:
as I know seplos v2, when in packed, bp00 it is master, and from there you access slaves units. When i connect with phone I only see bp00 pack, I connect to it and then I can change to slave packs.

patman15, 2025-02-18:
How does your statement relate to this issue?

bogi9999, 2025-02-18:
Hope this is what you need.

[SeplosV2.log](https://github.com/user-attachments/files/18850187/SeplosV2.log)

patman15, 2025-02-18:
Thanks, checked. The `AT+` replies are also in there. I could see incorrect replies, although the command was not in the middle of some data. I guess with a longer recording we could see that as well.
I'm not 100% sure but I fear the behavior is simply bad design of the BT setup and unavoidable.

maps0n, 2025-03-14:
> I'm not 100% sure but I fear the behavior is simply bad design of the BT setup and unavoidable.

So, since there will never be a new release by SEPLOS, we have to live with it.

One question: 
Is there a measure to restart the communication and would this produce valid data for a short while?
This would be sufficient to monitor the cell voltages, e.g. every minute.

patman15, 2025-03-15:
I would assume that the communication recovers automatically after some retries, i.e. that can take 10-20 minutes. If you feel it is a permanent issue, please check the 24h history of e.g. the voltage sensor. If it really stays `unavailable` please provide me with a debug log. Depending on the observed behaviour I might do some automated reset then

closed 2025-04-25 by github-actions[bot]