patman15/BMS_BLE-HA #190: Bluetooth taking a long time to connect.

tommysmsw, 2025-02-14:
I am not sure this is an issue with this. But I figured I would just mention it. When I restart HA it takes hours for one of my two Eco-worthy batteries to reconnect. It seems to happen every time and always just to the one battery.
I am pretty sure it is not a problem with the battery Bluetooth as I never had a problem connecting to my phone. But it could be a problem with the Bluetooth in my RPi? 
It's also the battery that is closer to the Pi so I figure that distance is not the issue. Plus the Pi is in the same cabinet as the batteries and maybe 2-3 feet away? 

Feel free to ignore this if I am the only mention of this. It could just be the Pi.

patman15, 2025-02-15:
Hi!

Being close is not necessary a good indicator for good connections. Also, the [RPi is not known for handling a number of devices in parallel well](https://www.home-assistant.io/integrations/bluetooth/#broadcom-bcm-based-adapters). You might want to try a USB adapter (see also the previous link) to get it better.

For the start, please check the output of `bluetoothctl info <MAC>` whether your device appears there after restart. If it is not there, it is definitely an issue with your BT setup. On HA 2025.02 and later you can also go to the [bluetooth integration](https://my.home-assistant.io/redirect/integration/?domain=bluetooth). On your BT adapter select `configure->advertisement monitor` and check it there.

Please also enable the `RSSI` sensor and monitor that one (post a graph).
My best tip would be a performance issue with the RPi internal adapter.

Best regards,
Patrick

tommysmsw, 2025-02-17:
So I did not write becak because when I updated my HA version from (I think) 2025.1.4 to 2025.2.4 it seemed to fix the issue. I did several restarts and both batteries immediately appeared. However, when I updated this add-on from 1.11.7 to 1.12 the issue came back. I tried to do a little investigation before reaching out to you but when I install 1.11.7 via "HACS" it actually installs 1.11.6 even though HACKS lists it as 17, the addon itself says 16. And I have the same issue. 

I wish I could give you more info (maybe it was working just because? and it had nothing to do with updating HA and maybe 1.12 did nothing to break it). I just figured I would mention it. At the very least maybe you know why HACS is giving me 16 when I ask for 17 and I can get 17 and report back again.

tommysmsw, 2025-02-17:
I forgot to mention that in 1.12 neither batter shows up. It's only been a few hours, but when I go back to 1.11.6, one battery shows immediately. So with 1.12 either they are now both taking a long time, or they are not showing at all. I have switched from 1.11.6 to 1.12 a few times to make sure I get the same results.

patman15, 2025-02-18:
Please consider my suggestions from https://github.com/patman15/BMS_BLE-HA/issues/190#issuecomment-2660798960 as I highly doubt this is related to the integration but to your BT setup. I need more info in order to help you. Also, a debug log of the integration would be helpful.

tommysmsw, 2025-02-18:
Is there just anyway you can get me 1.11.7 again? I know that one works fine. I switched from 1.11.6 to 1.12 many times yesterday and I get the exact same results when I switch. I feel like if it were my Bluetooth, I would get the same results no matter what version I use. No? In 1.11.6 I get one to work immediately, the other takes like 14 hours. In 1.11.7 they both work immediately and in 1.12 neither work. At least not for the 4-5 hours I waited. I am not sure how that could not be related to the version? Although you know way more about this than I do. For now, I would just love to be able to reinstall 1.11.7 again. Is there any way I can get that done? As I said, HACS installs 1.11.6 when you choose 1.11.7.

patman15, 2025-02-18:
> Is there just anyway you can get me 1.11.7 again?

Sure, just select if from HACS, I think I made a mistake and forgot to update the version number in the code, so it shows 1.11.6 when installed. Additionally, all the [release](https://github.com/patman15/BMS_BLE-HA/releases) are still available and downloadable. You can [manually install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) them.

>  I am not sure how that could not be related to the version?

Well, I'm pretty sure it is some startup timing issue on the BT or interference with other integrations. I did not change the code on the integration for your battery so I'm convinced it is another issue and I have seen such effects already. My prediction is that some random update from some other integration or HA will change the behavior again.
If you are happy, just close the issue, we can follow-up anytime.

closed 2025-02-19 by tommysmsw