patman15/BMS_BLE-HA #191: Ultimatron Battery

andrewjsmith76, 2025-02-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hi!

I have a LiFePo4 battery from Ultimatron (https://ultimatron-france.fr/categorie-produit/batterie-au-lithium/). It is equipped with BLE and I can connect to it and read out its properties using the accompanying app. Now, I'd like to monitor the battery status in my Home Assistant.

After disconnecting the app on my phone, I performed the steps described in the troubleshooting section. Running `bluetoothctl devices` in the terminal yields a bluetooth MAC address that I could match to the one in the app.

Running `bluetoothctl connect <MAC>` yielded a "connection successful" message (or something of that nature).

However, even after waiting a while, I couldn't find any entities in Home Assistant that belong to the battery. From the examples, I gather that the entities should contain "smartbat" ... searching for this word in the Developer Tools' "Current entity states" doesn't yield any results.

Am I doing something wrong?

Here's the output from the "advertisement monitor" in HA:
`{"name":"12200A220900102","address":"A4:C1:37:30:F9:EB","rssi":-66,"manufacturer_data":{"63979":"3037c1a4"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","0000fa00-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb"],"source":"DC:A6:32:07:55:72","connectable":true,"time":1739616953.9479675,"tx_power":null}`

Any help in this matter would be greatly appreciated!

Best regards,
Andrew

patman15, 2025-02-16:
Hi @andrewjsmith76! Can you try the main branch or wait for the next release? I think it is added there.

andrewjsmith76, 2025-02-17:
Hello @patman15,

Thank you for getting back to me... I'm a semi-noob in regard to Github... Have I understood correctly, that I need to simply re-download the current version from within HACS? Don't I get the main branch when I install this via HACS?

Sorry if I'm being dense...

Best regards,
Andrew

patman15, 2025-02-17:
Well, there is a README that explains how to manually download and update, but meanwhile I released 1.12.0, so you can simply try that one.
No, you don't get the main branch by default y you get the latest release. :smile:

andrewjsmith76, 2025-02-23:
Hello @patman15,

Thanks! I finally got around to trying this out. After updating to 1.12.0 and restarting HA, the battery was autodiscovered!

Thanks for your help!

Best regards,
Andrew

patman15, 2025-02-23:
Thanks for confirming!

closed 2025-02-23 by patman15