patman15/BMS_BLE-HA #194: Support TopBand Batteries

Fandu21, 2025-02-17:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Thank you for this amazing add-on!!  I am currently running version 1.12.0 and my TopBand BMS does not appear to currently be supported and I'm hoping that it can be added.

### Additional context

>[core-ssh ~]$ bluetoothctl info FC:45:C3:BC:D6:A8
Unable to open mgmt_socket
Device FC:45:C3:BC:D6:A8 (public)
        Name: ZM20210512010036
        Alias: ZM20210512010036
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (f000ffc0-0451-4000-b000-000000000000)
        Modalias: bluetooth:v000Dp0000d0110
        ManufacturerData Key: 0x0000 (0)
        ManufacturerData Value:
  fc 45 c3 bc d6 a8                                .E....
        RSSI: 0xffffffd0 (-48)
        AdvertisingFlags:
  05                                               .
[core-ssh ~]$`

[Link to TBEnergy IOS App](https://apps.apple.com/gb/app/tbenergy/id985141488)

patman15, 2025-02-18:
Hi @Fandu21! Looks like an Ective BMS, so implementation exists, although I see some differences. Can you try the [development branch](https://github.com/patman15/BMS_BLE-HA/tree/194-support-topband-batteries) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) and send me a debug log?

patman15, 2025-02-20:
Could it be that Bluetooth is turned off by the battery, when there is no current?

For the logs, if you are at integration, you can turn debug logs off and it will prompt you with a download then.

Fandu21, 2025-02-21:
I'm still able to connect to the battery using the ios app so I don't think it's an issue of the BMS turning off or anything.  So I did a little more digging and found something interesting.  While initially setting up the integration, bluetoothctl produced the following output:

>`[core-ssh ~]$ bluetoothctl devices
Unable to open mgmt_socket
Device CC:20:26:42:3C:78 CC-20-26-42-3C-78
Device D3:73:D6:5B:19:CA D3-73-D6-5B-19-CA
Device 34:13:43:1E:88:FD telink_mesh1
Device 75:76:B7:81:C4:BA 75-76-B7-81-C4-BA
Device 0A:38:B2:97:CF:CB 0A-38-B2-97-CF-CB
Device C0:C3:9A:B4:A8:8F C0-C3-9A-B4-A8-8F
Device 78:F7:A3:40:72:0D 78-F7-A3-40-72-0D
Device 49:F0:88:91:3C:73 49-F0-88-91-3C-73
Device 4C:A7:BE:26:24:F6 4C-A7-BE-26-24-F6
Device 4D:7F:52:59:6B:BD 4D-7F-52-59-6B-BD
Device 6B:37:AC:6E:40:B1 6B-37-AC-6E-40-B1
Device 70:45:1D:79:B2:92 70-45-1D-79-B2-92
Device 0C:E2:82:4A:6B:E7 0C-E2-82-4A-6B-E7
Device 66:34:67:1D:87:93 66-34-67-1D-87-93
Device F3:5C:11:F3:6C:C8 F3-5C-11-F3-6C-C8
Device C6:B2:63:82:E8:63 C6-B2-63-82-E8-63
Device D2:35:3D:B7:CB:8E D2-35-3D-B7-CB-8E
Device DF:04:5A:0A:9D:EB DF-04-5A-0A-9D-EB
Device E3:CA:68:91:73:A8 E3-CA-68-91-73-A8
Device 0B:08:A9:80:92:E3 0B-08-A9-80-92-E3
Device 25:98:05:27:A8:2E 25-98-05-27-A8-2E
Device 7C:51:BD:6C:38:60 7C-51-BD-6C-38-60
Device 40:17:60:14:D4:39 40-17-60-14-D4-39
Device 40:2A:01:E0:FF:B3 40-2A-01-E0-FF-B3
Device 4C:D5:85:F5:C9:8E 4C-D5-85-F5-C9-8E
Device 62:C1:B8:96:CB:2D 62-C1-B8-96-CB-2D
Device 47:AB:B7:04:08:3A 47-AB-B7-04-08-3A
Device 43:4F:92:10:1C:C7 43-4F-92-10-1C-C7
Device 65:77:66:9C:7F:17 65-77-66-9C-7F-17
Device FC:45:C3:BC:D6:A8 ZM20210512010036
Device 5E:56:52:A6:61:8F 5E-56-52-A6-61-8F
Device 46:F5:31:4B:3E:C6 46-F5-31-4B-3E-C6
Device 5D:1A:02:82:BF:A8 5D-1A-02-82-BF-A8
Device 34:13:43:1E:CF:F4 1FA178DE9D0E1212
Device B8:3A:9D:81:4B:57 ALAM (81:4B:56)
Device 5B:DB:DB:9F:D9:ED 5B-DB-DB-9F-D9-ED
Device 44:84:84:68:23:30 44-84-84-68-23-30
Device 44:C6:5D:6B:78:22 44-C6-5D-6B-78-22
Device 28:E6:A9:46:BF:13 65" OLED
Device 88:4A:EA:36:8B:AD Leviton DDMX1 V5.1
Device 60:A7:3C:E6:FB:BD 60-A7-3C-E6-FB-BD
Device 08:66:98:9F:C6:C5 08-66-98-9F-C6-C5
Device 50:8B:B9:EB:FB:97 TY`

However, now the output is quite different:
>`[core-ssh bms_ble]$ bluetoothctl devices
Unable to open mgmt_socket
Device 34:13:43:1D:7C:72 1FA178DE9D0E1212
Device 6E:BC:3B:92:7C:6F 6E-BC-3B-92-7C-6F
Device 18:87:4E:38:AE:E3 18-87-4E-38-AE-E3
Device 6B:AC:A8:5B:46:82 6B-AC-A8-5B-46-82
Device 56:C8:17:A1:A0:76 56-C8-17-A1-A0-76
Device 11:EE:A7:D2:E5:08 11-EE-A7-D2-E5-08
Device 5E:12:CD:6D:22:79 5E-12-CD-6D-22-79
Device 63:5B:37:F1:C7:7A 63-5B-37-F1-C7-7A
Device 44:C6:5D:6B:78:22 44-C6-5D-6B-78-22
Device 7B:91:D0:99:4C:A5 7B-91-D0-99-4C-A5
Device 4C:BD:31:05:58:00 4C-BD-31-05-58-00
Device 77:AE:79:9C:6C:59 77-AE-79-9C-6C-59
Device 7C:73:1B:C2:0B:30 7C-73-1B-C2-0B-30
Device 7E:A5:A9:E3:6E:27 7E-A5-A9-E3-6E-27
Device 88:4A:EA:36:8B:AD Leviton DDMX1 V5.1
Device 50:8B:B9:EB:FB:97 TY
Device 08:66:98:9F:C6:C5 08-66-98-9F-C6-C5
Device 60:A7:3C:E6:FB:BD 60-A7-3C-E6-FB-BD
Device 4F:2A:A9:B5:0C:50 4F-2A-A9-B5-0C-50`

What finally worked was removing the integration, restarting HA, and then re-installing the integration.  The output from bluetoothctl has returned to the expected.  With things functioning again, I enabled logging, left things to setting for a few minutes, then moved the battery out of BLE signal range, waiting a few minutes, and then brought the battery back in range.

Log [here](https://www.dropbox.com/scl/fi/88mzgwd0nsythfdg2n1p8/home-assistant_bms_ble_2025-02-21T00-58-49.027Z.log?rlkey=e2nva9sxw67owt0m95ulrtcse&st=w8aj74w0&dl=0).  I had a gander through it and didn't see anything too insightful, let me know if I can provide anything else that might help.

patman15, 2025-02-21:
I'm not sure how you ended up. The log contains a lot of errors and it doesn't look like it is working. Are you saying that you now have working setup? Can you get me a log of that one as well?

Fandu21, 2025-02-25:
With the [updated error checking](https://github.com/patman15/BMS_BLE-HA/pull/200), I'm no longer getting any data from the battery at all.  [Log here](https://www.dropbox.com/scl/fi/pv6d57scx6xs47p3dl77z/home-assistant_bms_ble_2025-02-25T03-01-34.120Z.log?rlkey=n7848kc9ishw1jcruiouxfnb8&st=511kofip&dl=0)

patman15, 2025-02-25:
Ok, I checked the log again, the parsing is correct. A lot of the received data is garbage, too short, or has incorrect checksum. By chance, are you using the Raspberry Pi internal BT? I do not see anything I could fix more from coding point of view. So I'll merge that code soon. Let me know if you need more help or if it is ok like this.

Fandu21, 2025-02-26:
Yes I'm using the built-in BT hardware on the RPi4b and I'm using the USB3 ports so that may explain some of the checksum issues.

I left the battery overnight and I'm getting valid data every 30 seconds.  That's fine for my needs.  However there does seem to be an issue with the connection recovering after an interruption.  I intentionally moved the battery out of BT range and then brought it back within range, expecting the connection to automatically re-establish and data to resume, but that's not the behavior I'm seeing.

After bringing the battery back within BT range, the debug logs show packets being received but all data is invalid.  [Log here](https://www.dropbox.com/scl/fi/e6mj7zw9y2yfacrelluol/home-assistant_bms_ble_2025-02-26T02-51-13.619Z.log?rlkey=knmbuzynwjjawxg91wf9b10i8&st=a0t0tmv0&dl=0)

Restarting HA does not fix the issue, the only way I can coax things to work is by removing the BLE integration and then re-installing it.

patman15, 2025-02-26:
> After bringing the battery back within BT range, the debug logs show packets being received but all data is invalid.

Ok, I see a chance to fix that in the protocol parsing (thanks for testing), but the basic issue with the RPi messing up data will persist. For that I strongly recommend to take a look at [proper adapters](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters), e.g. I took this [LogiLink](https://www.amazon.de/dp/B093C98PFF) for no money (no I do not get anything commission 😉)
I'll provide you an update that should further enhance robustness.

patman15, 2025-02-26:
@Fandu21 I pushed a small fix that should bring the battery back in cases like you tested yesterday. Btw. I assume that you just kept the battery to short out of range before the connection was considered to be lost, otherwise it would have resynced. Anyway thanks for pointing it out, now it should even cover that case. 😃

Fandu21, 2025-02-27:
Tested and the connection now re-establishes itself automatically.  Awesome!

closed 2025-02-27 by Fandu21