patman15/BMS_BLE-HA #195: Wattcycle Smart Battery BMS

Trst23, 2025-02-19:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

The BLE BMS is recognized but I always get a error like this:

[image]

The BLE Proxy Device info:

{"name":"XDZN_001_7A01","address":"C0:D6:3C:4F:7A:01","rssi":-81,"manufacturer_data":{"54976":"3c4f7a01ff"},"service_data":{},"service_uuids":[],"source":"5C:01:3B:6A:1C:AC","connectable":true,"time":1739976609.7685628,"tx_power":null}

I would be happy if this type can be integrated.

kind regards

patman15, 2025-02-19:
This is an issue with your BT setup. Your RSSI is pretty low, see README. Try to get the battery a bit closer and see if you can get read of any disturbances on the wireless channel. The battery was obviously detected and the type should work.

Trst23, 2025-02-19:
Thank you for the fast response.
You are right. I am wondering because the battery was only some meters away.
But after you response I put it 1m next to the battery and it was recognized.

Thanks a lot.

closed 2025-02-19 by Trst23