patman15/BMS_BLE-HA #197: Creabest battery not detected

patman15, 2025-02-19:
> Hi @patman15 
> I also have the same problem with another battery (Creabest).
> {"name":"100R0002E3","address":"BA:03:EC:11:09:09","rssi":-76,"manufacturer_data":{"21330":"2134ba03ec110909010500010000"},"service_data":{},"service_uuids":["000003c1-0000-1000-8000-00805f9b34fb"],"source":"2C:CF:67:AB:82:92","connectable":true,"time":1739968077.0545566,"tx_power":0}
> An another one with JK-BMS:
> {"name":"40716310726","address":"C8:47:80:1E:32:67","rssi":-44,"manufacturer_data":{"2917":"88a0c847801e3267"},"service_data":{},"service_uuids":["0000ffe0-0000-1000-8000-00805f9b34fb","0000fee7-0000-1000-8000-00805f9b34fb"],"source":"2C:CF:67:AB:82:92","connectable":true,"time":1739968077.1905568,"tx_power":null}
> 
> The message 'No supported devices found when try to add new integration.' comes up when trying to add the "integration".
> It is a "Homeassistant Core" installation:
> https://www.home-assistant.io/installation/linux#install-home-assistant-core
> 
> Core 2025.2.4
> BMS BLE Commit a96a425 

 _Originally posted by @Gruni22 in [#184](https://github.com/patman15/BMS_BLE-HA/issues/184#issuecomment-2668533334)_

@Gruni22 I carved out your request to be able to work on it. 😃

Gruni22, 2025-02-19:
@patman15 Thanks.
If you need any further details, please let me know.
I switched back to HAOS for now.
Nevertheless, the Creabest is still missing there.w

patman15, 2025-02-19:
Can you try this [branch](https://github.com/patman15/BMS_BLE-HA/tree/197-creabest-battery-not-detected) via [manual install](https://github.com/patman15/BMS_BLE-HA#manual)? Note, you need to have HA2025.2.2+. It should fix the Creabest issue.

patman15, 2025-02-19:
Hard to tell, but most likely an issue in the BT setup. Could be an issue with the bluez version/config, the bleak library, Linux Kernel, ... this is really hard to debug or figure out, especially remotely. I have a number of setups and one is significantly slower in detecting devices than HAOS. No idea why. :wink:

Gruni22, 2025-02-19:
bluez is a good factor. I compiled manually the latest version with BLE support (activated manually).
If I can get running HAOS without internet connection, I am happy how it is.
And I can understand, that you do not want put effort on it. Nevertheless, in the supervised config it should also work.
(I hate all that limited docker thingy)

patman15, 2025-02-20:
@Gruni22 I updated to HA2025.2.2, can I merge the update or are there any open issues?

Gruni22, 2025-02-20:
Not related to the Battery Plugin itself. You can merge it. 

Thanks!

closed 2025-02-20 by patman15