patman15/BMS_BLE-HA #203: i have a jbd bms that is not getting added automatically

yokonetworks, 2025-02-24:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

i see it in advertisement monitor (Bluetooth proxy via esp) 
how can i add this manually? becuase its proxy it doesnt seem to allow me to use bluetoothctl pair commands in temrinal

{"name":"RV BMS","address":"A4:C1:38:BE:6D:7C","rssi":-35,"manufacturer_data":{"28028":"be38c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"5C:01:3B:94:AA:F8","connectable":true,"time":1740354298.8764884,"tx_power":null}

thanks!

imbdb, 2025-02-24:
@yokonetworks The device name need to be in a particular pattern to be detected.
Here are patterns for JBD
```
 "SP0?S*",
  "SP1?S*",
  "SP2?S*",
  "GJ-*",  # accurat batteries
  "SX1*",  # Supervolt v3
  "DP04S*", # ECO-WORTHY, DCHOUSE
  "121?0*",  # Eleksol, Ultimatron
  "12200*",
  "12300*",
```

These patterns are the model name of your BMS, I also had setup a custom name but then I renamed my device to the model name and it got detected (Had to restart Homeassistant once).

patman15, 2025-02-24:
Hi @yokonetworks, @imbdb is correct. Unfortunately, so far all manufacturers that use JBD BMS do not manage to get the `manufacturer_data` correct. So I'm left with auto detection via name pattern. 😞  You can rename the BMS in HA though. Let me know if a default pattern is missing, then I can add it.

imbdb, 2025-02-24:
@patman15 I am not sure about this, Can we use mac addresses for detection?
I have JBD BMS, @yokonetworks also has one, We can determine if the BMSes use the same manufacturer or not.

patman15, 2025-02-24:
@imbdb the problem is that JBD or their users put the MAC address into the `manufacturer_id`. Unfortunately, they do it reverse so the manufacturer specific part of the MAC is not in the first 16bit of the data, which is defined by BT as the `manufacturer_id`. Consequently, each device has its own ID. :-(
The BT filtering does not allow me to use the native address, so also this option is gone.

imbdb, 2025-02-24:
@patman15 
I am new to Home Assistant addons, I found the below method in documentation that might help us get list all the Bluetooth devices .
https://developers.home-assistant.io/docs/core/bluetooth/api/#fetching-all-discovered-devices

Just asking you if we have already tried this, if not, I can try to raise a PR with this functionality.

patman15, 2025-02-24:
Hi! Yeah, I'm aware of that functionality, but due to the evolution of the project I have not considered it so far. Unfortunately, it is only half the solution, most likely it needs further additions, i.e. a shared scanner instance, which means a major rework of the auto-detection. Not sure it's worth the effort since other people did not spend a bit of time implementing a specification reasonably...
Also, all the phone apps must have the same issue.
If you want to give it a try, happy to receive a PR, but please consider the test coverage as well! Functionality is maximum half the effort needed.

imbdb, 2025-03-06:
@patman15 That makes sense, I don't think its such a large issue.

closed 2025-04-16 by github-actions[bot]