patman15/BMS_BLE-HA #204: Support for Shelly Bluetooth Gateway Proxy

kaffeetas, 2025-02-24:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Is there a way to use the shelly Bluetooth functionally instead of esphome bluetooth proxy?
Have multipble Shellys (plus1PM, miniPMG3) spread in my house and pretty close my batteries with daly bms.
Shelly & Passive BLE works great with some BTHome devices.
Set a Shelly to active mode (in HA) and i can see the Daly BMS in the logs on the shelly:
  {
    "ts": 1740423233.003,
    "level": 2,
    "data": "shelly_notification:208 Event from script:1: {\"component\":\"script:1\",\"id\":1,\"event\":\"ble.scan_result\",\"data\":[2,[[\"40:16:09:01:43:01\",-76,\"AgEFBP8CAQQ=\",\"\"]]],\"ts\":1740423232.96}",
    "fd": 1
  },

however BMS does not appear on HA side. I'm unsure if its just not configured correctly or if functionality is missing.

patman15, 2025-02-24:
If you are using HA 2025.2+, can you please post the information that is given by the Bluetooth advertisement monitor? Actually it should also work flawlessly with Shelly proxies.

kaffeetas, 2025-02-24:
[image]
Did a quick update - that bluetooth monitor is a great improvement :-)

patman15, 2025-02-25:
Can you please click the device(s), copy the data, and paste it here? The table does not contain everything.
Btw. did you check your device is not listed among the disabled or hidden ones?

kaffeetas, 2025-02-25:
Here is the content of one of the BMS:
{"name":"DL-401609014301","address":"40:16:09:01:43:01","rssi":-79,"manufacturer_data":{"258":"04"},"service_data":{},"service_uuids":[],"source":"E4:65:B8:F1:9D:5C","connectable":false,"time":1740472441.337187,"tx_power":null}

kaffeetas, 2025-02-25:
I use 16S LiFePo 250A BMS from Daly.

figured also out that "active" with shellys means "acitvely" asking devices to send data but no possibilty to communicate back to the device :-(.
I did set up a ESP32 as Bluetooth proxy and now details of bluetooth advertisement monitor says:

{"name":"DL-40160901534C","address":"40:16:09:01:53:4C","rssi":-87,"manufacturer_data":{"258":"04"},"service_data":{},"service_uuids":[],"source":"64:E8:33:8D:93:94","connectable":true,"time":1740482255.341024,"tx_power":null}

Thanks for your great support by the way :-)

kaffeetas, 2025-02-25:
thanks a lot - this worked like a charme! It found the BMS right after homeassistant started in the devices section and all data seems correct so far, however i did not thest the fancy stuff like conrolling the FETs so far ;-). How do you want me to send you the debug log? You can contact me with my nick @gmail.com

patman15, 2025-02-26:
Hi! You can just attach the file to any comment here, but actually, since you said it works like a charm, I think I'm good. I will soon merge the changes for the next release which will close this ticket automatically. Feel free to open a new one in case you experience any troubles.

patman15, 2025-02-26:
> however i did not thest the fancy stuff like conrolling the FETs so far

It's simply not there and currently I have no plans to implement it. See the big [*caution section*](https://github.com/patman15/BMS_BLE-HA#provided-information) for reasoning.

closed 2025-02-28 by patman15

Tomblarom, 2025-07-11:
> EDIT: bad news, according to [this thread](https://shelly-forum.com/thread/18876-bluetooth-gateway-active-connection/) Shelly proxies do not support active connections. So you will need another BT adapter to get it working.

That's a shame. I also do have multiple Shelly devices. Do you know, if this is a hardware limitation or a software limitation, that could potentially be cleared through an update from Shelly? I briefly gasped over the linked articles, but was not able to find a definite answer.

patman15, 2025-07-12:
@Tomblarom I think it is a hardware (memory) limitation as active connections require quite some memory. At least that is my conclusion from the [big red warning box at ESPhome](https://esphome.io/components/bluetooth_proxy.html).