patman15/BMS_BLE-HA #211: JK BMS: Wrong temperature value in 1.13.0

SkeLLLa, 2025-03-03:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

JK BMS

### Describe the issue

After update a few attributes were added in temperature sensor with values `0`. Those `0` value I guess represent that the sensor actually is not connected.
And temperature sensor itself displays "average" temperature among all of them.

Therefore as result the average temperature dropped and shows non representative value.

### Reproduction steps

```text
1. Update integration to 1.13.0 
2. Restart HA
3. See updated sensor and it's attributes
```

### Debug logs

Might be related to this PR: https://github.com/patman15/BMS_BLE-HA/pull/196

patman15, 2025-03-03:
Hmm, can you please get me a debug log? This temperature sensor thing is really annoying as there seems to be no consistent concept over the versions. Some BMSs report -200 when not connected others obviously 0. I need to check some details in the log on how to fix that. Thanks!

SkeLLLa, 2025-03-03:
[home-assistant_bms_ble_2025-03-03T16-08-23.625Z.log](https://github.com/user-attachments/files/19056152/home-assistant_bms_ble_2025-03-03T16-08-23.625Z.log)

Yes, I was just in process of removing non-related data from it.

patman15, 2025-03-03:
Thanks! Yes that line and the actual frames. Can please also let me know the HW/SW revision of your BMS so I can correlate it with the data sent? You should find it in the log after first connect.

SkeLLLa, 2025-03-03:
It's easier to find it in previous issue :): https://github.com/patman15/BMS_BLE-HA/issues/90#issuecomment-2486711686

SkeLLLa, 2025-03-03:
No worries. Let me know if you need anything else.

patman15, 2025-03-03:
Found out that the sensors are different for SW version <11 vs 11-14 vs >14. 🤔 
@SkeLLLa can you try the linked branch?

> Might be it worth showing not average

But still it would be wrong. None of the options is perfect, so I better keep the current one and fix it.

SkeLLLa, 2025-03-04:
@patman15 wrote you back in PR.

patman15, 2025-03-04:
@SkeLLLa thanks for testing, will merge.

closed 2025-03-04 by patman15

SkeLLLa, 2025-03-04:
Thank you.

Regarding those "avg" vs "max/median" - usually if you displaying some analytical function on a graph average value is the least meaningful one, since it doesn't give you an idea what's going on with measure comparing to other metrics. Median is good when you have more values e.g. 10+. But max is more convenient or even 2 sensors like max and min are useful. Since users may want to prevent overheating using automations as well as do low temperature protection (if for some reason values in BMS are set not correctly).

patman15, 2025-03-04:
Totally agreed, but since everybody chooses a different style there is always the option to use [templates](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#i-want-to-know-the-maximum-cell-voltage). There you can choose which sensors to use and also the function you want to apply. 😃 Since the average uses all sensors, it also ignores the fact that some are for cells, others for the BMS, or even the ambient temperature. There are just too many variants to make a good general solution.