patman15/BMS_BLE-HA #214: seplos v3 SP47xxx No supported devices found via Bluetooth

vinzeno, 2025-03-05:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

hello, I can't add my bms seplos 3.0
It works very well by bluetooth and it is identified under the name SP47xxx in the peripherals. Do you have a track please

vinzeno, 2025-03-05:
[image]

vinzeno, 2025-03-05:
Hello Patrick, thank you for your feedback.
What more do you want, is that okay with you?

[image]
[image]

vinzeno, 2025-03-05:
It's last firmware

[image]

patman15, 2025-03-05:
@vinzeno I made a modification, can you please try the branch linked to this ticket via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual)? I'm not sure it will work, as the advertisement looks strange and I made it according to all other Seplos batteries. But let's see.
Please provide a short debug log in case it recognizes the battery. Thanks!

vinzeno, 2025-03-05:
I love you. It's perfect and all the value are good. Thank you very much.

patman15, 2025-03-05:
😊 Ok, I will merge it for the next version. Just keep the patch till then.

vinzeno, 2025-03-06:
I have never exported the logs, I have enabled the collection in the plugin but I don't know how to extract it, if you can help me with that?

patman15, 2025-03-06:
You just need to stop the collection and it will ask you where to save.

sylvaingarciaha, 2025-03-06:
ok thank , but it'ss to big and the zip file don't pass

vinzeno, 2025-03-06:
ok thank , but it'ss to big and the zip file don't pass

patman15, 2025-03-06:
Just start recording again and stop after only a few minutes. Then it will be smaller.

vinzeno, 2025-03-09:
[home-assistant_bms_ble_2025-03-09T10-20-08.483Z.log](https://github.com/user-attachments/files/19148898/home-assistant_bms_ble_2025-03-09T10-20-08.483Z.log)

patman15, 2025-03-09:
@vinzeno  Your log is flooded with SSL errors and did not contain anything from the integration. But never mind, I'm good.

vinzeno, 2025-03-23:
Hello, after the update 1.14.0n the plugin don't ron correctly. Impossible to see the bms. I have test to delete et re add the bms but it d'ont work. The bms ron correcty with the apps bluetooth seplos.

[image]

[home-assistant_bms_ble_2025-03-23T07-28-40.313Z.log](https://github.com/user-attachments/files/19407617/home-assistant_bms_ble_2025-03-23T07-28-40.313Z.log)

vinzeno, 2025-03-23:
[image]

patman15, 2025-03-25:
Your Bluetooth system is overloaded. Check the BT monitor of your adapter or add additional ones.

vinzeno, 2025-03-25:
Thanks for your feedback.
I have a good Sena Technologies controller, though.
It started working after 2 hours, and everything works now.

patman15, 2025-03-25:
> It started working after 2 hours, and everything works now.

That does not sound right, there is for sure still another Bluetooth issue. But if it works, just remember it in case you have other troubles. :wink: