patman15/BMS_BLE-HA #216: SEPLOS BMS 10E (V2) not detected

patman15, 2025-03-08:
> I have two Mason SEPLOS Battery packs, each with SEPLOS BMS 10E (V2) and BT Modul.
> The Mastepack (BP0) is connected by CAN bus to a Victron Multiplus GX and the two packs are interconnected by RS485
> There is a design flaw with this SEPLOS BMS: When the CAN bus is in use, the desktop software can only monitor the slave battery pack(s), but not the master.  :-(
> So, I am using the iOS app to monitor both packs:
> So, I am really excited about @patman15 's  work to give access via home assistant:
> 
> I'd like to provide feedback information as well, but I am stuck at setting the system up.
> • Installed a ESP32 BT Proxy via ESPhome, and add "active: true" in yaml
> • I have downloaded the BLE Battery Management System (BMS) V1.13.0 in the HACS
> • When I try to add this integration, there is the message: "Keine unterstützen Geräte via Bluetooth gefunden."
> • The advertise monitor proofs that the ESP32 Proxy found the two BMS as BP00 and BP01 with RSSI of -79
> • In terminal the command     bluetoothctl devices >> No default controller available"
> • manually connecting does not work neither
> 
> Did I forget a step or did something wrong?
> 
> Right click in the advertise monitor not the BP01 und BP00 entries produces: {"name":"BP01\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","address":"04:7F:0E:13:21:33","rssi":-79,"manufacturer_data":{},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"94:54:C5:E8:38:24","connectable":true,"time":1741291451.996939,"tx_power":null}
> 
> {"name":"BP00\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","address":"04:7F:0E:13:21:DB","rssi":-77,"manufacturer_data":{},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"94:54:C5:E8:38:24","connectable":true,"time":1741291452.2929354,"tx_power":null}
> 
>  

 _Originally posted by @maps0n in [#171](https://github.com/patman15/BMS_BLE-HA/issues/171#issuecomment-2704839607)_

patman15, 2025-03-08:
Hi maps0n, please do not mix issues, it makes it very difficult to follow different discussions then. You have a detection problem, the post was about the stacked battery system.

Regarding your issue: The problem is the name that seems to be wrongly transmitted by the BT proxy. It should be `BP00` and `BP01` without all the `\u0000`. Can you please post the raw advertisement message of the packs, using [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)?

Thanks,
Patrick

maps0n, 2025-03-09:
Hello Patrick,
sorry for mixing and thank you for opening this case.

Attached are screenshots – I hope they comprise the information you need for the investigation.
Please advise, if something is missing – I am not familiar with Bluetooth nor the app.

TIA for looking into it.   BR Herb

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[BP01.txt](https://github.com/user-attachments/files/19153051/BP01.txt)

patman15, 2025-03-10:
It's an issue with the proxy. You can either try to recompile the proxy using a current version of ESPHome (if not done already) or use a [BT dongle](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters)?
Could you please also go to the [bluetooth integration](https://my.home-assistant.io/redirect/integration/?domain=bluetooth). On your BT adapter select *configure->advertisement monitor*, click the device in question and provide the information via *copy to clipboard*.

patman15, 2025-03-10:
> Is the yaml file I posted correct?

I can't see an issue with it.

> Attempt to integrate BLE Battery Management still error:
> "Keine unterstützen Geräte via Bluetooth gefunden."

The reason is still the wrongly communicated name by the proxy: `BP01\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000`
it shall be `BP01`.

> Currently no BT Dongle available- I am going to order one.
👍 

You can modify the following two lines
https://github.com/patman15/BMS_BLE-HA/blob/867f614e10370a2a2330a828a85a4229a8103de0/custom_components/bms_ble/manifest.json#L137

https://github.com/patman15/BMS_BLE-HA/blob/867f614e10370a2a2330a828a85a4229a8103de0/custom_components/bms_ble/plugins/seplos_v2_bms.py#L71

to `BP0*` then it will work also on your system. Note: any update will overwrite your changes!

maps0n, 2025-03-10:
Hi Patrick,
Your suggestions to replace BP0? to BP0* did the trick and the integration is up. 
So, I am starting to looking around.

Tomorrow the BT dongles will arrive and I will keep one as backup, when the files are overwritten.

Pending is what is wrong with my exp32 bluetooth proxy. 

Thank you very much for the integration you created and especially for your support on this issue here.

BR herb 

P.S.
I changed in the yaml file the framework to  type: esp-idf – was recommend by others - of course did address the the name issue.

patman15, 2025-03-10:
Hi! Just to confirm (and probably document in FAQ) you changed the platform from Arduino to esp-idf and that solves the name issue with the `\u0000`, correct?

maps0n, 2025-03-10:
Sorry, I wanted to write:  ... "of course did **not** address the the name issue."

One Question:
I experience that the entities data disappears randomly and comes back.
Where should I report about this?

maps0n, 2025-03-14:
Hello @patman15 

Update:
Today I replaced the esp32 BT proxy with the SABRENT BT-UB40 bluetooth dongle connected to my HA yellow system.

The original codes lines with "BP0?" are working fine.

No more need to use the modification "BP0*", which you proposed to handle the strange naming of the esp32 BT proxy.

Thanx again for your work on this integration and your support.

I also succeeded to get all individual cell voltages as outlined in the documentation. 
So,  no need to walk the basement and to start the iOS BMA app, if I want to monitor the individual cell voltage.  👍 
This was my motivation to the integration running. All other relevant data were already covered by the hass-victron installation – My SEPLOS BMS is connected with CAN the Victron system.

FYI: No change regarding the issue #189

BR herb

patman15, 2025-03-15:
Hi Herb! You are welcome! So this issue is mitigated? Can we close the topic for now?
I'm preparing for the next release so I'm in clean-up mode. :smiley:

maps0n, 2025-03-15:
Yes, this specific issue is solved. Your integration is fine.

closed 2025-03-15 by patman15