patman15/BMS_BLE-HA #218: Integration of Dyness Junior Box

shopf, 2025-03-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hello.
I have bought a 1.6 kWh battery from Dyness model Junior Box and would like to have it in Home Assistant.
The battery has WiFi and Bluetooth. BLE Battery Management Systems cannot be found at the moment.

Thank you in advance for your support.

### Additional context

Result from bluetoothctl info 90:EB:48:0D:CA:C4:
```
Unable to open mgmt_socket
Device 90:EB:48:0D:CA:C4 (public)
Name: R07E8546681A00F9
Alias: R07E8546681A00F9
Paired: no
Bonded: no
Trusted: no
Blocked: no
Connected: no
LegacyPairing: no
RSSI: 0xffffffcc (-52)
AdvertisingFlags: 06
```

I am a Home Assistant and Python beginner and am currently trying to run the Service Explorer script. Hopefully I can do it.

patman15, 2025-03-10:
Hi @shopf! I can take a look when you have more information, especially about which BMS is used. Please also consider the ask in the issue template, which is the most helpful input:
> To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

closed 2025-04-20 by github-actions[bot]

shopf, 2025-06-15:
Hello,
sorry for the late reply.
I just went to the Novelbits site and tried to capture the Bluetooth traffic with an iPadOS 18.5. I can't get it to work with the XCode PacketLogger. Does this only work with a Mac? Unfortunately I don't have a Mac.
Could this also work with an iPad or iPhone?
Many thanks for your feedback.