patman15/BMS_BLE-HA #226: JBD AP21 BMS not detected

ktrumper, 2025-03-19:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hi Patrick, Thank you for taking a look.  This is the advertisement data for the 1st BMS.  I renamed it to the model #, I don't recall exactly how it came from the factory though.

{"name":"ap21s002-l21s","address":"A4:C1:37:42:3E:D9","rssi":-84,"manufacturer_data":{"16089":"4237c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"DC:A6:32:2C:81:C5","connectable":true,"time":1742345641.5779326,"tx_power":null}

patman15, 2025-03-19:
Hmm, this one is a bit more challenging. The current identification patterns are here:
https://github.com/patman15/BMS_BLE-HA/blob/702144139b0810d1d69cd3fe5961bf92440c78ab/custom_components/bms_ble/plugins/jbd_bms.py#L63-L71

I could add a further one. Looking at the first three patterns, I would assume that your battery came with a name matching `AP2?S*`. Would adding that pattern help you? (Note: I think it is case sensitive ...)

ktrumper, 2025-03-19:
I renamed the battery to SP2... and that did the trick!  Maybe it's worth adding the AP2... to the list for others, I can confirm this is working.  

I noticed the module stays connected so I can no longer login with the Android app.  Not a big deal but what's the best way to disconnect the BLE BMS in home assistant?

ktrumper, 2025-03-20:
That works for me thanks!

patman15, 2025-03-20:
Still need the issue for the change of the detection pattern.

closed 2025-03-20 by patman15