patman15/BMS_BLE-HA #231: Seplos v3 BMS ("CSY...") not detected

patman15, 2025-03-23:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

> source: https://community.home-assistant.io/t/ble-battery-management-systems-custom-integration/728506/53

```
{“name”:"CSY012405290042 ",“address”:“C0:D6:3C:4C:A5:B1”,“rssi”:-78,“manufacturer_data”:{},“service_data”:{},“service_uuids”:[“0000fff0-0000-1000-8000-00805f9b34fb”],“source”:“3C:8A:1F:A9:03:F8”,“connectable”:true,“time”:1742727695.8344843,“tx_power”:null}
```

patman15, 2025-03-23:
Please see the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/feature-seplos-v3-pattern-CSY) and install it using the [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) method. Let me know if it works and please provide a debug log for verification. I would be also interested which type of battery this is, since the BT name is totally different from the other Seplos systems I've seen so far. You didn't rename, right?

broekieman, 2025-03-23:
It shows up now but keeps initializing. Where do I find debug log, is this file OK? I attached the bms pdf. I did not rename. 

[logs-3.txt](https://github.com/user-attachments/files/19411635/logs-3.txt)

broekieman, 2025-03-23:
Screenshot bms

[image]

patman15, 2025-03-24:
> It shows up now but keeps initializing. Where do I find debug log, is this file OK?
> 
Yes, that is the correct log. It seems that your BMS is too far away, the signal strength is very weak: `rssi: -81`. Please try to get above -75 dBm.

patman15, 2025-03-24:
@broekieman Ok, that did not help. In general this no issue with the integration, your Bluetooth system fails to connect. Can you try a different BT adapter?

broekieman, 2025-03-24:
I have only one esp32 to create a Bluetooth proxy.

patman15, 2025-03-24:
Well, either your BMS requires [pairing](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#my-bms-needs-a-pin-how-can-i-enter-it) or it is some issue with your BT setup that I cannot debug from remote. You could try another [high-performance adapter](https://www.amazon.de/dp/B093C98PFF) but it's not guaranteed that it helps.

broekieman, 2025-03-24:
If I want to use my phone app, I can only connect through the app and not by the normal Bluetooth setting. I also tried a shelly plug as proxy but it's says: "Deze adapter heeft geen ondersteuning voor het maken van actieve (GATT) verbindingen".

Any suggestion how to pair my proxy with the bms?

broekieman, 2025-03-24:
Don't know how, but after removing the map and a new manual installation it works.

Do have an example for a card to show the cells? 

[image]

patman15, 2025-03-24:
> Don't know how, but after removing the map and a new manual installation it works.
> 
Which map?

> Do have an example for a card to show the cells?
> 
[Mushroom template card](https://github.com/piitaya/lovelace-mushroom) together with the an appropriate template similar to the one in the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking).

broekieman, 2025-03-24:
> Which map?
>
>
> [Mushroom template card](https://github.com/piitaya/lovelace-mushroom) together with the an appropriate template similar to the one in the [...]

I meant the folder with your code.

closed 2025-03-24 by patman15

broekieman, 2025-03-24:
Could you please help me to create the helper templete for individual cell voltage? 

[image]

patman15, 2025-03-24:
Hi!

1. My Dutch is really bad, so I do not know what you want to tell me with this screenshot.
2. Did you have a look at the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking)?

broekieman, 2025-03-24:
> Hi!
> 
> 1. My Dutch is really bad, so I do not know what you want to tell me with this screenshot.
> 2. Did you have a look at the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking)?

I don't know how to create the card for cell voltages. The screenshot is the helper I made.

broekieman, 2025-03-25:
> Hi! The screenshot does not show a helper, but the delta voltage sensor. Please follow my advice regarding the FAQ. Everything you need is there to create individual sensors and then you can add them to the dashboard. If you have specific questions please raise a new issue of type `question`.

I was so smart to keep your text:"The index [0] can be in the range from 0 to the number of cells-1, i.e. 0-3 for a 4 cell battery." in the helper. It works now. 😂Can you share your cards code so I can adjust them to.my sensor names?