patman15/BMS_BLE-HA #236: Possible to make it work? Compatible?

HannuEKN, 2025-03-27:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

This is two 330Ah LiPo4, a new product  from "batterilgret.se"
=> https://www.batterilagret.se/sortiment/lithiumbatterier/lithiumbatterier-12v/330ah-sbl-lithiumbatteri--bluetooth-och-varme--underseat-29477

An earlier model (discontinued, 150Ah) did work "out of the box" with "BMS-BLE", when BMS-BLE was just installed.

Of some reason "info" on the first battery doesn't show all of the values the second does. 

# bluetoothctl info F9:56:0B:2C:E4:2F
Unable to open mgmt_socket
Device F9:56:0B:2C:E4:2F (public)
	Name: SBL-12330BLH1-242055
	Alias: SBL-12330BLH1-242055
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
	ManufacturerData Key: 0x007b (123)
	ManufacturerData Value:
  02 ff ff 7d                                      ...}            
	RSSI: 0xffffffb9 (-71)
	AdvertisingFlags:
  06                                               .               

# bluetoothctl info C2:33:20:61:82:60
Unable to open mgmt_socket
Device C2:33:20:61:82:60 (public)
	Name: SBL-12330BLH1-242049
	Alias: SBL-12330BLH1-242049
	Paired: no
	Bonded: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (00002600-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
	ManufacturerData Key: 0x007b (123)
	ManufacturerData Value:
  02 ff ff 7d                                      ...}            
	RSSI: 0xffffffb8 (-72)
	AdvertisingFlags:
  06                                               .               

# cat /etc/os-release  # raspberry pi 4
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.19.4
PRETTY_NAME="Alpine Linux v3.19"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://gitlab.alpinelinux.org/alpine/aports/-/issues"

# uname -a
Linux core-ssh 6.6.74-haos-raspi #1 SMP PREEMPT Mon Mar 17 11:21:04 UTC 2025 aarch64 Linux

BT -> Config -> Advertisement 
{"name":"SBL-12330BLH1-242049","address":"C2:33:20:61:82:60","rssi":-72,"manufacturer_data":{"123":"02ffff7d"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","00002600-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb"],"source":"D8:3A:DD:92:51:B3","connectable":true,"time":1743106203.7045412,"tx_power":null}

{"name":"SBL-12330BLH1-242055","address":"F9:56:0B:2C:E4:2F","rssi":-84,"manufacturer_data":{"123":"02ffff7d"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"D8:3A:DD:92:51:B3","connectable":true,"time":1743107563.8768537,"tx_power":null}

### Additional context

https://play.google.com/store/search?q=sbl+connect&c=apps - android app, there should be one for ios too.

Don't know if there is a "SDK"-like doc somewhere, have not located anything like it.

patman15, 2025-03-28:
Hi!
I added the missing detection pattern to the JBD BMS. Please try the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/236-add-sbl-battery) via [manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation).
Best regards, Patrick

HannuEKN, 2025-03-29:
Hmm... the direct link to the branch doesn't work for me, both SBL-12330BLH1-* popped up as discovered immediately after ha-restart, and SBL-12330BLH1-242049 added "immediately", SBL-12330BLH1-242055 took a bit longer though.
About to explore the 13 added entities. :-)

patman15, 2025-03-29:
> SBL-12330BLH1-242055 took a bit longer though.

This most likely due to the [limitation of the RPi4 internal adapter](https://www.home-assistant.io/integrations/bluetooth/#cypress-based-adapters). If it starts bugging you, think about buying a [high-performance adapter](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters).

closed 2025-03-30 by patman15