patman15/BMS_BLE-HA #241: Volthium battery device not found

gilbertth, 2025-04-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Trying to connect a Volthium battery throw ble proxy (esp32). the log of the esphome shows the name of the batteries but integration says device not found. Cant seem to find device in bluetootctl devices. The name of the two batteries are advertised V-12V200Ah-0215 and V-12V200Ah-0289

### Additional context

debug here:
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/esphome-web-7c2600.yaml...
INFO Starting log output from 192.168.86.64 using esphome API
INFO Successfully connected to esphome-web-7c2600 @ 192.168.86.64 in 0.118s
INFO Successful handshake with esphome-web-7c2600 @ 192.168.86.64 in 0.096s
[13:35:17][I][app:100]: ESPHome version 2025.2.2 compiled on Apr  1 2025, 10:40:26
[13:35:17][C][wifi:600]: WiFi:
[13:35:17][C][wifi:428]:   Local MAC: 78:21:84:7C:26:00
[13:35:17][C][wifi:433]:   SSID: 'VIDEOTRON8369'[redacted]
[13:35:17][C][wifi:436]:   IP Address: 192.168.86.64
[13:35:17][C][wifi:440]:   BSSID: B0:E4:D5:2D:52:C2[redacted]
[13:35:17][C][wifi:441]:   Hostname: 'esphome-web-7c2600'
[13:35:17][C][wifi:443]:   Signal strength: -65 dB ▂▄▆█
[13:35:17][V][wifi:445]:   Priority: 0.0
[13:35:17][C][wifi:447]:   Channel: 6
[13:35:17][C][wifi:448]:   Subnet: 255.255.255.0
[13:35:17][C][wifi:449]:   Gateway: 192.168.86.1
[13:35:17][C][wifi:450]:   DNS1: 192.168.86.1
[13:35:17][C][wifi:451]:   DNS2: 0.0.0.0
[13:35:17][C][logger:177]: Logger:
[13:35:17][C][logger:178]:   Max Level: VERY_VERBOSE
[13:35:17][C][logger:179]:   Initial Level: VERY_VERBOSE
[13:35:17][C][logger:181]:   Log Baud Rate: 115200
[13:35:17][C][logger:182]:   Hardware UART: UART0
[13:35:17][C][bluetooth_proxy:091]: Bluetooth Proxy:
[13:35:17][C][bluetooth_proxy:092]:   Active: YES
[13:35:17][C][bluetooth_proxy:093]:   Connections: 3
[13:35:17][C][bluetooth_proxy:094]:   Raw advertisements: YES
[13:35:17][C][bluetooth_proxy.connection:017]: BLE Connection:
[13:35:17][C][esp32_ble_client:048]:   Address: 
[13:35:17][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[13:35:17][C][esp32_ble_client:083]:   State: IDLE
[13:35:17][C][bluetooth_proxy.connection:017]: BLE Connection:
[13:35:17][C][esp32_ble_client:048]:   Address: 
[13:35:17][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[13:35:17][C][esp32_ble_client:083]:   State: IDLE
[13:35:17][C][bluetooth_proxy.connection:017]: BLE Connection:
[13:35:17][V][mdns:122]:     TXT: board = esp32dev
[13:35:17][V][mdns:122]:     TXT: network = wifi
[13:35:17][C][esphome.ota:073]: Over-The-Air updates:
[13:35:17][C][esphome.ota:074]:   Address: esphome-web-7c2600.local:3232
[13:35:17][C][esphome.ota:075]:   Version: 2
[13:35:17][C][safe_mode:018]: Safe Mode:
[13:35:17][C][safe_mode:020]:   Boot considered successful after 60 seconds
[13:35:17][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:35:17][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[13:35:17][C][api:140]: API Server:
[13:35:17][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:35:17][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:35:17][V][bluetooth_proxy:059]: Proxying raw packet from 02:00:07:B9:99:77, length 39. RSSI: -78 dB
[13:35:17][V][bluetooth_proxy:061]: Proxying 1 packets
[13:35:17][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 2199152859511
  rssi: -78
  address_type: 0
  data: '**V-12V200Ah-0215
	V-12V200Ah-0215'**
}
}
[13:35:18][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:35:18][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:35:18][V][bluetooth_proxy:059]: Proxying raw packet from 02:00:07:B9:99:77, length 39. RSSI: -74 dB
[13:35:18][V][bluetooth_proxy:061]: Proxying 1 packets
[13:35:18][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 2199152859511
  rssi: -74
  address_type: 0
  data: 'V-12V200Ah-0215
	V-12V200Ah-0215'
}
}
[13:35:18][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:35:18][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:35:18][V][bluetooth_proxy:059]: Proxying raw packet from 02:00:07:C3:10:B4, length 39. RSSI: -74 dB
[13:35:18][V][bluetooth_proxy:061]: Proxying 1 packets
[13:35:18][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 2199153479860
  rssi: -74
  address_type: 0
  data: 'V-12V200Ah-0289
	V-12V200Ah-0289'
}
}

patman15, 2025-04-02:
Hi!
Hmm, I need to adapt the detection pattern, but there could be a second issue with the proxy. Can you please provide me the advertisement data from the BT monitor? (please see the [trouble shooting guide](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#if-your-device-is-not-recognized))

gilbertth, 2025-04-02:
Hi do you mean this ? Btw many thanks for this. Will be extremely useful to me.

[image]

patman15, 2025-04-02:
>  Btw many thanks for this. Will be extremely useful to me.

You're welcome.
Please try the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/241-volthium-battery-device-not-found) via [manual install](https://github.com/patman15/BMS_BLE-HA/tree/241-volthium-battery-device-not-found#installation) and provide a debug log for final verification. Hope it works!

gilbertth, 2025-04-02:
Cool ! problem solved ! Thanks again. Only issue data is available and then after 30 seconds (new polling i suppose) becomes unavailable. after another 30 seconds data is available. This cycle repeats itself.

patman15, 2025-04-02:
@gilbertth Please provide me a debug log of a few minutes, then I can check if what's happening.

gilbertth, 2025-04-02:
Here is the log.

[logs_esphome-web-7c2600_logs (2).txt](https://github.com/user-attachments/files/19573345/logs_esphome-web-7c2600_logs.2.txt)

patman15, 2025-04-02:
I would need the debug log of the entity, not of the proxy. Please go to the integration and enable the debug log, wait a few minutes, disable it again, then it will ask you for download. Thanks.

gilbertth, 2025-04-02:
Hi having a hard time to include the log file maybe too big is there a limit to a file attached ? Is there another way to send you the file ?

patman15, 2025-04-03:
Hi! My contact data is on my profile, you could also zip it or just cut it down to the last 2-3 minutes. All options are fine for me.

gilbertth, 2025-04-03:
Hi! ok here is a zip file. Thanks.

[bms_ble.txt.zip](https://github.com/user-attachments/files/19588833/bms_ble.txt.zip)

patman15, 2025-04-03:
Sorry to bug you again, but this protocol is not in debug mode. Please follow the [troubleshooting guide](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#in-case-you-have-troubles-youd-like-to-have-help-with) to retrieve the correct log.

gilbertth, 2025-04-03:
Cant seem to download the diagnostic file with my chrome browser. I followed the troubleshooting guide and the diagnostic file is not found. i let the log recording on for 3-4 minutes is it enough to produce a file ? Do you know another way to retrieve the diagnostic file ?

[image]

[image]

patman15, 2025-04-03:
> i let the log recording on for 3-4 minutes is it enough to produce a file ? 

Yes. What I need is the debug log that is generated if you press the button that your left arrow points to. The diagnosis data is interesting, but won't help in your case.

> Do you know another way to retrieve the diagnostic file ?

You can go to system -> protocols, select HomeAssitant Core and see/download the logs also there.

patman15, 2025-04-03:
That's ok, but the log looks a bit weird. Need to investigate if this can actually happen like this or if something else is wrong.

gilbertth, 2025-04-03:
I have an app for the battery but not sure how to capture the bluetooth traffic.

patman15, 2025-04-04:
> I have an app for the battery but not sure how to capture the bluetooth traffic.

Please check one of the two links I gave for Android/iOS!

gilbertth, 2025-04-04:
Ok capturing the bluetooth traffic with the proposed method is difficult. The batteries are in a van and difficult to move. The bluetooth range of the app is limited and i have a desktop mac. i understand that the phone must be hooked to the mac for the log to be tranferred. Is there another ios solution that would snif bluetooth without the use of a computer attached ?

On an other solution, could extending the polling request from 30 secs to 1 minute from home assistant resolve the issue ?

gilbertth, 2025-04-05:
Hi after different test (changing the interval polling wasn't successful), it seems that the wifi distance was the problem. I moved the esphome proxy closer to a wifi router and the problem doesn't appear anymore. This is now fully functional.

Again many thanks !

patman15, 2025-04-05:
@gilbertth ok, thanks for the info. I will close this issue and merge the changes for the next release.

closed 2025-04-05 by patman15