patman15/BMS_BLE-HA #242: Perfektium LiFePO4 BMS Integration

DerMatze3279, 2025-04-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Perfektium LiFePO4 Battery BMS Integration would be great as used by many campers today.

patman15, 2025-04-02:
Hi! If you could provide additional information, as mentioned in the issue template
> Add any other context (documentation) or screenshots about the feature request here. For new BMS types, please provide the output of `bluetoothctl info <MAC>` and the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py). To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

I could have a look.

DerMatze3279, 2025-04-02:
Sorry Patrick, for sure! You can find the bluetoothctl and the Bluetooth traffic traces attched. The service explorer script ist failing on my end, any suggestions?

service_explorer.py
➜  /share python service_explorer.py  --address A4:C1:38:E4:77:CD
2025-04-02 20:23:11,062 __main__ INFO: starting scan...
2025-04-02 20:23:21,399 __main__ ERROR: could not find device with address 'A4:C1:38:E4:77:CD'  

bluetoothctl info A4:C1:38:E4:77:CD
Device A4:C1:38:E4:77:CD (public)
        Name: PKT2201PB121000084
        Alias: PKT2201PB121000084
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0x77cd (30669)
        ManufacturerData.Value:
  e4 38 c1 a4                                      .8..            
        RSSI: 0xffffffd2 (-46)
        AdvertisingFlags:
  06                                               .               
[CHG] Device 61:50:C3:0B:89:44 RSSI: 0xffffffc2 (-62)
[CHG] Device 44:5C:ED:D2:CE:D1 RSSI: 0xffffffc1 (-63)

[Archive.zip](https://github.com/user-attachments/files/19574117/Archive.zip)

patman15, 2025-04-02:
Hi! Thanks. Looks like a JBD BMS. I modified the detection pattern, please try the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/242-perfektium-lifepo4-bms-integration) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation). A debug log would be cool to finally check if it works.

DerMatze3279, 2025-04-03:
That was fast, wow!!! Thank you so much, works. You can find the log file attached.

Matthias

patman15, 2025-04-03:
Integration works fine, but you seem to have an issue with the BT system being pretty busy. You might want to consider using a [high-performance adapter](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters) and/or monitor the `link quality` sensor.

DerMatze3279, 2025-04-03:
Thanks you Patrick, will do!

-

closed 2025-04-04 by patman15