patman15/BMS_BLE-HA #245: New BMS Integration - DALY Smart Active Balance BMS 150A

elliotbailey13, 2025-04-02:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I see there is already an integration for DALY BMS devices but it doesn't work with their Smart Balancing BMS product.

### Additional context

[Product Site](https://www.dalybms.com/copy-smart-nmc-bms-20s-21s-24s-27s-32s-30a-250a-common-port-product/)

2025-04-02 12:37:15.160 DEBUG (MainThread) [custom_components.bms_ble] confirm step for JHB-501901019E89
2025-04-02 12:37:16.423 DEBUG (MainThread) [custom_components.bms_ble] confirm step for JHB-501901019E89
2025-04-02 12:37:16.424 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JQVNTE07AFMJPERSRPJXP7V0 version=1 domain=bms_ble title=JHB-501901019E89 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=50:19:01:01:9E:89>
2025-04-02 12:37:16.424 DEBUG (MainThread) [custom_components.bms_ble.daly_bms::JHB-501901019E89:9E89)] initializing Daly Smart BMS, BT address: 50:19:01:01:9E:89
2025-04-02 12:37:16.425 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for JHB-501901019E89 (50:19:01:01:9E:89) as Daly Smart BMS
2025-04-02 12:37:16.425 DEBUG (MainThread) [custom_components.bms_ble] JHB-501901019E89: advertisement: {'name': 'JHB-501901019E89', 'address': '50:19:01:01:9E:89', 'rssi': -94, 'manufacturer_data': {260: b'\x01P\x19\x01\x01\x9e\x89JHB'}, 'service_data': {}, 'service_uuids': [], 'source': '58:BF:25:36:80:E8', 'advertisement': AdvertisementData(local_name='JHB-501901019E89', manufacturer_data={260: b'\x01P\x19\x01\x01\x9e\x89JHB'}, tx_power=-127, rssi=-94), 'device': BLEDevice(50:19:01:01:9E:89, JHB-501901019E89), 'connectable': True, 'time': 1388960.177989017, 'tx_power': None}
2025-04-02 12:37:16.425 DEBUG (MainThread) [custom_components.bms_ble] JHB-501901019E89: BMS data update
2025-04-02 12:37:16.425 DEBUG (MainThread) [custom_components.bms_ble.daly_bms::JHB-501901019E89:9E89)] connecting BMS
2025-04-02 12:37:16.425 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 1
2025-04-02 12:37:21.887 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 1, last rssi: None)
2025-04-02 12:37:22.389 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 2
2025-04-02 12:37:25.978 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 2, last rssi: None)
2025-04-02 12:37:26.480 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 3
2025-04-02 12:37:32.460 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 3, last rssi: None)
2025-04-02 12:37:32.961 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 4
2025-04-02 12:37:39.014 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 4, last rssi: None)
2025-04-02 12:37:39.515 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 5
2025-04-02 12:37:45.632 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 5, last rssi: None)
2025-04-02 12:37:46.133 DEBUG (MainThread) [bleak_retry_connector] 50:19:01:01:9E:89 - 50:19:01:01:9E:89: Connection attempt: 6
2025-04-02 12:37:46.427 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching JHB-501901019E89 data in 30.002 seconds (success: True)
2025-04-02 12:37:46.427 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry JHB-501901019E89 for bms_ble

Logger: homeassistant.config_entries
Source: config_entries.py:753
First occurred: 11:29:06 AM (2 occurrences)
Last logged: 12:37:46 PM

Error setting up entry JHB-501901019E89 for bms_ble
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 753, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/__init__.py", line 56, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 307, in async_config_entry_first_refresh
    await self._async_refresh(
        log_failures=False, raise_on_auth_failed=True, raise_on_entry_error=True
    )
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/coordinator.py", line 101, in _async_update_data
    if not (bms_data := await self._device.async_update()):
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/plugins/basebms.py", line 288, in async_update
    await self._connect()
  File "/config/custom_components/bms_ble/plugins/basebms.py", line 235, in _connect
    self._client = await establish_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/bleak_retry_connector/__init__.py", line 366, in establish_connection
    await client.connect(
    ...<3 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 328, in connect
    connected = await super().connect(**kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak_esphome/backend/client.py", line 77, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak_esphome/backend/client.py", line 297, in connect
    await self._client.bluetooth_device_connect(
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/aioesphomeapi/client.py", line 598, in bluetooth_device_connect
    await connect_future
asyncio.exceptions.CancelledError

patman15, 2025-04-02:
@elliotbailey13 Try to improve your signal strength! `rssi=-94` should be well above -80 dBm.

elliotbailey13, 2025-04-02:
Hello @patman15 , good call. I turned on a Bluetooth proxy on another ESPHome device that was closer to the BMS and it worked! Thank you for the quick response.

closed 2025-04-02 by elliotbailey13