patman15/BMS_BLE-HA #246: Unknown BMS not being detected

ProfsRetroFix, 2025-04-03:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I purchased a BMS for my car battery, and it is being detected by Home Assistant as an 'iBeacon' via a Bluetooth proxy (Atom mini), as such I am unable to run bluetoothctl. Unfortunately it is pretty generic and I don't have a brand name.

The iBeacon entry identifies it's vendor as "Nanjing Qinheng Microelectronics Co., Ltd."

I switched the BT proxy into very verbose mode, and caught an advertisement entry for it. Details attached. Apologies for the briefness - I have no idea what I'm doing(!), but please let me know if there's anything else I can provide.

### Additional context

[ble.txt](https://github.com/user-attachments/files/19588493/ble.txt)

patman15, 2025-04-03:
Hi! Unfortunately, I cannot identify the BMS clearly from that information, I need to ask you to follow the guidance in the ticket template:
> For new BMS types, please provide the output of `bluetoothctl info <MAC>` and the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py). To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

ProfsRetroFix, 2025-04-04:
> Hi! Unfortunately, I cannot identify the BMS clearly from that information, I need to ask you to follow the guidance in the ticket template:

Unfortunately as stated, I can't run bluetoothctl as it's coming through via BT proxy from the garage. I'll try and capture the BT traffic later today however and come back to you!

patman15, 2025-04-04:
Hmm, looks like nothing I have seen before. Can you provide a link to the app you are using? Might get a bit challenging.

ProfsRetroFix, 2025-04-04:
Just my luck eh? Sure, I scanned the barcode on the device itself and it [lead here](https://play.google.com/store/apps/details?id=com.dc.battery.monitor2)

patman15, 2025-04-04:
The app seems to get updated data every second (does this match your observations?). The data sample is only 16 bytes which means, there won't be much information in it. Also it looks encrypted, scrambled, something like that. Also there is not really much information in the app except for the overall voltage. Do you have more information somewhere than just the voltage?
Sadly, for the first time, I need to say that this does not look like a promising candidate. Still I'll have a second look some other day.

ProfsRetroFix, 2025-04-04:
Indeed, it does seem to be pretty much updating live while the app is open. Moving away and coming back the next day, it does a more full sync and fills in all the blanks from the previous day on a line graph within the app.

To be honest it's pretty barebones, only voltage and charge level, although the charge level is pretty inaccurate. It reads at 0% most of the time, but I'm obviously able to use the car and it's reading a voltage.

Silly me assuming they were all basically the same. Thanks for looking into it.

patman15, 2025-04-06:
Sorry for that, but it does not sound being worth the effort. If there is something else, let me know. Meanwhile I'll close this topic.

closed 2025-04-06 by patman15