patman15/BMS_BLE-HA #250: device communication timed out

goodcat432157, 2025-04-05:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hello, 

the integration recognizes my Solarcharger from Redodo 40A BT-ROCC2440 but after loading I always get the error: device communication timed out 

What could be the reason ?

[image]

[BMS REDODO](https://www.redodopower.de/products/redodo-40-a-mppt-solarladeregler?srsltid=AfmBOorvAjnv1iCaaNAPQldyCEVqkQGLFCkiVUlIlIKdl0toMm0gXA-P)

patman15, 2025-04-05:
Most likely it uses a different protocol, as it is not a BMS. Can you provide a debug log or diagnostic data of the device?

goodcat432157, 2025-04-05:
Hi, unfortunately I only have access to the Solar Charger via an Android app, so I can't download any logs.

patman15, 2025-04-05:
Please see the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#in-case-you-have-troubles-youd-like-to-have-help-with).

goodcat432157, 2025-04-05:
I have attached the log file

[error_log.txt](https://github.com/user-attachments/files/19615195/error_log.txt)

goodcat432157, 2025-04-06:
Too bad I thought I could read out the values, I'll just deactivate it. Thanks for the feedback

closed 2025-04-24 by patman15