patman15/BMS_BLE-HA #257: Different manufacturer ID in JBD BMS

robermf, 2025-04-07:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hello. 
The JBD BMS has a different name, so the add on shows "No supported devices found via Bluetooth"
I have trying but myself but I can't get it works.

Is any chance to add this new ID? Thanks!

➜  ~ bluetoothctl info A4:C1:37:24:EB:A0
Device A4:C1:37:24:EB:A0 (public)
        Name: xiaoxiang BMS V9.0
        Alias: xiaoxiang BMS V9.0
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0xc1a4 (49572)
        ManufacturerData.Value:
  37 24 eb a0                                      7$..            
        RSSI: 0xffffffae (-82)
        AdvertisingFlags:
  05                                               .

patman15, 2025-04-07:
Hi @robermf!

You can do a manual install of the main branch, I already added it, but it's not yet released.

robermf, 2025-04-08:
Thank you patman15 :)
I tried with the last released but still didn't find it. I will keep trying. 

Regards!