patman15/BMS_BLE-HA #260: Adding additional bms kills first installed

Tulexcorp, 2025-04-11:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

[home-assistant_bms_ble_2025-04-11T15-46-19.014Z.log](https://github.com/user-attachments/files/19709940/home-assistant_bms_ble_2025-04-11T15-46-19.014Z.log)

Sorry, very new to this, please let me know if I'm not doing this right.
Running HA as HAOS on dedicated mini pc. 6 seplos V2 batteries. Master battery connected to Sol-ark 15K via can. All batteries in parallel.
Can read batteries 2-6 (1-5) in Seplos PC software. Can see all 6 batteries on phone via Seplos bluetooth app.  

Set up BP00, it works fine. It shows as a device with 13 entities. Talking to someone else who is playing with this, I got the impression that it will use that as a master and allow you to see the other batteries. That isn't the case for me. So, added BP01. For a short time, it shows no device for BP01, just entities, and I can see data for BP01. Then all entities for both BP00 and BP01 become unavailable. BP01 then shows as a device.

Tulexcorp, 2025-04-12:
[home-assistant_bms_ble_2025-04-12T03-32-03.691Z.log](https://github.com/user-attachments/files/19717110/home-assistant_bms_ble_2025-04-12T03-32-03.691Z.log)

Did a fresh install to get hopefully a more useful log. This time same result. First BMS runs with no issue. Install second BMS, looks OK for a short time, then all entities go to unavailable. This time both BMSs showed as devices from the beginning. Also a bit different is that both BMSs at install asked what area to install. Reboot the whole system, both BMSs come up for about 5 minutes, then back to unavailable entities.

patman15, 2025-04-12:
Hi! I had a look at your logs (thanks for them), but could not spot any issue with the integration. The BMSs simply stop answering. Probably it's due to interference or an issue with the system itself. There is a thread #171 where I tried to figure out how the system with it's main / additional batteries works, but I couldn't conclude from the feedback.
In theory the master should report all sub-batteries, but when I query it, it just reports data of the master element. Also some people reported that if (correctly (?)) connected, the BT of the sub-batteries shuts down and disappears.
I never heard that the master disappears. Which BT adapter are you using? There are [high-performance adapters](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters) and less optimal ones, especially with many devices (as you have) it makes a difference.
Happy to help, but since I do not have such batteries, I'm bound to observations by others.

Tulexcorp, 2025-04-12:
I have my mini PC within a foot or so of the batteries. I even opened the batteris to expose the chip on the Seplos display. Being a generic mini PC, all I can tell is that bluetooth in HA reports it as a 5 slot Realtek. IF I can get this working, I will have to add additional bluetooth to have enough slots for 6 batteries, so either trying a different adapter or bluetooth proxy would be my next step. While playing around with an older version, I did see attributes showing more than 16 cells under one delta entity. But there was no way to know what they went to, just a huge list of values.

patman15, 2025-04-12:
Try to increase the distance to the batteries, like 3-6 foot. Sometime being too close makes it worse. I can't think of any software issue so far. Let me know if you have other observations, maybe that triggers something with me. Sorry for not being able to help more at the moment.

Tulexcorp, 2025-04-12:
I've moved it farther away, no difference. What's odd to me is that I can load any of the BMSs, it runs with no issue. It's adding another that kills all that are added. I have a new bluetooth adapter (from their list, which is hard to do, the list is old) on the way, will try that and update.

patman15, 2025-04-12:
Which type of install are you using for HA? Did you take an HAOS image? If not, you may want to consider that. I had strange issues with BT and never figured out why, but it was due to my setup of the OS.

Tulexcorp, 2025-04-12:
I did the recommended method of booting Ubuntu from a USB and installing through that.

patman15, 2025-04-13:
Not sure, which variant that is, `supervised`? If unsure, please check the [system health page](https://my.home-assistant.io/redirect/system_health/) and send a screenshot. For BT issues I recommend to try the [HA OS](https://www.home-assistant.io/installation/#advanced-installation-methods) method at least temporary.

Tulexcorp, 2025-04-13:
This is HA installed as an operating system. Nothing else on the mini pc. 

[image]

patman15, 2025-04-13:
Hmm, looks good. Then I'm slightly running out of ideas. Maybe the other BT adapter brings a change. I could only think of disturbances on the wireless channel, but this is really hard to diagnose remotely.

closed 2025-04-15 by Tulexcorp

patman15, 2025-04-15:
> Apparently, the Realtek bluetooth adapters are a known issue, they don't have a reset pin, so HA can't reset them automatically.
> 
Thanks for the info, I might add a hint for others to the FAQ.

> Just for reference, of the 13 entities for each battery, only a few are different for the master battery. Not an issue.
>
That sounds reasonably realistic.

> "Problem" - Master is fine, says OK, all the others say problem, "show more" shows the problem being "unavailable". 
>
That is strange.

> Battery Current, Cycles, Volt, and Battery Level all say Pack Value under Attributes, but far as I can tell, the value is actually the master battery's value, not some pack average. Didn't check Link Quality or Signal strength, as I left those unenabled.
> 
Did you install the version 1.15.0? I'm a bit confused because for Seplos v2 I did not implement the pack values. Did you use some development version?
Anyways if you want to help fixing that, please post a debug log and your observations to #171.

> Thank you so much for trying to figure this out. This is an awesome integration.
>
@Tulexcorp, thanks for the feedback!

Tulexcorp, 2025-04-15:
Running clean install of 1.15.0, no special version. I'm OK with how it is now, I'm really only pulling the cell delta for now. I did provide the info in #171. Of note, because the Realtek adapter is being used to communicate with battery 1, the master, I don't know if that could be causing the "problem" issue. If I recall, for a short time,  the slave batteries didn't say problem.