patman15/BMS_BLE-HA #263: Battery not found via bt-proxy

snipah, 2025-04-13:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Creabest CBT Power

### Describe the issue

Battery is not found via bt proxy, even though it shows up in advertisement monitor:

[image]
[image]

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

No HA-log as no device found

snipah, 2025-04-13:
Set log-level to VERBOSE and it show, that the batteries MAC is being proxied:

[image]

patman15, 2025-04-14:
The shown advertisement information is wrong (reported UUID and manufacturer data). If I'm correct, this is the same battery as in https://github.com/patman15/BMS_BLE-HA/issues/59#issuecomment-2383190424, so I assume that it already worked at least once.
I would say, it's an issue with the proxy, try to use another BT adapter.

snipah, 2025-04-14:
You're totally right. Works via Raspi 4 directly, but I want to get it running via proxy. What could be a reason for wrong advertisement info? Too much load on the ESP32?

patman15, 2025-04-14:
>  bt-proxy is only missing "0000fff0-0000-1000-8000-00805f9b34fb"

It's mixing it up, it adds `ffb0`:
 `0000*ffb0*fff0-0000-1000-8000-00805f9b34fb` so the UUID is too long and thus out of spec.
I would guess for a memory problem, but I have never looked into the implementation. You need to ask the developers. Sorry.

snipah, 2025-04-16:
...read through your FAQ/ Troubleshooting and added a USB-stick to my Raspi4. Night-and-Day-Difference. Crazy how much better bluetooth reception is in comparison to onboard-bt on Raspberry.

closed 2025-04-16 by snipah