patman15/BMS_BLE-HA #264: gokwh bms not reorganized

boraXa, 2025-04-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

i have a gokwh battery with BMS DGJBD (200A) and I cant integrate. the message is “No supported devices found via Bluetooth”.

do you have any idea how can I add (maybe manually)

thanks a lot

[image]

[image]

patman15, 2025-04-15:
Hey! Did you rename the battery from the manufacturer default, i.e. what was it's original name (in case you changed it in the app)?

boraXa, 2025-04-15:
Hey Patrick,

I dont change the name of the bms

[image]

patman15, 2025-04-16:
Hi @boraXa, could you please install the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/264-gokwh-bms-not-reorganized) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation), [enable debug logging](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics), and then send me the log for verification. Should work in principle.

boraXa, 2025-04-16:
Hi Patrick,

I tried install manually but without success.
I cannot enable debug because the because the "card" cannot install ...

Thanks a lot

[image]

boraXa, 2025-04-16:
is working NOW :))))

Thanks a lot fro your support!!

Have a great easter!!! 

[image]

boraXa, 2025-04-16:
Hey Patrick,

now if it still works do you realize that we want even more (give it a finger and it takes your whole hand) :)))

is it possible that we can have access to the two options (charge/discharge - on/off)?

the idea is that we can make an automation in HA for example when it gets below 20% to stop the discharge and when it gets to 80% to stop the charge (it's just an example)

thanks a lot

patman15, 2025-04-16:
> is it possible that we can have access to the two options (charge/discharge - on/off)?

I know this is tempting, but for the time being I need to say that I won't do that. Multiple reasons:
1. Unreliability of Bluetooth, protocol information, testing, etc., see the [caution information](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information)
2. In order to avoid side-effects, I don't do any write operation to the BMS. This could lead to unintended behavior.
3. I try to keep all BMS implementations on the same level
4. I need to maintain currently 17 different protocols (still counting) which is a huge effort. Adding active settings would almost double the effort.
5. Working on getting it a HA core integration will consume a huge amount of time. I'd like to spend it there.

Thanks for the feedback, I will merge it for the next release.

Best regards, happy Easter,
Patrick

closed 2025-04-19 by patman15