patman15/BMS_BLE-HA #267: Ective BMS (LC100) - Device communication timed out

To-Lue, 2025-04-17:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Hello,
at first, i've happy to find this integration which should work with my two Ective LC100 batteries.

For the bluetooth connection i use an esphome bluetooth proxy (ESP32 D1 mini).
This proxy shwos me in advertise mode a lot of bluetooth devices around my proxy.
My two batteries are also found and listed by your bms integration.
If I want to add this new listed batteries, i'm getting an time out error.

The proxy works fine with a shelly blu motion device and my mopeka gas sensor (for testing).

Here are some information (advertise mode information about the two batteries, debugging info and my esphome config).

I hope, you can help me in this case :-)

Thank's a lot and have a nice day.

Greetings,
Tobias

[ESPHome-Config.txt](https://github.com/user-attachments/files/19791037/ESPHome-Config.txt)
[home-assistant_bms_ble_2025-04-17T07-30-48.801Z_short.txt](https://github.com/user-attachments/files/19791039/home-assistant_bms_ble_2025-04-17T07-30-48.801Z_short.txt)
[NWJ20230302020637.txt](https://github.com/user-attachments/files/19791040/NWJ20230302020637.txt)
[NWJ20230302020760.txt](https://github.com/user-attachments/files/19791038/NWJ20230302020760.txt)

patman15, 2025-04-17:
Hi @To-Lue! Please check your `RSSI` it's pretty low so I assume your proxy is not close enough to the batteries. Try to rearrange your setup, I see values -76, -77, -84. Please see [reference table](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information) for good values.
Other than that I do not see an issue. Parts of the data are received but no complete message.

To-Lue, 2025-04-17:
Hi @patman15 ,
thank's a lot for your support and your really fast response :)

I've done the bt-proxy near to the batteries.
Now the RSSI is -37 but with the same result.

With the Andoird App, i can check both batteries without any problems (not while trying to connect with the bt-proxy).

[home-assistant_bms_ble_2025-04-17T10-53-09.514Z_short.txt](https://github.com/user-attachments/files/19793033/home-assistant_bms_ble_2025-04-17T10-53-09.514Z_short.txt)

patman15, 2025-04-17:
Hi @To-Lue!

There are some suspicious warnings in your log, related to the proxy:
```
2025-04-17 12:50:38.758 WARNING (MainThread) [aioesphomeapi.connection] vpn-bt-proxy @ 192.168.9.10: Connection error occurred: [Errno 104] Connection reset by peer
2025-04-17 12:51:46.987 WARNING (MainThread) [homeassistant.components.esphome.manager] BT-Proxy - Harry - VPN: [W][component:237]: Component esp32_ble_tracker took a long time for an operation (60 ms).
2025-04-17 12:51:46.988 WARNING (MainThread) [homeassistant.components.esphome.manager] BT-Proxy - Harry - VPN: [W][component:238]: Components should block for at most 30 ms.
```
I can see that BT messages are getting lost which relates to (at least) the last two messages. Do you have another option to connect via BT? E.g. a BT-USB dongle? The other option would be to increase the timeout, but I'm pretty sure you are suffering from some Bluetooth issue.

To-Lue, 2025-04-17:
Hi @patman15 ,
thank you again for your help!

Unfortunately, I only have the BT proxy there; I can't connect a USB dongle to the ESP.
I also noticed that my ping processing time is high (approx. 120-150ms), which would likely also lead to a timeout.

Is there a way to extend the times?
If so, where can I increase them?

Thank you very much!

To-Lue, 2025-04-17:
Hi @patman15 ,
thank's for the information about the timeout and the location in the source code. Thanks to the adjustment, the integration now works without issues. All information can be read and displayed. I'll inevitably have to adjust this value again after an update. Thank you very much!

Greetings,
Tobias

patman15, 2025-04-19:
@To-Lue I increased the value in the next release to 5s, hope that works for you as well.

closed 2025-04-26 by patman15

To-Lue, 2025-08-05:
Hi Patrick,
Since the last update, I have been experiencing constant timeout issues
again.
Until the update with the increased timeout times, your add-on was running
flawlessly.
Can you help me with this?
The timeout management has changed, so I can no longer manually adjust the
timeout to a specific value.

Thank's a lot and have a nice day.

Greetings,
Tobias