patman15/BMS_BLE-HA #269: BMS found in Advertisement monitor but "no Device found"

kiwiscanwifi, 2025-04-19:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

I have 13x new  JK-BD4A20S4P
I can see them under Advertisement monitor but then I am stuck on what happens next. I can not see my UUID on the manifest list

I am using a M5Stack Atom Lite

Advertisement monitor example:

Address: 98:DA:20:08:AF:35
Name: 15C17003E50AJ <- What I named it
Source: F4:65:0B:08:8B:E8

Advertisement data

Manufacturer data
2917	38 38 61 30 39 38 64 61 32 30 30 38 61 66 33 35

Service data
Service UUIDs
 0000fee7ffe0-0000-1000-8000-00805f9b34fb

Am I missing something or is the model I sourced use a different format for the UUID

Thanks

patman15, 2025-04-19:
Hi! This seems to be the same issue as #263 and due to a bug(?) in the proxy. Check that you have a minimal proxy configuration or try another BT adapter.

kiwiscanwifi, 2025-04-19:
2025.3.3
Upgrading to 2024.4.0 now.
Homeassistant is a VM and the BT is the M5Stack Atom Lite.
It was configured using the https://esphome.io/projects/index.html 

I didn't see the other issue before.. Yes I have a different string with "fee7" being included.

patman15, 2025-04-19:
`0000fee7ffe0-0000-1000-8000-00805f9b34fb` is an invalid UUID, so there is an issue with your BT setup or a HW issue.
`0000xxxx-0000-1000-8000-00805f9b34fb` would be correct. So the one above it is too long. Probably wrongly assembled, no idea, why.

julep21, 2025-05-03:
Hello,
I have the same problem with  a JK-B2A8S20P BMS.
HA in VM, BTproxy is the SMLIGHT SLZN-06M with the manufacturer's precompiled BT proxy firmware.
Advertisement monitor shows "service_uuids": "0000fee7ffe0-0000-1000-8000-00805f9b34fb" for the JK-B2A8S20P BMS.

Two Litime 100Ah batteries were immediately recognised by Ble BM integration. Their UUID is indeed in the correct format "0000ffe0-0000-1000-8000-00805f9b34fb".

PS: I deleted a third Litime battery from the Ble BM integration. How do I get it back ? Thanks

closed 2025-06-05 by patman15