patman15/BMS_BLE-HA #271: Seplos slaves not populating certain entities

Tulexcorp, 2025-04-19:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Seplos V2

### Describe the issue

Now that we've verified Seplos V2 needs each BMS installed, I've installed the latest version.

Still seeing entities "Problem" and "Runtime" as Problem and Unknown respectively.
Don't care about runtime, but hoping Problem can be fixed.

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

[home-assistant_bms_ble_2025-04-19T15-39-10.054Z.log](https://github.com/user-attachments/files/19821785/home-assistant_bms_ble_2025-04-19T15-39-10.054Z.log)

patman15, 2025-04-19:
Hi! Thanks for being precise and raising this again. Please try this [branch](https://github.com/patman15/BMS_BLE-HA/tree/fix-code-improvements). `Problem` should be fixed, `runtime` is only available during discharging, could that be the issue?

Tulexcorp, 2025-04-19:
"Problem" didn't change.

 It appears you are correct about run time only showing during discharge, as 3 of the 6 batteries that showed run time were not charging. The 3 that showed run time seemed to be cycling through values, they were up and down. But, not all of the batteries not charging populated with run time.  Again, nice to have it all working right, but runtime isn't something I'm going to use, so not an issue with me.

[home-assistant_bms_ble_2025-04-19T19-53-16.880Z.log](https://github.com/user-attachments/files/19822661/home-assistant_bms_ble_2025-04-19T19-53-16.880Z.log)

patman15, 2025-04-20:
Hi!
> "Problem" didn't change.

I fixed the wrong parser. 😩 Anyway, I updated the branch and fixed the protocol again, now it should definitely work.

> Edit to add - It's conceivable that the batteries not populating run time were not discharging, but simply not charging.

The value is (for Seplos v2) generated by the integration and available when `current < 0`, see the [sensor description](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information).

Tulexcorp, 2025-04-20:
Everything looks great now! Thanks again.

Probably shouldn't admit it in public, but my mind was thinking run time was how long the device was running. Probably from being a literal person, I would have expect it to say run time left for what it's reporting. That, and I didn't know it actually did that type of calculation, I thought it was just reporting raw data.

patman15, 2025-04-20:
> Everything looks great now! Thanks again.
> 
Cool, then I'll go with that version into the next release. Strange that nobody else complained ...

> Probably shouldn't admit it in public, but my mind was thinking run time was how long the device was running.
>
😆 I can tell you, you are neither the first nor the only one. I checked with other implementations, the term is commonly used (sometimes together with *remaining* but that makes the name so long ...)

Tulexcorp, 2025-04-22:
So what's interesting is that when all the batteries are not in some sort of fault mode, they all said OK for problem. 
Because all my batteries hit cell protection at full charge, they all say "Y" for warning on the battery display. But in HA, only the master battery shows problem, the rest say OK. 

[home-assistant_bms_ble_2025-04-22T21-37-21.965Z.log](https://github.com/user-attachments/files/19857216/home-assistant_bms_ble_2025-04-22T21-37-21.965Z.log)

patman15, 2025-04-23:
> Because all my batteries hit cell protection at full charge, they all say "Y" for warning on the battery display. But in HA, only the master battery shows problem, the rest say OK.

That is correct, only the master battery reports problems. All others do not send that information, so they are ok. That was the original bug that the integration expected problems also from the extension batteries.

> Will also add that problem isn't clearing from battery 1 even after all displays show no warnings.

The manual says this means `Waiting for intermittent power replenishment` what ever that is. Probably not really a problem? I could mask that if you think this is wrong to be reported.

> Rebooted HA, no change.

The integration has no stored values, so rebooting does not help to change values as they are all read live.

Tulexcorp, 2025-04-23:
Not sure what to think. I waited until the batteries were definitely below any high values, maybe down to 97% after reading 99.9%.
The app showed no faults, the batteries showed no faults on the displays.  But problem still remained in HA. Did an uninstall of the integration, re-installed, and it was gone. Maybe just timing?

The faults show up as two different faults in the apps. Don't remember exactly, but a yellow one that says something like intermittent power supply, and a red one that is overvoltage protection I think. Not even sure if both cause a warning on the display, the display just says warning Y/N. The yellow one isn't really a problem, it's just a notice from what I understand. The red one is what causes the battery to stop charging I would assume. I don't know how much control you have over all of that. I may even be able to turn off the warning in the app, not sure. Turning it off in the app might not stop what it sends out. But if you want to try to mask the one you pointed out, we can see what it does.

patman15, 2025-04-23:
You can take a look at the [protocol spec](https://github.com/user-attachments/files/15930423/Seplos.bluetooth.protocol.2.pdf). On page 16 you can find *surface15-Alarm event table*. I enabled all but event 7 and event 8. Your issue came from event 6, bit 1, *Waiting for intermittent power replenishment*

I would like to only exclude obvious useless ones, like `calibration not done` or `internal bit`. Everything with `warning` (why would you want to have a warning, when you do not want to be notified ...) or `error`/`failure` should be in to raise users attention. In a later stage I might add an attribute about the type of error, but this a huge effort, or (more likely) I might just add an attribute with the error code and let the user handle it.
If you want you can go through the table and suggest a list of items to be filtered.

Tulexcorp, 2025-04-23:
I would say just that one beyond what you've already excluded.  I don't understand the BMS well enough to suggest excluding alarms.  But if the waiting for intermittent power is the yellow one in the apps, and I'm fairly sure it is, it's not really an alarm, it comes on every time full charge is reached, so not needed. Any type of attribute, even a code, would be awesome. Most likely, 90+%  of them won't ever come into play, so it's not like we really need the attributes spelled out.

patman15, 2025-04-23:
Maybe that means something like the adsorption state? No clue.
Ok, will do a bit later.

Jalle19, 2025-04-24:
The "intermittent power supply" warning can be disabled from the Seplos app. For me it happens any time the battery is fully charged. I think the root cause is that when the inverter is in "passthrough" mode, a small current may sometimes flow to the battery. This is detected by thr BMS as something erroneus.

Jalle19, 2025-04-24:
"Intermittent Charging Function", that's the one. Disable it and that warning should go away.

Tulexcorp, 2025-04-24:
Don't want to disable that. That setting basically tells the battery to go into low charge state at around 96% to prevent overvoltage.

Tulexcorp, 2025-04-24:
Still the same. The two alarms on the battery are Monomer High Voltage Alarm and Total Pressure High Pressure Alarm. Both yellow alarms, neither stops the battery from working or continuing to charge.

IF one or both of those is causing "Problem" on battery one, it's probably doing what it's supposed to. Obviously it would be nice if only real issue alarms caused "Problem" to trip, but who's to say what a real issue is. All batteries (well, can't see the master battery) show the same alarms in the Seplos app. Didn't see the intermittent alarm on any batteries, so that alone wasn't what was tripping "Problem".

Tulexcorp, 2025-04-24:
Adding that I can't be confident that the intermittent alarm wasn't present in the Seplos PC app, as I checked again and realized that only 2 alarms fit on the screen, you have to scroll down to see more. Can't say if it was there at the time or not.

Tulexcorp, 2025-04-25:
In my situation, being closed loop with a Sol-Ark, I can't tell the inverter to control the batteries. It simply won't take it. My settings in the inverter are updated by the BMS. So, I have to let the BMS do all the work. What's been explained to me from people who actually understand this BMS and who are in the same situation is that you have to drive them to certain alarms. You tell it when you want it to cut off, but it shows as an alarm at that point. . So it's going to charge until it hits a parameter, then it's going to give me a yellow alarm. Because there are yellow and red alarms, and people told me to ignore the yellow alarms, I guess I was forgetting that not everyone has the same situation. I Apologized for the tail chasing, I just wasn't putting it all together. For the everyday person who can control their BMS from the inverter, you are correct, you want the alarms.

patman15, 2025-04-25:
@Tulexcorp does your statement therefore mean, that you are ok with me merging the current implementation of #275 to the next release? I'm planning to release 1.16 on the weekend (hopefully).

Tulexcorp, 2025-04-25:
Absolutely.

closed 2025-04-25 by patman15