patman15/BMS_BLE-HA #273: Felicity LUX-S-1600LG01

Ichhabekeinenase, 2025-04-20:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Hi,
I dont know if it is a Bug or Something but my BMS wont connect! I have the Debug log (The Important Parts).  
Bluetooth is an ESP32-S3 running ESPHome Bluetooth Proxy.

Thanks

2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JS99TGK8PVA30KYWGF1QGHJ5 version=1 domain=bms_ble title=F100216000124300034 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=A8:17:10:67:C8:3C>
2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble.felicity_bms::F100216000124300034:C83C)] initializing Felicity Solar LiFePo4 battery, BT address: A8:17:10:67:C8:3C
2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for F100216000124300034 (A8:17:10:67:C8:3C) as Felicity Solar LiFePo4 battery
2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble] F100216000124300034: advertisement: {'name': 'F100216000124300034', 'address': 'A8:17:10:67:C8:3C', 'rssi': -54, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': [], 'source': 'CC:DB:A7:97:83:7C', 'advertisement': AdvertisementData(local_name='F100216000124300034', tx_power=-127, rssi=-54), 'device': BLEDevice(A8:17:10:67:C8:3C, F100216000124300034), 'connectable': True, 'time': 1775252.574678882, 'tx_power': None}
2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble] F100216000124300034: BMS data update
2025-04-20 11:59:37.230 DEBUG (MainThread) [custom_components.bms_ble.felicity_bms::F100216000124300034:C83C)] connecting BMS
2025-04-20 11:59:37.230 DEBUG (MainThread) [bleak_retry_connector] A8:17:10:67:C8:3C - A8:17:10:67:C8:3C: Connection attempt: 1
2025-04-20 12:00:00.341 WARNING (MainThread) [homeassistant.helpers.service] Referenced devices 61f541da1241be0485c01e40eced8765 are missing or not currently available
2025-04-20 12:00:17.231 DEBUG (MainThread) [bleak_retry_connector] A8:17:10:67:C8:3C - A8:17:10:67:C8:3C: Timed out trying to connect (attempt: 1, last rssi: None)
2025-04-20 12:00:17.482 DEBUG (MainThread) [bleak_retry_connector] A8:17:10:67:C8:3C - A8:17:10:67:C8:3C: Connection attempt: 2
2025-04-20 12:00:18.913 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching F100216000124300034 data in 41.683 seconds (success: True)
2025-04-20 12:00:18.914 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry F100216000124300034 for bms_ble
Traceback (most recent call last):

patman15, 2025-04-20:
Hi! Hard to say from this short snippet. Can you provide a complete log? There is even a traceback missing that could be important.

Ichhabekeinenase, 2025-04-20:
Of Course, Im sorry. Heres a cleaned up log (I have some other Errors in my Home Assistant). Let me know if i removed something important! 

[home-assistant_bms_ble_2025-04-20T10-35-03.589Z.log](https://github.com/user-attachments/files/19825970/home-assistant_bms_ble_2025-04-20T10-35-03.589Z.log)

Ichhabekeinenase, 2025-04-20:
Here is another one!

[home-assistant_bms_ble_2025-04-20T10-04-19.628Z.log](https://github.com/user-attachments/files/19825982/home-assistant_bms_ble_2025-04-20T10-04-19.628Z.log)

Ichhabekeinenase, 2025-04-20:
Here is another one!

[home-assistant_bms_ble_2025-04-20T10-04-19.628Z.log](https://github.com/user-attachments/files/19825990/home-assistant_bms_ble_2025-04-20T10-04-19.628Z.log)

patman15, 2025-04-20:
The first log shows an issue with the proxy:
```
2025-04-20 12:01:30.806 WARNING (MainThread) [aioesphomeapi.connection] bluetooth-proxy @ 192.168.1.122: Connection error occurred: Ping response not received after 90.0 seconds
2025-04-20 12:01:33.910 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for bluetooth-proxy @ 192.168.1.122: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.122', port=6053))]: [Errno 113] Connect call failed ('192.168.1.122', 6053) (SocketAPIError)
2025-04-20 12:02:48.079 DEBUG (MainThread) [custom_components.bms_ble] F100216000124300034: device communication failed: A8:17:10:67:C8:3C - A8:17:10:67:C8:3C: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address A8:17:10:67:C8:3C was found: The proxy/adapter is out of connection slots or the device is no longer reachable
```
which hints to a WiFi issue, the connection to the proxy could be bad. In the second one it look like the BMS does not answer, which could be caused by long delays. You can try to increase the timeout to 5s (which will be in the next release):
https://github.com/patman15/BMS_BLE-HA/blob/97f5bd4e24c18648ddb944b51a54e8e26e28e61a/custom_components/bms_ble/plugins/basebms.py#L43

Let me know if one of the hints helps.

Best regards,
Patrick

Ichhabekeinenase, 2025-04-21:
Ok after some Testing i looked in the ESPHome log! Everytime it connects and then Disconnects! The Log in Home Assistant stays the Same!
I have Tried the Timeout Patch! Should i increase the Timeout lenght?
I have tried another ESP32 and its the same Output on both!

Have you seen something like this before?

```
[11:09:48][I][esp32_ble_client:263]: [0] [A8:17:10:67:C8:3C] Connected
[11:09:48][D][esp32_ble_tracker:116]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[11:09:48][D][esp32_ble_tracker:282]: Starting scan...
[11:09:48][D][esp32_ble_client:191]: [0] [A8:17:10:67:C8:3C] ESP_GATTC_REG_FOR_NOTIFY_EVT
[11:09:50][D][esp32_ble_client:416]: [0] [A8:17:10:67:C8:3C] Event 46
[11:09:50][D][esp32_ble_client:297]: [0] [A8:17:10:67:C8:3C] cfg_mtu status 0, mtu 247
[11:09:50][D][esp32_ble_client:191]: [0] [A8:17:10:67:C8:3C] ESP_GATTC_WRITE_DESCR_EVT
[11:09:50][D][esp32_ble_client:191]: [0] [A8:17:10:67:C8:3C] ESP_GATTC_WRITE_CHAR_EVT
[11:09:53][I][esp32_ble_client:148]: [0] [A8:17:10:67:C8:3C] Disconnecting (conn_id: 0).
[11:09:53][D][esp32_ble_tracker:116]: connecting: 0, discovered: 0, searching: 0, disconnecting: 1
[11:09:53][D][esp32_ble_client:191]: [0] [A8:17:10:67:C8:3C] ESP_GATTC_CLOSE_EVT
[11:09:53][D][esp32_ble_tracker:116]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
```

closed 2025-04-26 by patman15