patman15/BMS_BLE-HA #276: No supported devices found via Bluetooth

MichaNW, 2025-04-26:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have an ECTIVE LC 80 LT 12V LiFePO4 and when I go to BLE Battery Management it says “No supported devices found via Bluetooth”. What else can I do?

{"name":"xxxxx\f","address":"30:55:44:37:92:F2","rssi":-127,"manufacturer_data":{"65535":"3055443792f2"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000ffe0-0000-1000-8000-00805f9b34fb","f000ffc0-0451-4000-b000-000000000000"],"source":"2C:CF:67:61:FC:46","connectable":true,"time":1745649294.4176664,"tx_power":null}

patman15, 2025-04-26:
Does your device match the listed name patterns?
> Ective, Topband batteries (show up as $PFLAC…, NWJ20…, ZM20…)

If not, please let me know the first 5 characters of the device name.

MichaNW, 2025-04-27:
{"name":"Petroll20126\f","address":"30:55:44:37:92:F2","rssi":-127,"manufacturer_data":{"65535":"3055443792f2"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000ffe0-0000-1000-8000-00805f9b34fb","f000ffc0-0451-4000-b000-000000000000"],"source":"2C:CF:67:61:FC:46","connectable":true,"time":1745649294.4176664,"tx_power":null}

patman15, 2025-04-27:
That is the original name from the manufacturer, or did you change it?

MichaNW, 2025-04-27:
this is the name I got from the dealer :-) and is made up of my name and a number on the battery

patman15, 2025-04-27:
Hmm, you can either try to rename your battery via the app to match the patterns listed above or edit the integration to recognize your pattern (will be overwritten by updates). I'll have a look if I can find a way to detect the ECTIVE batteries without using the name. I will let you know if I find one.

MichaNW, 2025-04-27:
Great,
Thank you very much for your help

Prima,
Vielen dank für deine Hilfe

patman15, 2025-04-27:
Try the attached [branch](https://github.com/patman15/BMS_BLE-HA/tree/276-ective-lc-80-lt---no-supported-devices-found-via-bluetooth) via manual install. I removed the name pattern. Just hoping it does not interfere with other devices ... Let me know if it works so I can add it to the next release. Thanks!

MichaNW, 2025-04-27:
you are the best, it worked.
Many thanks for your help

closed 2025-04-27 by patman15

patman15, 2025-07-26:
@MichaNW unfortunately I will need to revoke this change with the next version, as it conflicts with #404. In case you need to re-add your battery the name needs to be `NWJ20`*. You can rename it in HA as you like, but the device name needs to be like this. Sorry for that.