patman15/BMS_BLE-HA #282: Eco Worthy BMS wird mit HA nicht erkannt

mmeudy, 2025-04-30:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Hallo,
ich versuche gerade die Daten meiner ECO Worthy Batterie in den HA zu bekommen.
Da habe ich die Intergartion gefunden.
Leider wird mein BMS nicht erkannt.

Ich habe die Patman App auch probiert, mit dieser verbindet sich das BMS aber die ausgelesen Daten stimmen absolut nicht.
Bei der Patman App kann ich die Adresse der Batterie angeben und damit findet er sie, das geht beim BLE leider nicht.

Wie kann ich mit dem BLE die Batterie finden ?

Gruß
Marcus

mmeudy, 2025-04-30:
Das wird mir für die Batterie angezeigt. 

{"name":"ECO0AA8","address":"4E:FC:FA:D6:0A:A8","rssi":-80,"manufacturer_data":{"64590":"fad60aa8"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb","00000001-0000-1000-8000-00805f9b34fb"],"source":"EC:62:60:8C:14:3A","connectable":false,"time":1746008298.4872713,"tx_power":null}

mmeudy, 2025-04-30:
Könnten Sie mir sagen wie ich es manuell installieren kann ?

patman15, 2025-04-30:
Please see the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) and use the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/282-eco-worthy-bms-wird-mit-ha-nicht-erkannt).

closed 2025-05-07 by patman15