patman15/BMS_BLE-HA #284: ECO-WORTHY server rack battery BMS support/ not detected

sircambridge, 2025-05-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

[image]

I recently got an ECO-WORTHY 100ah server rack battery. turns out the BLE advertisement name is different from the supported one. here is the attached info

`{"name":"ECO-LFP48100-3U-8EC7","address":"AA:C2:37:01:8E:C7","rssi":-70,"manufacturer_data":{"49834":"37018ec700014c03"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"F4:4E:FC:7A:4D:C4","connectable":true,"time":1746083069.327507,"tx_power":null}`

unw1red, 2025-05-01:
I have the same batteries and had a similar issue. I had to use their iOS app to rename each battery starting with "DP04S" and then they all connected and reported proper data.

patman15, 2025-05-02:
@sircambridge, @unw1red thanks for the info, I will add the `ECO-LFP` name pattern to the integration.

patman15, 2025-05-05:
@sircambridge, @unw1red can you please try the [branch](https://github.com/patman15/BMS_BLE-HA/tree/284-eco-worthy-server-rack-battery-bms-support-not-detected) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) and let me know if it works?

unw1red, 2025-05-05:
Installed. Looking through and things look the same. Should I delete my batteries and add them again?

patman15, 2025-05-05:
> Should I delete my batteries and add them again?

Yes please, but this time keep the name `ECO-LFP48100`... There shouldn't be the need to rename to `DP04S`...

unw1red, 2025-05-05:
Done and added batteries back. Everything seems to be working as usual. Should anything appear different?

unw1red, 2025-05-05:
[image]

In use

unw1red, 2025-05-05:
I have one battery that is discovered in the scans, but will not connect.

{"name":"ECO-LFP48100-04","address":"AA:C2:37:01:39:6F","rssi":-89,"manufacturer_data":{"49834":"3701396f00011507"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"CC:50:E3:A9:26:BE","connectable":true,"time":1746451619.3341558,"tx_power":null}

unw1red, 2025-05-05:
1. BLE Proxy (ESP32) is hanging right in front of rack full of all five batteries.
2. I have two proxies with each set to three connections.

patman15, 2025-05-05:
Try to check the advertisement monitor to see to which proxy which BMS is connected and try to fix the RSSI value. Too close is not necessarily good. With a value of -89 dBm it's highly unlikely that it will work.

closed 2025-05-07 by patman15