patman15/BMS_BLE-HA #285: Read all data through master battery

unw1red, 2025-05-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

With experience in Eco-worthy server rack batteries and JBD BMS, I know that the master battery can read all parameters from other "chained" batteries. Is it at all possible that when reading the transmissions from the master battery, that the rest of the packs data is in this payload? Is it possible?

patman15, 2025-05-02:
Hi! Could be possible. I have such an implementation, e.g. for Seplos. I would need a BT packet trace using the vendor app in order to identify the necessary commands. Can you provide that?

unw1red, 2025-05-02:
I’m willing to help in anyway I can. I don’t quite understand what you are asking for, but I am technical and shouldn’t need too much guidance. ---Kelly D Griffin479.372.1790 Voice

patman15, 2025-05-02:
The guidance from the issue template hopefully helps:

> To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

unw1red, 2025-05-02:
I hope this is what you are needing. DP04S01 is the master battery.

[home-assistant_2025-05-02T16-55-43.355Z.log](https://github.com/user-attachments/files/20015959/home-assistant_2025-05-02T16-55-43.355Z.log)

patman15, 2025-05-02:
No, sorry, these are the messages from the integration itself. Please check the links in my previous message.

unw1red, 2025-05-03:
I am sorry. I have tried to do this with two Android devices (Pixel 9 and OnePlus) and they are not rooted. Therefore, I do not have access to the capture file.

patman15, 2025-05-03:
You do not need a rooted phone. You just have to have adb and developer mode enabled.

unw1red, 2025-05-03:
I do. I preform the capture with Bluetooth HCI Packet enabled. USB Debug is enabled. Turn off/on Bluetooth. Run local EW app. Stop capture. Can't find the capture file. bug report does not have the file inside of the .zip file. I have used both a Pixel 9 Pro and a OnePlus 10 Plus.

patman15, 2025-05-03:
Thanks for the efforts, doesn't sound too promising though. The only other option you could try is recording directly via Wireshark, as also described in the guide as the first option.
Maybe that works for you?

unw1red, 2025-05-03:
I got it to work, but now the file is in the data directory which I do not have permission for because my phone is not rooted. Any ideas?

patman15, 2025-05-04:
`adb root` and the `adb pull` did not work?
The second option is still using Wireshark, right?
Other than that I would need to crawl the Internet. My phone has it on the debug zip, so I lack a reference. Sorry.

unw1red, 2025-05-04:
I don't give up easily. I figured out how to capture BT on my Mac. Here is a capture after running through different commands on the JBD app. Summary is looking at the stack and then I looked at pack data for all of the attached batteries (1-5).

[Capture of EW BLE Traffic.zip](https://github.com/user-attachments/files/20027560/Capture.of.EW.BLE.Traffic.zip)

patman15, 2025-05-05:
Thanks a lot! I had a quick look and the data looks like what I need. Will see if I can make use of it and let you know soon.

patman15, 2025-05-11:
Hi @unw1red! I had a look again and it seems that this mode of communication uses a totally different protocol than before. Also, in the log there is no proof that the communication really goes via a single device, as it does not show the BT addresses for some reason. Since the benefit over multiple connections is rather limited, I decided to defer this implementation till I have finished some other important changes I need to do.
Let me know if I should rethink my priority due to urgent needs from your side.
Best regards,
Patrick

unw1red, 2025-05-12:
I appreciate you looking into this. Please close request.

patman15, 2025-09-13:
@unw1red I had a look again and unfortunately I just now noticed that the communication is legacy Bluetooth and not BLE so I cannot implement this feature for this specific BMS, as Home Assistant via it's underlying library (Bleak) does not support legacy Bluetooth but only BLE. Sorry for that.