patman15/BMS_BLE-HA #286: Support for eco-worthy

shaf, 2025-05-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hi,

Please be kind to add support for the following device

{"name":"ECO-WORTHY 0B_5AD4","address":"E2:E0:5A:78:3C:31","rssi":-86,"manufacturer_data":{"57570":"5a783c31"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"30:AE:A4:0A:6B:06","connectable":true,"time":1746117657.0329983,"tx_power":null}

Thank you

### Additional context

Playstore app

https://play.google.com/store/apps/details?id=uni.UNI5B1C700&hl=en_GB

patman15, 2025-05-02:
Thanks for the info I will take care of it soon..

patman15, 2025-05-05:
@@shaf can you please check the linked branch (see *development* section on the right) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation)? Please send me a debug log to verify. Note: your `RSSI`  is pretty low with -86 dBm.

shaf, 2025-05-05:
> @[@shaf](https://github.com/shaf) can you please check the linked branch (see _development_ section on the right) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation)? Please send me a debug log to verify. Note: your `RSSI` is pretty low with -86 dBm.

Thank you sir,

Installed as 

`
git checkout origin/286-support-for-eco-worthy
rsync -av custom_components/bms_ble /config/custom_components/
`

Unfortunately sees the device but end up in an initialization loop, do you know what BMS this battery is using ?

`
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTFYX55Q2274YV2589Q3G6FH version=1 domain=bms_ble title=ECO-WORTHY 0B_5AD4 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=E2:E0:5A:78:3C:31>
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] initializing ECO-WORTHY BW02, BT address: E2:E0:5A:78:3C:31
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for ECO-WORTHY 0B_5AD4 (E2:E0:5A:78:3C:31) as ECO-WORTHY BW02
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble] ECO-WORTHY 0B_5AD4: advertisement: {'name': 'ECO-WORTHY 0B_5AD4', 'address': 'E2:E0:5A:78:3C:31', 'rssi': -63, 'manufacturer_data': {57570: b'Zx<1'}, 'service_data': {}, 'service_uuids': ['0000fff0-0000-1000-8000-00805f9b34fb'], 'source': '30:AE:A4:0A:6B:06', 'advertisement': AdvertisementData(local_name='ECO-WORTHY 0B_5AD4', manufacturer_data={57570: b'Zx<1'}, service_uuids=['0000fff0-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-63), 'device': BLEDevice(E2:E0:5A:78:3C:31, ECO-WORTHY 0B_5AD4), 'connectable': True, 'time': 528899.331174421, 'tx_power': None}
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble] ECO-WORTHY 0B_5AD4: BMS data update
2025-05-05 11:16:55.535 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] connecting BMS
2025-05-05 11:16:55.535 DEBUG (MainThread) [bleak_retry_connector] E2:E0:5A:78:3C:31 - E2:E0:5A:78:3C:31: Connection attempt: 1
2025-05-05 11:16:55.915 DEBUG (MainThread) [bleak_retry_connector] E2:E0:5A:78:3C:31 - E2:E0:5A:78:3C:31: Connected after 1 attempts
2025-05-05 11:16:56.212 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b'\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r\x8d\r\xee\r\xe5\r\xd4\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\x87\x00v\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18p\x04')
2025-05-05 11:16:56.212 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
2025-05-05 11:16:56.414 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00d\x00d\x05\x85\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00I2")
2025-05-05 11:16:56.414 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
2025-05-05 11:16:56.474 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b'\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r\x8d\r\xee\r\xe5\r\xd3\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\x87\x00v\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xb5B')
2025-05-05 11:16:56.474 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
2025-05-05 11:16:56.673 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00d\x00d\x05\x85\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00I2")
2025-05-05 11:16:56.673 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
2025-05-05 11:16:56.897 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b'\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r\x8d\r\xee\r\xe5\r\xd4\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\x87\x00v\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18p\x04')
2025-05-05 11:16:56.897 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
[... cut ...] x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb4\xf2")
2025-05-05 11:17:01.084 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] invalid frame type: 'e2'
2025-05-05 11:17:01.109 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] disconnected from BMS
`

patman15, 2025-05-05:
> Unfortunately sees the device but end up in an initialization loop, do you know what BMS this battery is using ?

Thanks for the log. That is for sure an ECO-WORTHY BMS but it uses a different frame format than all others I have seen so far. I need to do a small adaptation to compensate for that change.

shaf, 2025-05-08:
Sorry Patrick,

I have no idea what I'm doing but I merged in the device_id and made the following change to 
plugins/ecoworthy_bms.py

```
#    _HEAD: Final[tuple] = (b"\xa1", b"\xa2", b"\xe2")
#    _HEAD: Final[tuple] = (b"\xe2", b"\xe0")
    _HEAD: Final[tuple] = (b"\xa1", b"\xa2", b"\xe2", b"\xe0")
```

No luck getting data :(

```
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTR80PEGFRCPPQXNDK7H3SGE version=1 domain=bms_ble title=ECO-WORTHY 0B_5AD4 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=E2:E0:5A:78:3C:31>
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] initializing ECO-WORTHY BW02, BT address: E2:E0:5A:78:3C:31
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for ECO-WORTHY 0B_5AD4 (E2:E0:5A:78:3C:31) as ECO-WORTHY BW02
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble] ECO-WORTHY 0B_5AD4: advertisement: {'name': 'ECO-WORTHY 0B_5AD4', 'address': 'E2:E0:5A:78:3C:31', 'rssi': -61, 'manufacturer_data': {57570: b'Zx<1'}, 'service_data': {}, 'service_uuids': ['0000fff0-0000-1000-8000-00805f9b34fb'], 'source': '30:AE:A4:0A:6B:06', 'advertisement': AdvertisementData(local_name='ECO-WORTHY 0B_5AD4', manufacturer_data={57570: b'Zx<1'}, service_uuids=['0000fff0-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-61), 'device': BLEDevice(E2:E0:5A:78:3C:31, ECO-WORTHY 0B_5AD4), 'connectable': True, 'time': 806644.63918423, 'tx_power': None, 'raw': b'\x02\x01\x06\x03\x02\xf0\xff\x13\tECO-WORTHY 0B_5AD4\x07\xff\xe2\xe0Zx<1'}
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble] ECO-WORTHY 0B_5AD4: BMS data update
2025-05-08 16:25:58.559 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] connecting BMS
2025-05-08 16:25:58.560 DEBUG (MainThread) [bleak_retry_connector] E2:E0:5A:78:3C:31 - E2:E0:5A:78:3C:31: Connection attempt: 1
2025-05-08 16:25:58.888 DEBUG (MainThread) [bleak_retry_connector] E2:E0:5A:78:3C:31 - E2:E0:5A:78:3C:31: Connected after 1 attempts
2025-05-08 16:25:59.135 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00c\x00d\x05D\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00&\x83")
2025-05-08 16:25:59.343 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b'\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r(\r)\r0\r+\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\xdc\x00\xca\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18M-')
2025-05-08 16:25:59.540 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00c\x00d\x05D\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00&\x83")
2025-05-08 16:25:59.748 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b'\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r)\r)\r0\r+\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\xdc\x00\xca\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\x7f\xf3')
2025-05-08 16:25:59.949 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00c\x00d\x05D\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00&\x83")
2025-05-08 16:26:00.163 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa2\x00\x00\x08\x03V\x00\x04\r\'\r)\r0\r*\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x02\x00\xdc\x00\xca\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\xfc\x18\x0c\x07")
2025-05-08 16:26:00.356 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00c\x00d\x05D\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00&\x83")
[... cut ...]
2025-05-08 16:26:03.942 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] RX BLE data: bytearray(b"\xe2\xe0Zx<1\xa1\x00\x00\x08\x03D\x00\x08\x00c\x00d\x05D\x00\x00\x00\x00\'\x10\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00&\x83")
2025-05-08 16:26:03.966 DEBUG (MainThread) [custom_components.bms_ble.ecoworthy_bms::ECO-WORTHY 0B_5AD4:3C31)] disconnected from BMS

```

shaf, 2025-05-08:
> Please update from the branch should work now. I need to do a bunch of cleanups later ...

You are a genius ! Thank you, it's working ❤️

[image]

patman15, 2025-05-08:
Ok great! It's getting late but I need to clean the code before merging. Can I please ask you to test it again, after I did the cleaning?

shaf, 2025-05-08:
Sure thing Patrick, will do ... btw some more data, I presume I need to RTFM to get this extra data into sensors

```
2025-05-08 20:18:25.421 DEBUG (MainThread) [custom_components.bms_ble] ECO-WORTHY 0B_5AD4: BMS data sample {'battery_level': 100, 'voltage': 13.52, 'current': 0.0, 'problem_code': 0, 'design_capacity': 100.0, 'cell_count': 4, 'temp_sensors': 2, 'cell#0': 3.365, 'cell#1': 3.386, 'cell#2': 3.391, 'cell#3': 3.384, 'temp#0': 18.9, 'temp#1': 17.3, 'delta_voltage': 0.026, 'cycle_charge': 100.0, 'cycle_capacity': 1352.0, 'power': 0.0, 'battery_charging': False, 'temperature': 18.1, 'problem': False}
2025-05-08 20:18:25.421 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching ECO-WORTHY 0B_5AD4 data in 0.477 seconds (success: True)
```

patman15, 2025-05-11:
@shaf could you please one final time update the branch and check if it still works as expected? If you are ok, I would merge and close the topic. Thanks!

shaf, 2025-05-11:
> [@shaf](https://github.com/shaf) could you please one final time update the branch and check if it still works as expected? If you are ok, I would merge and close the topic. Thanks!

Hi Patrick,

Sorry it's been a busy weekend and I've only just had a moment to take a look at this, so I've installed the updated branch as

```
➜  BMS_BLE-HA git:(98e3245) git fetch --all
remote: Enumerating objects: 39, done.
remote: Counting objects: 100% (30/30), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 39 (delta 25), reused 26 (delta 25), pack-reused 9 (from 1)
Unpacking objects: 100% (39/39), 12.35 KiB | 972.00 KiB/s, done.
From https://github.com/patman15/BMS_BLE-HA
   98e3245..22a8456  286-support-for-eco-worthy -> origin/286-support-for-eco-worthy
   fc9a9ef..ad8bd5d  main                       -> origin/main
➜  BMS_BLE-HA git:(98e3245) git status
HEAD detached at 98e3245
nothing to commit, working tree clean
➜  BMS_BLE-HA git:(98e3245) git checkout origin/286-support-for-eco-worthy
Previous HEAD position was 98e3245 add additional protocol version
HEAD is now at 22a8456 fixed tests
➜  BMS_BLE-HA git:(22a8456) git status
HEAD detached at origin/286-support-for-eco-worthy
nothing to commit, working tree clean
➜  BMS_BLE-HA git:(22a8456)
➜  BMS_BLE-HA git:(22a8456) rsync -nav custom_components/bms_ble /config/custom_components/
sending incremental file list
bms_ble/plugins/
bms_ble/plugins/ecoworthy_bms.py

sent 1,034 bytes  received 25 bytes  2,118.00 bytes/sec
total size is 186,102  speedup is 175.73 (DRY RUN)
➜  BMS_BLE-HA git:(22a8456) rsync -av custom_components/bms_ble /config/custom_components/
sending incremental file list
bms_ble/plugins/
bms_ble/plugins/ecoworthy_bms.py

sent 7,443 bytes  received 45 bytes  14,976.00 bytes/sec
total size is 186,102  speedup is 24.85
➜  BMS_BLE-HA git:(22a8456)
```

Restarted HA and at face value it looks fine but I'll let it run overnight and look at the charts and update you in the morning
I can't see anything broken but would be prudent to get some runtime.

Thank you for your help - will update in the morning

shaf, 2025-05-11:
Hi @patman15 

Tests are good, feel free to merge and close this issue
Thank you for your support !

closed 2025-05-12 by patman15